esphome/issues #898: State from homeassistant binary_sensor in lambda is Nan

ulph, 2019-12-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

hass.io on a rpi3b

**ESP (ESP32/ESP8266, Board/Sonoff):**

wemos lolin32

**Affected component:**

lambda, on homeassistant imported binary sensors

**Description of problem:**

Logic in lambdas based on homeassistant imported binary sensors is broken, as the values never change. 

See log for a hint. The underlying values seemingly are 'on'/'off' etc.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ulfsar_esp32_1
  platform: ESP32
  board: lolin32

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ulfsar Esp32 1 Fallback Hotspot"
    password: "zcy6coHh6JdS"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

time:
  - platform: homeassistant
    id: esptime
    
sensor:
  - platform: homeassistant
    id: balkongdorr_temperature
    entity_id: sensor.balkongdorr_temperature
    internal: true
  - platform: homeassistant
    id: sovrum_temperature
    entity_id: sensor.sovrum_temperature
    internal: true
  - platform: homeassistant
    id: ytterdorr_ov_temperature
    entity_id: sensor.ytterdorr_ov_temperature
    internal: true
  - platform: homeassistant
    id: trappa_temperature
    entity_id: sensor.trappa_temperature
    internal: true
  - platform: homeassistant
    id: ytterdorr_nv_temperature
    entity_id: sensor.ytterdorr_nv_temperature
    internal: true

  - platform: homeassistant
    id: balkongdorr_openclose
    entity_id: binary_sensor.balkongdorr_openclose
    internal: true
  - platform: homeassistant
    id: sovrum_humidity
    entity_id: sensor.sovrum_humidity
    internal: true
  - platform: homeassistant
    id: ytterdorr_ov_openclose
    entity_id: binary_sensor.ytterdorr_ov_openclose
    internal: true
  - platform: homeassistant
    id: trappa_humidity
    entity_id: sensor.trappa_humidity
    internal: true
  - platform: homeassistant
    id: ytterdorr_nv_openclose
    entity_id: binary_sensor.ytterdorr_nv_openclose
    internal: true

  - platform: homeassistant
    id: balkongdorr_vibration
    entity_id: binary_sensor.balkongdorr_vibration
    internal: true
  - platform: homeassistant
    id: sovrum_pressure
    entity_id: sensor.sovrum_pressure
    internal: true
  - platform: homeassistant
    id: ytterdorr_ov_vibration
    entity_id: binary_sensor.ytterdorr_ov_vibration
    internal: true
  - platform: homeassistant
    id: trappa_pressure
    entity_id: sensor.trappa_pressure
    internal: true
  - platform: homeassistant
    id: ytterdorr_nv_vibration
    entity_id: binary_sensor.ytterdorr_nv_vibration
    internal: true
    
font:
  - file: 'slkscr.ttf'
    id: font_slkscr
    size: 8
    
i2c:
  sda: 22
  scl: 21
  scan: False

display:
  - platform: ssd1306_i2c
    model: "SH1106 128x64"
    address: 0x3C
    lambda: |-
    
      auto row = 0;
      it.printf(0, (row++)*8, id(font_slkscr), TextAlign::TOP_LEFT , "sovr");
      it.printf(0, (row++)*8, id(font_slkscr), TextAlign::TOP_LEFT , "vard");
      it.printf(0, (row++)*8, id(font_slkscr), TextAlign::TOP_LEFT , "balk");
      it.printf(0, (row++)*8, id(font_slkscr), TextAlign::TOP_LEFT , "h.ov");
      it.printf(0, (row++)*8, id(font_slkscr), TextAlign::TOP_LEFT , "h.nv");

      row = 0;
      it.printf(127, (row++)*8, id(font_slkscr), TextAlign::TOP_RIGHT , "%.1fC", id(sovrum_temperature).has_state()?id(sovrum_temperature).state:99.9);
      it.printf(127, (row++)*8, id(font_slkscr), TextAlign::TOP_RIGHT , "%.1fC", id(trappa_temperature).has_state()?id(trappa_temperature).state:99.9);
      it.printf(127, (row++)*8, id(font_slkscr), TextAlign::TOP_RIGHT , "%.1fC", id(balkongdorr_temperature).has_state()?id(balkongdorr_temperature).state:99.9);
      it.printf(127, (row++)*8, id(font_slkscr), TextAlign::TOP_RIGHT , "%.1fC", id(ytterdorr_ov_temperature).has_state()?id(ytterdorr_ov_temperature).state:99.9);
      it.printf(127, (row++)*8, id(font_slkscr), TextAlign::TOP_RIGHT , "%.1fC", id(ytterdorr_nv_temperature).has_state()?id(ytterdorr_nv_temperature).state:99.9);

      row = 0;
      it.printf(127-32, (row++)*8, id(font_slkscr), TextAlign::TOP_RIGHT , "%.1fp", id(sovrum_humidity).has_state()?id(sovrum_humidity).state:-1);
      it.printf(127-32, (row++)*8, id(font_slkscr), TextAlign::TOP_RIGHT , "%.1fp", id(trappa_humidity).has_state()?id(trappa_humidity).state:-1);
      it.printf(127-32, (row++)*8, id(font_slkscr), TextAlign::TOP_RIGHT , "%s", id(balkongdorr_openclose).has_state()?(id(balkongdorr_openclose).state?"OPN":"-"):".");
      it.printf(127-32, (row++)*8, id(font_slkscr), TextAlign::TOP_RIGHT , "%s", id(ytterdorr_ov_openclose).has_state()?(id(ytterdorr_ov_openclose).state?"OPN":"-"):".");
      it.printf(127-32, (row++)*8, id(font_slkscr), TextAlign::TOP_RIGHT , "%s", id(ytterdorr_nv_openclose).has_state()?(id(ytterdorr_nv_openclose).state?"OPN":"-"):".");

      row = 0;
      it.printf(127-64, (row++)*8, id(font_slkscr), TextAlign::TOP_RIGHT , "%.0f", id(sovrum_pressure).has_state()?id(sovrum_pressure).state:-1);
      it.printf(127-64, (row++)*8, id(font_slkscr), TextAlign::TOP_RIGHT , "%.0f", id(trappa_pressure).has_state()?id(trappa_pressure).state:-1);
      it.printf(127-64, (row++)*8, id(font_slkscr), TextAlign::TOP_RIGHT , "%s", id(balkongdorr_vibration).has_state()?(id(balkongdorr_vibration).state?"MTN":"-"):".");
      it.printf(127-64, (row++)*8, id(font_slkscr), TextAlign::TOP_RIGHT , "%s", id(ytterdorr_ov_vibration).has_state()?(id(ytterdorr_ov_vibration).state?"MTN":"-"):".");
      it.printf(127-64, (row++)*8, id(font_slkscr), TextAlign::TOP_RIGHT , "%s", id(ytterdorr_nv_vibration).has_state()?(id(ytterdorr_nv_vibration).state?"MTN":"-"):".");

```

**Logs (if applicable):**

```
[00:46:46][W][homeassistant.sensor:014]: Can't convert 'on' to number!
[00:46:46][D][sensor:092]: 'ytterdorr_nv_vibration': Sending state nan  with 1 decimals of accuracy

```

**Additional information and things you've tried:**

https://www.home-assistant.io/integrations/binary_sensor/

brandond, 2019-12-05:
You're trying to import a binary_sensor as a sensor. One of these things has a boolean as the state, the other has a number. Did you maybe mean:

```yaml
binary_sensor:
  - platform: homeassistant
    id: ytterdorr_nv_vibration
    entity_id: binary_sensor.ytterdorr_nv_vibration
    internal: true
```

ulph, 2019-12-05:
... Yes obviously. Sleep deprived night shift hacking, thanks a lot.

closed 2019-12-05 by OttoWinter