esphome/issues #900: Pullup with Dallas temperatur sensor

JAltberg, 2019-12-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio in Raspberry Pi 3b+

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Affected component:**

Dallas temperature sensor, water resistent probe

**Description of problem:**
Hi!

All I read is that you have to use 4,7k ohm resistor to use dallas temperature sensor. I can't find any code to use pullup (in my case ESP8266). Why is that?

In ESP8266 I have read that the internal is 30-100k ohm, that should work as well. Saw that you can use pullup for DHT temperature sensor.

Thanks!
Jocke

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

hgregorian, 2019-12-06:
I use dallas sensors (wired in "normal" mode) in a few places with a 4.7k ohm resistor.  As long as you've got it wired up in normal or parasitic mode properly, just follow the procedure here: https://esphome.io/components/sensor/dallas.html

There's no specific "code" consideration you need to make to make it work.

brandond, 2019-12-07:
I think you're asking about code to use the internal pull-up instead of an external resistor to 3v3? You can't.

OttoWinter, 2019-12-07:
Dallas integration already uses pullups for all 1-wire operations.

As @brandond said, it won't work though (or at least it won't work well), you should use an external resistor instead.

closed 2019-12-07 by OttoWinter