esphome/issues #901: Can't connect to API while receiving UART data

DJTerentjev, 2019-12-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io/Docker Raspberry PI
**ESP (ESP32/ESP8266, Board/Sonoff):**
Wemos D1mini

**Affected component:**
https://esphome.io/custom/uart.html

**Description of problem:**
I am trying to get data from RS485-UART converter. I tried different pins of D1 mini and get data. 
But if I try to flash D1 mini with OTA (UART is connected) it get flushed with no problem, but then can't connect to API.  Got errors: WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.80: timed out). Re-Trying in 30 seconds. But same time I can flush D1 mini again. If I disconnect UART or data line of RS485 and flush D1 mini, the I can get connection. Then I can connect data line again and see the incoming data.
Looks like data from UART disturb connection to API. I tried a lot of time and same result if there is data incoming to UART, then no connection to API. 

I use next script to reed data:
```
#include "esphome.h"
int byteReceived;

class MyCustomComponent : public Component, public UARTDevice {
 public:
  MyCustomComponent(UARTComponent *parent) : UARTDevice(parent) {}

  void setup() override {
    // nothing to do here
  }
  void loop() override {
    if (available()) {
    ESP_LOGD("custom", "Receiving...");
  while (available()) {
    byteReceived = read();   
	ESP_LOGD("custom", "Data reseived is: %f",  byteReceived);
    }
   }
  }
}; 
```
**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: my_name
  platform: ESP8266
  board: d1_mini
  includes:
    - my_custom_component2.h
wifi:
  ssid: my_ssid
  password: my_password
  manual_ip:
    static_ip: 192.168.1.80
    gateway: 192.168.1.1
    subnet: 255.255.255.0
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "My_name Fallback Hotspot"
    password: "ooooopop"

captive_portal:

# Enable logging
logger:
  level: debug

# Enable Home Assistant API
api:
  password: "my_password"

ota:
  password: "my_password"

# Example configuration entry

uart:
  id: uart_bus
  tx_pin: D7
  rx_pin: D6
  baud_rate: 9600

custom_component:
- lambda: |-
    auto my_custom = new MyCustomComponent(id(uart_bus));
    return {my_custom};
```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/my.yaml...
INFO Starting log output from 192.168.1.80 using esphome API
INFO Connecting to 192.168.1.80:6053 (192.168.1.80)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.80: [Errno 111] Connection refused). Re-Trying in 1 seconds
INFO Connecting to 192.168.1.80:6053 (192.168.1.80)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.80: [Errno 111] Connection refused). Re-Trying in 1 seconds
INFO Connecting to 192.168.1.80:6053 (192.168.1.80)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.80: [Errno 111] Connection refused). Re-Trying in 2 seconds
INFO Connecting to 192.168.1.80:6053 (192.168.1.80)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.80: [Errno 111] Connection refused). Re-Trying in 3 seconds
INFO Connecting to 192.168.1.80:6053 (192.168.1.80)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.80: [Errno 111] Connection refused). Re-Trying in 5 seconds
INFO Connecting to 192.168.1.80:6053 (192.168.1.80)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.80: [Errno 111] Connection refused). Re-Trying in 7 seconds
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.80: [Errno 111] Connection refused). Re-Trying in 11 seconds
INFO Connecting to 192.168.1.80:6053 (192.168.1.80)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.80: [Errno 111] Connection refused). Re-Trying in 17 seconds
INFO Connecting to 192.168.1.80:6053 (192.168.1.80)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.80: [Errno 111] Connection refused). Re-Trying in 25 seconds
```

**Additional information and things you've tried:**

OttoWinter, 2019-12-07:
I don't see any issue with your code (except maybe `%f` being the wrong format for type `uint8_t`, but that won't cause this problem).

You're going to need to get logs from over USB (serial logs). The logs you posted are really of little help here.

DJTerentjev, 2019-12-07:
Please see if this log is sufficient. One I saw the data I just pull D1 mini power otherwise log will be flooded with data from UART. And one more thing, with same code, but using PollingComponent booting going fine. But in this case, I guess, some data going to be missing. Is it correct? Thank you very much for such a good tool. I am new to programming and ESPhome make it much easer. Could you please help me with two questions. This setup() function will be complete all the code inside or in some cases it can be triggered by API before completed and start over?  And second one. If there is a way to print variable to log other than state of sensor?
This is log from serial port:
```
Using 'COM9' as serial port.
Unexpected error: could not open port 'COM9': PermissionError(13, 'Access is denied.', None, 5)
[21:24:01][D][wifi:319]: Found networks:
[21:24:01][I][wifi:365]: - 'My_home' (52:8B:F3:6B:F7:39) ▂▄▆█
[21:24:01][D][wifi:366]:     Channel: 8
[21:24:01][D][wifi:367]:     RSSI: -51 dB
[21:24:01][I][wifi:365]: - 'My_home' (04:BF:6D:22:45:9C) ▂▄▆█
[21:24:01][D][wifi:366]:     Channel: 3
[21:24:01][D][wifi:367]:     RSSI: -62 dB
[21:24:01][I][wifi:193]: WiFi Connecting to 'My_home'...
[21:24:03][I][wifi:423]: WiFi Connected!
[21:24:03][C][wifi:283]:   SSID: 'My_home'
[21:24:03][C][wifi:284]:   IP Address: 192.168.1.53
[21:24:03][C][wifi:286]:   BSSID: 52:8B:F3:6B:F7:39
[21:24:03][C][wifi:287]:   Hostname: 'mercury'
[21:24:03][C][wifi:291]:   Signal strength: -54 dB ▂▄▆█
[21:24:03][C][wifi:295]:   Channel: 8
[21:24:03][C][wifi:296]:   Subnet: 255.255.255.0
[21:24:03][C][wifi:297]:   Gateway: 192.168.1.1
[21:24:03][C][wifi:298]:   DNS1: (IP unset)
[21:24:03][C][wifi:299]:   DNS2: (IP unset)
[21:24:03][D][wifi:432]: Disabling AP...
[21:24:03][C][ota:029]: Over-The-Air Updates:
[21:24:03][C][ota:030]:   Address: 192.168.1.53:8266
[21:24:03][C][ota:032]:   Using Password.
[21:24:03][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
[21:24:03][I][app:058]: setup() finished successfully!
[21:24:03][I][ota:379]: Waiting for OTA attempt.
[21:24:03][I][app:100]: ESPHome version 1.14.3 compiled on Dec  7 2019, 21:23:01
[21:24:03][C][wifi:415]: WiFi:
[21:24:03][C][wifi:283]:   SSID: 'My_home'
[21:24:03][C][wifi:284]:   IP Address: 192.168.1.53
[21:24:03][C][wifi:286]:   BSSID: 52:8B:F3:6B:F7:39
[21:24:03][C][wifi:287]:   Hostname: 'mercury'
[21:24:03][C][wifi:291]:   Signal strength: -55 dB ▂▄▆█
[21:24:03][C][wifi:295]:   Channel: 8
[21:24:03][C][wifi:296]:   Subnet: 255.255.255.0
[21:24:03][C][wifi:297]:   Gateway: 192.168.1.1
[21:24:03][C][wifi:298]:   DNS1: (IP unset)
[21:24:03][C][wifi:299]:   DNS2: (IP unset)
[21:24:03][C][logger:175]: Logger:
[21:24:03][C][logger:176]:   Level: DEBUG
[21:24:03][C][logger:177]:   Log Baud Rate: 115200
[21:24:03][C][logger:178]:   Hardware UART: UART0
[21:24:03][C][captive_portal:169]: Captive Portal:
[21:24:03][C][ota:029]: Over-The-Air Updates:
[21:24:03][C][ota:030]:   Address: 192.168.1.53:8266
[21:24:03][C][ota:032]:   Using Password.
[21:24:03][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
[21:24:09][D][ota:072]: Starting OTA Update from 192.168.1.56...
[21:24:09][D][ota:243]: OTA in progress: 0.1%
[21:24:10][D][ota:243]: OTA in progress: 14.3%
[21:24:11][D][ota:243]: OTA in progress: 27.5%
[21:24:12][D][ota:243]: OTA in progress: 40.6%
[21:24:13][D][ota:243]: OTA in progress: 54.4%
[21:24:14][D][ota:243]: OTA in progress: 66.0%
[21:24:15][D][ota:243]: OTA in progress: 79.3%
[21:24:17][D][ota:243]: OTA in progress: 92.4%
[21:24:17][I][ota:267]: OTA update finished!
[21:24:17][I][app:137]: Rebooting safely...
[21:24:17][W][wifi_esp8266:337]: Event: Disconnected ssid='My_home' bssid=52:8B:F3:6B:F7:39 reason='Association Leave'
[21:24:17]
[21:24:17] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[21:24:17]
[21:24:17]load 0x4010f000, len 1384, room 16 
[21:24:17]tail 8
[21:24:17]chksum 0x2d
[... 8 lines cut ...]
[21:24:24][C][wifi:033]: Setting up WiFi...
[21:24:24][D][wifi:304]: Starting scan...
[21:24:25][D][custom:013]: Receiving...
[21:24:25][D][custom:016]: Data reseived is: -5486124068793688683255936251187209270074392635932332070112001988456197381759672947165175699536362793613284725337872111744958183862744647903224103718245670299614498700710006264535590197791934024641512541262359795191593953928908168990292758500391456212260452596575509589842140073806143686060649302051520512.000000
[21:24:25][D][custom:016]: Data reseived is: -5486124068793792227217307376177438024762263809808135130022363935411170353265780834745015092813701490773359721696876380609693142579844097156903921501119915190729522022652588784361152235317876401781602123082373543687274670624183701899770715820937754767373866954335145884566923056755081430489417966138949632.000000
[21:24:25][D][custom:016]: Data reseived is: -5486124068793960333884004026161574120608454656806497744935657449291008824652167758109930813663969022633246186373848017119498134379370261827584096254492218661010383415924075698901476484477641672903159326507807629480262187141219272976334928882530568892146233793991967163061276605542768592032595326657363968.000000
[21:24:25][D][custom:016]: Data reseived is: -5486124068793999315140049626157895534138006157559741249843087829321116296277996609904693879368378885093509714414884918339163060014042995664263557066868404972959568666537753824012276310369761156061781286722401040678636104014734767718726630462030351877600695669854418474 [... cut ...] 781602123082373543687274670624183701899770715820937754767373866954335145884566923056755081430489417966138949632.000000
[21:24:25][D][custom:016]: Data reseived is: -5486124068793960333884004026161574120608454656806497744935657449291008824652167758109930813663969022633246186373848017119498134379370261827584096254492218661010383415924075698901476484477641672903159326507807629480262187141219272976334928882530568892146233793991967163061276605542768592032595326657363968.000000
Serial port closed!
```

DJTerentjev, 2019-12-07:
I think I pulled power early, so I repeat try to catch useful information. This what I've got:
```
[22:09:21]Soft WDT reset
[22:09:21]
[22:09:21]>>>stack>>>
[22:09:21]
[22:09:21]ctx: cont
[22:09:21]sp: 3ffffb30 end: 3fffffc0 offset: 01b0
[22:09:21]3ffffce0:  3fff1b57 00000000 3fff1b0f 4021aeb2  
[22:09:21]3ffffcf0:  00000005 00000166 3fff0984 3ffea2e0  
[22:09:21]3ffffd00:  00000005 3fff19f1 3fff0984 40217d88  
[22:09:21]3ffffd10:  3ffffd50 3ffffd40 00000008 40218199  
[22:09:21]3ffffd20:  3fff19f1 000001d3 3fff199c 402217d4  
[22:09:21]3ffffd30:  3ffffe00 3ffffdf0 3fff0984 402182e9  
[22:09:21]3ffffd40:  00004bc6 3fff19f1 3fff199c 4020532c  
[22:09:21]3ffffd50:  3ffea2e0 00000010 00000000 40205222  
[22:09:21]3ffffd60:  3ffffe00 3ffffdf0 00000010 3ffea2e0  
[22:09:21]3ffffd70:  00000005 00000015 3fff199c 402054a4  
[22:09:21]3ffffd80:  00000010 40255d30 00000000 3fff07c0  
[22:09:21]3ffffd90:  00000000 4bc6a7f0 43126e97 00000102  
[22:09:21]3ffffda0:  3fff213c 3ffea2e0 3fff2148 4020ae1c  
[22:09:21]3ffffdb0:  3ffffe00 3ffffdf0 00000010 402062f0  
[22:09:21]3ffffdc0:  00000000 00000000 4bc6a7f0 4020ae48  
[22:09:21]3ffffdd0:  3ffffe00 3ffffdf0 00000010 4020644f  
[22:09:21]3ffffde0:  3ffffe00 3ffffdf0 00000010 40206470  
[22:09:22]3ffffdf0:  001ea385 00000005 3fff07c0 00000000  
[22:09:22]3ffffe00:  00000085 ff000000 3fff07c0 4021c81a  
[22:09:22]3ffffe10:  3fff1dd4 00001a7f 3fff1dd4 4020b9d9  
[22:09:22]3ffffe20:  3fff1dd4 00000005 3fff07c0 4021c688  
[22:09:22]3ffffe30:  3fff1304 00000005 3fff07c0 4021c70c  
[22:09:22]3ffffe40:  3fff1dd4 00000005 3fff07c0 4020a9f8  
[22:09:22]3ffffe50:  3fff1304 00000006 3fffff20 4021baa4  
[22:09:22]3ffffe60:  00000100 0000000c 00000003 40225b8e  
[22:09:22]3ffffe70:  00000000 00000000 00000000 3fff2104  
[22:09:22]3ffffe80:  3fff07c0 3ffffed8 00000000 4020bfe2  
[22:09:22]3ffffe90:  3fff1f88 00000000 00000000 3fff1fb8  
[22:09:22]3ffffea0:  00000000 00000000 00000001 4021dea0  
[22:09:22]3ffffeb0:  3501a8c0 4021dea0 0101a8c0 4021dea0  
[22:09:22]3ffffec0:  00ffffff 4021dea0 00000000 4021dea0  
[22:09:22]3ffffed0:  00000000 00000000 3fff1ef8 00000000  
[22:09:22]3ffffee0:  00000000 3fff1f18 00000000 00000000  
[22:09:22]3ffffef0:  00000001 4021dea0 3501a8c0 4021dea0  
[22:09:22]3fffff00:  0101a8c0 4021dea0 00ffffff 4021dea0  
[22:09:22]3fffff10:  00000000 4021dea0 00000000 00000000  
[22:09:22]3fffff20:  3fff1d1c 4021dea0 4020b9f8 4020ba40  
[22:09:22]3fffff30:  0101a8c0 4021dea0 00ffffff 4021dea0  
[22:09:22]3fffff40:  00000000 4021dea0 00000000 3fff1dbc  
[22:09:22]3fffff50:  3fff1dc0 3fff1dc0 0101a8c0 4021dea0  
[22:09:22]3fffff60:  00ffffff 4021dea0 00000000 4021dea0  
[22:09:22]3fffff70:  00000000 feefeffe feefeffe feefeffe  
[22:09:22]3fffff80:  3fff2074 00000001 feefeffe feefeffe  
[22:09:22]3fffff90:  feefeffe feefeffe feefeffe 3fff0ab8  
[22:09:22]3fffffa0:  3fffdad0 00000000 3fff0a88 4021a0ac  
[22:09:22]3fffffb0:  feefeffe feefeffe 3ffe8534 40100d1d  
[22:09:22]<<<stack<<<
[22:09:22]
[22:09:22] ets Jan  8 2013,rst cause:1, boot mode:(3,7)
[22:09:22]
[22:09:22]load 0x4010f000, len 1384, room 16 
[22:09:22]tail 8
[22:09:22]chksum 0x2d
[22:09:22]csum 0x2d
[22:09:22]v8b899c12
[22:09:22]~ld
[22:09:22][I][logger:156]: Log initialized
[22:09:22][C][ota:364]: There have been 1 suspected unsuccessful boot attempts.
[22:09:22][I][app:028]: Running through setup()...
[22:09:22][C][uart:111]: Setting up UART bus...
[22:09:22][C][wifi:033]: Setting up WiFi...
[22:09:22][D][wifi:304]: Starting scan...
[... 123 lines cut ...]
[22:09:29]3fffff70:  00000000 feefeffe feefeffe feefeffe  
[22:09:29]3fffff80:  3fff2074 00000001 feefeffe feefeffe  
[22:09:29]3fffff90:  feefeffe feefeffe feefeffe 3fff0ab8  
[22:09:29]3fffffa0:  3fffdad0 00000000 3fff0a88 4021a0ac  
[22:09:29]3fffffb0:  feefeffe feefeffe 3ffe8534 40100d1d  
[22:09:29]<<<stack<<<
[22:09:29]
[22:09:29] ets Jan  8 2013,rst cause:1, boot mode:(3,7)
[22:09:29]
[22:09:29]load 0x4010f000, len 1384, room 16 
[22:09:29]tail 8
[22:09:29]chksum 0x2d
[22:09:29]csum 0x2d
[22:09:29]v8b899c12
[22:09:29]~ld
[22:09:29][I][logger:156]: Log initialized
[22:09:29][C][ota:364]: There have been 3 suspected unsuccessful boot attempts.
[22:09:29][I][app:028]: Running through setup()...
[22:09:29][C][uart:111]: Setting up UART bus...
[22:09:29][C][wifi:033]: Setting up WiFi...
[22:09:29][D][wifi:304]: Starting scan...
[22:09:29][D][custom:013]: Receiving...
```
Look like it will continue again and again.
Actually I am not sure, may be this is my prints to log?

brandond, 2019-12-07:
Your code is getting stuck in a loop printing the UART data which triggers the watchdog timer, whick reboots the node repeatedly until it eventually goes into safe mode.

DJTerentjev, 2019-12-07:
If there is a way I can delay printing until completely booted?

brandond, 2019-12-07:
Probably. Individually printing every byte that comes in (and printing it as a float, even) is not going to work very well though, no matter when you do it. The device will be stuck in a loop until the serial rx buffer is empty.

DJTerentjev, 2019-12-07:
Totally agree with that. See I moved a bit forward using PollingComponent. I was able to convert data to string and publish to HA sensor. But what I don't understand which function is better to use loop() or update()  when I have continues data inflow and want consonantly analyse it.  In Arduino you have only loop() and you can publish value any time you want.

glmnet, 2019-12-12:
loop is called by the component itself and is called 60 times per seconds if I'm not mistaken

update is called by `PollingComponent` and happens depending on the `update_interval` you setup.

so if you are doing sensor reading and publishing probably update is best suited. But if you need to attend some logic continuously then loop is more preferable.

In any case you code should "end" that is reach the end of the function as quickly as possible for stuff to go smooth.

DJTerentjev, 2019-12-13:
You are right. Thank you.

closed 2019-12-13 by DJTerentjev