esphome/issues #903: Dallas - random "Scratch pad checksum invalid!"

wizu, 2019-12-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome in docker, running latest build - esphome/esphome-armv7

**ESP (ESP32/ESP8266, Board/Sonoff):**

esp32 devkitv1

**Affected component:**

https://esphome.io/components/sensor/dallas.html

I have two one wire DS18b20 sensors connected to the same pin. From time to time one of them returns NaN and in logs there is "Scratch pad checksum invalid!" but I have no idea what to do about it.
I power my devkit from random usb charger, and sensors from ESP.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
    name: esp32_devkitv1_b
    platform: ESP32
    board: esp32dev

dallas:
    - pin: 23

sensor:
    - platform: dallas
      address: 0x0300000B87031928
      name: "Temperature Furnace Water"
    - platform: dallas
      address: 0x5F01186C54FAFF28
      name: "Temperature Furnace room"
```

**Logs (if applicable):**

```
[19:27:25][C][logger:175]: Logger:
[19:27:25][C][logger:176]:   Level: VERBOSE
[19:27:25][C][logger:177]:   Log Baud Rate: 115200
[19:27:25][C][logger:178]:   Hardware UART: UART0
[19:27:25][C][dallas.sensor:070]: DallasComponent:
[19:27:25][C][dallas.sensor:071]:   Pin: GPIO23 (Mode: INPUT)
[19:27:25][C][dallas.sensor:072]:   Update Interval: 60.0s
[19:27:25][D][dallas.sensor:077]:   Found sensors:
[19:27:25][D][dallas.sensor:080]:     0x0300000B87031928
[19:27:25][D][dallas.sensor:080]:     0x5F01186C54FAFF28
[19:27:25][C][dallas.sensor:085]:   Device 'Temperature Furnace Water'
[19:27:25][C][dallas.sensor:085]:     Unit of Measurement: '°C'
[19:27:25][C][dallas.sensor:085]:     Accuracy Decimals: 1
[19:27:25][C][dallas.sensor:085]:     Icon: 'mdi:thermometer'
[19:27:25][V][dallas.sensor:085]:     Unique ID: 'dallas-0300000B87031928'
[19:27:25][C][dallas.sensor:093]:     Address: 0x0300000B87031928
[19:27:25][C][dallas.sensor:094]:     Resolution: 12
[19:27:25][C][dallas.sensor:085]:   Device 'Temperature Furnace room'
[19:27:25][C][dallas.sensor:085]:     Unit of Measurement: '°C'
[19:27:25][C][dallas.sensor:085]:     Accuracy Decimals: 1
[19:27:25][C][dallas.sensor:085]:     Icon: 'mdi:thermometer'
[19:27:25][V][dallas.sensor:085]:     Unique ID: 'dallas-5F01186C54FAFF28'
[19:27:25][C][dallas.sensor:093]:     Address: 0x5F01186C54FAFF28
[19:27:25][C][dallas.sensor:094]:     Resolution: 12
[19:27:25][C][ota:029]: Over-The-Air Updates:
[19:27:25][C][ota:030]:   Address: 192.168.1.160:3232
[19:27:25][C][api:095]: API Server:
[19:27:25][C][api:096]:   Address: 192.168.1.160:6053
[19:27:27][V][api.connection:567]: Hello from client: 'Home Assistant 0.102.2 (192.168.1.181)'
[19:27:27][D][api.connection:583]: Client 'Home Assistant 0.102.2 (192.168.1.181)' connected successfully!
[19:28:12][D][dallas.sensor:148]: 'Temperature Furnace Water': Got Temperature=49.6°C
[19:28:12][V][sensor:013]: 'Temperature Furnace Water': Received new state 49.562500
[19:28:12][D][sensor:092]: 'Temperature Furnace Water': Sending state 49.56250 °C with 1 decimals of accuracy
[19:28:12][D][dallas.sensor:148]: 'Temperature Furnace room': Got Temperature=34.8°C
[19:28:12][V][sensor:013]: 'Temperature Furnace room': Received new state 34.750000
[19:28:12][D][sensor:092]: 'Temperature Furnace room': Sending state 34.75000 °C with 1 decimals of accuracy
[19:29:12][D][dallas.sensor:148]: 'Temperature Furnace Water': Got Temperature=49.4°C
[19:29:12][V][sensor:013]: 'Temperature Furnace Water': Received new state 49.375000
[19:29:12][D][sensor:092]: 'Temperature Furnace Water': Sending state 49.37500 °C with 1 decimals of accuracy
[19:29:12][W][dallas.sensor:141]: 'Temperature Furnace room' - Scratch pad checksum invalid!
[19:29:12][V][sensor:013]: 'Temperature Furnace room': Received new state nan
[19:29:12][D][sensor:092]: 'Temperature Furnace room': Sending state nan °C with 1 decimals of accuracy
[19:29:21][I][ota:046]: Boot seems successful, resetting boot loop counter.
[19:30:12][D][dallas.sensor:148]: 'Temperature Furnace Water': Got Temperature=49.2°C
[19:30:12][V][sensor:013]: 'Temperature Furnace Water': Received new state 49.187500
[19:30:12][D][sensor:092]: 'Temperature Furnace Water': Sending state 49.18750 °C with 1 decimals of accuracy
[19:30:12][W][dallas.sensor:141]: 'Temperature Furnace room' - Scratch pad checksum invalid!
[19:30:12][V][sensor:013]: 'Temperature Furnace room': Received new state nan
[19:30:12][D][sensor:092]: 'Temperature Furnace room': Sending state nan °C with 1 decimals of accuracy
[19:31:12][D][dallas.sensor:148]: 'Temperature Furnace Water': Got Temperature=49.1°C
[19:31:12][V][sensor:013]: 'Temperature Furnace Water': Received new state 49.062500
[19:31:12][D][sensor:092]: 'Temperature Furnace Water': Sending state 49.06250 °C with 1 decimals of accuracy
[19:31:12][W][dallas.sensor:141]: 'Temperature Furnace room' - Scratch pad checksum invalid!
[19:31:12][V][sensor:013]: 'Temperature Furnace room': Received new state nan
[19:31:12][D][sensor:092]: 'Temperature Furnace room': Sending state nan °C with 1 decimals of accuracy

```

I've tried diffrent power source and reconnecting components.

OttoWinter, 2019-12-07:
If it just happens time to time then it's likely just a bad connection somewhere, too long wires or incorrect resistance.

You can filter out NAN values using the `filter_out` sensor filter.

wizu, 2019-12-07:
I've tried yet another power source - this time my main phone charger and all the issues were gone. It looks like, a cheap phone chargers aren't good enough.

closed 2019-12-07 by wizu