esphome/issues #905: Esphome Addon - Compile Fail - Segmentation Errror (Raspbian on RP4)

Lipown, 2019-12-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Raspberry Pi 4 Running Hassio ESPHome addon
**ESP (ESP32/ESP8266, Board/Sonoff):**

Weemos D1
**Affected component:**

https://esphome.io/components/esphome.html

**Description of problem:**
I wanted to try ESP Home on Weemos D1 mini instead of Tasmota. I have Raspberry Pi 4 running Hassio. When I want to compile binary the output is error below. 

I found that I need to update firmware of the Pi which I did, however cmd uname -a shows still the old one. The rpi-update saying I have the latest.

To be sure I used that tool on SSD I am running system from plus from SD card. Both boots are updated but I still obtain the error in compilation of binary.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: tt
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: ""
  password: ""

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Tt Fallback Hotspot"
    password: "2z7JDI3ZY5zq"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
INFO Reading configuration /config/esphome/test.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/test
Processing test (board: esp01_1m; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/test/.pioenvs/test/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/components/api/user_services.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/components/api/util.cpp.o
xtensa-lx106-elf-g++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [/data/test/.pioenvs/test/src/esphome/components/api/util.cpp.o] Error 4
Compiling /data/test/.pioenvs/test/src/esphome/components/captive_portal/captive_portal.cpp.o
========================== [FAILED] Took 5.92 seconds ==========================
```

**Additional information and things you've tried:**
```
pi@rpi:~ $ sudo uname -a
Linux rpi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
pi@rpi:~ $ sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Your firmware is already up to date
```

OttoWinter, 2019-12-07:
I assume the referenced issue is https://github.com/esphome/issues/issues/677 ?

The other option is that it's just the RPi running out of memory. So please also check if the RPi has low memory while executing the compiler.

Also, I'm not sure if the fix described in #677 also applies for RPi 4s. The messages posted there look like it was exclusive to the RPi 3 (the RPi 4 uses a different chipset, so the fix by the rpi team may not be applicable to both RPis)

Lipown, 2019-12-07:
Yes,  you are right.  Available RAM when executing is arround 2500MB.

Lipown, 2019-12-07:
Solved by using another approach of setting the SSD as boot device https://jamesachambers.com/raspberry-pi-4-usb-boot-config-guide-for-ssd-flash-drives/

The following guide does not correctly update firmware when using rpi-update, so use the first one: https://www.stewright.me/2019/10/run-raspbian-from-a-usb-or-ssd-on-a-raspberry-pi-4/

closed 2019-12-07 by Lipown