esphome/issues #907: wind sensor TX23 not working with the current TX20/TX23 implementation

SeByDocKy, 2019-12-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

According to https://next.esphome.io/components/sensor/tx20.html  the TX23 should be also working with the current implementation with the following pinout
TX23 wires <=> connexion
red(VCC)                      +3.3
yellow(GND)                GND
brown/copper(data)    GPIOpin
green(DTR)                  GND

Since officially the yellow and green should be connected togather to GND, I used a 3 wires servo cable to connect the sensor to my ESP32  where the I seletect as data pin the GPIO2

**Problem-relevant YAML-configuration entries:**
In my ESPhome config file I have in the sensor section
```
- platform: tx20
    wind_speed:
      name: "station_meteo_vitesse_vent"
      unit_of_measurement: "m/s"
      icon: "mdi:weather-windy"
      accuracy_decimals: 1
    wind_direction_degrees:
      name: "station_meteo_direction_vent"
    pin:
      number: GPIO2
      #mode: INPUT
    id: tx20_id
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

And unfortunatly no signal is received ....  I tried also with a 10Ko resistance between the data pin and the VCC ... no more success.

I read here: https://forum.arduino.cc/index.php?topic=62753.0    in post #3 & 4 than probably the TX23 protocol is a little bit different where the green (DTR) should be going to high to low every 2s.

SeByDocKy, 2019-12-07:
I just checked with this arduino lib: https://github.com/egzumer/Arduino-LaCrosse-TX23-Library

with an arduino uno and the same wiring and it's working ..... so clearly there is a bug/missing for the TX23 implementation in ESPhome

SeByDocKy, 2019-12-11:
No news ? I don't how to contact the author of this Addon for ESPhome .... I am sure he can solve the problem in a couple of minutes.

SeByDocKy, 2019-12-15:
Ok I wrote a custom sensor component but it's not the ideal situation since for example the pin data must be defined in the corresponding .h file 
here are my finding if anyone need to integrate the TX23 w/ ESPhome
conf yaml file for an ESP32
```
esphome:
  name: station_meteo
  platform: ESP32
  board: esp32dev
  includes:
    - tx23_custom_sensor.h
  libraries:
    - "https://github.com/egzumer/Arduino-LaCrosse-TX23-Library"
wifi:
  ssid: !secret esphome_ssid
  password: !secret esphome_password

sensor:
- platform: custom
    id: tx23_id
    lambda: |-
      auto TX23 = new TX23CustomSensor();
      App.register_component(TX23);
      return {TX23->speed_sensor, TX23->direction_sensor};

    sensors:
    - name: "station_meteo_vitesse_vent"
      unit_of_measurement: "m/s"
      accuracy_decimals: 1
      icon: "mdi:weather-windy"
    
    - name: "station_meteo_direction_vent"
      unit_of_measurement: "°"
      icon: "mdi:compass-outline" 
```

and the corresponding TX_custom_sensor.h
```
#include "esphome.h"
#include "LaCrosse_TX23.h"

#define DATAPIN 2

class TX23CustomSensor : public PollingComponent, public Sensor {
 public:
  LaCrosse_TX23 anemometer = LaCrosse_TX23(DATAPIN);
  Sensor *speed_sensor = new Sensor();
  Sensor *direction_sensor = new Sensor();
  TX23CustomSensor() : PollingComponent(3000) {}
  void setup() override {
    
  }
  void update() override {
	static const char *dirTable[] = {"N", "NNE", "NE", "ENE", "E", "ESE", "SE", "SSE", "S", "SSW", "SW", "WSW", "W", "WNW", "NW", "NNW"};
	float speed;
	int direction;
	bool reading;
	
	reading = anemometer.read(speed, direction);
   		
	speed_sensor->publish_state(speed);
	direction_sensor->publish_state(direction*22.5);
	
    ESP_LOGD("custom", "reading is: %d", reading);
    ESP_LOGD("custom", "speed is: %3.2f", speed);
	ESP_LOGD("custom", "direction is: %s", dirTable[direction]);
  }
};
```

OttoWinter, 2019-12-15:
CC: @TheEggi (https://github.com/esphome/esphome/pull/275)

TheEggi, 2019-12-26:
Seems that the TX23U has a different start frame - problem is that I don't have the possibility to test this and therefore will not make a change to the code. What could be done is to remove the TX23 portion in the description for now.

SeByDocKy, 2019-12-26:
> Seems that the TX23U has a different start frame - problem is that I don't have the possibility to test this and therefore will not make a change to the code. What could be done is to remove the TX23 portion in the description for now.

I can beta test if you need ....

RyuzakiKK, 2020-02-16:
Same here, I bought a TX23 and it is not working with the "tx20" platform.

@TheEggi if you are willing to work on that I can test it too.

I started to compare the tx20 implementation with the tx23 arduino library https://github.com/egzumer/Arduino-LaCrosse-TX23-Library ( [tx20 reference](https://www.john.geek.nz/2011/07/la-crosse-tx20-anemometer-communication-protocol) and [tx23 reference](https://www.john.geek.nz/2012/08/la-crosse-tx23u-anemometer-communication-protocol/) ) but I can't guarantee that I'll be able to get it to work.

SeByDocKy, 2020-02-16:
> Same here, I bought a TX23 and it is not working with the "tx20" platform.
>
> @TheEggi if you are willing to work on that I can test it too.
>
> I started to compare the tx20 implementation with the tx23 [...]

I managed to write a custom components. I did 2 videos 

https://www.youtube.com/watch?v=1qMPK9LsWVw

https://www.youtube.com/watch?v=J2yG0CH2fS0

My TX23 is working perfectly with ESPhome now.....

RyuzakiKK, 2020-02-16:
@SeByDocKy Awesome, thanks! I'll try it as soon as I have some spare time (maybe later today).

Anyway the biggest difference between the tx20 and the tx23 seems to be the pullup resistor and that you need to pull down TxD for 500ms in order to get a reading. Instead in the tx20 it just transmit data every 2 seconds.

I guess this is the reason why, with the default tx20 library, we didn't get any readings at all.

SeByDocKy, 2020-02-16:
> @SeByDocKy Awesome, thanks! I'll try it as soon as I have some spare time (maybe later today).
>
> Anyway the biggest difference between the tx20 and the tx23 seems to be the pullup resistor and that [...]

As far I remember I tried the original TX20 code with the pullup resistor without success .... Seems there are some other differences as well.

TheEggi, 2020-02-19:
> Same here, I bought a TX23 and it is not working with the "tx20" platform.
>
> @TheEggi if you are willing to work on that I can test it too.
>
> I started to compare the tx20 implementation with the tx23 [...]

The next few months are already quite busy. So I don't think that there will be an update of the implementation in the near future.

Willi24c, 2020-03-23:
Hi,
I am complete rookie and was trying to use TX23 for Home Assistant using ESPHome. The above custom sensor component works fine for me. Only one request: The direction is shown in degrees but i like to see the "human readable" direction as N or E or NNE or .....
Because I am not C (or other) programmer - just interested user - it would be nice if SeByDocKy or Otto or TheEggi or any other specialist could give me some hint how to do that.
Thank you very much in advance.
Harald

TheEggi, 2020-03-24:
> Hi,
> I am complete rookie and was trying to use TX23 for Home Assistant using ESPHome. The above custom sensor component works fine for me. Only one request: The direction is shown in degrees but i [...]

Can be done with the following template:

```
text_sensor:
  - platform: template
    name: "Wind Direction"
    lambda: |-
      return id(tx20_id).get_wind_cardinal_direction();
    update_interval: 2s
```

Willi24c, 2020-03-24:
Hi TheEggi,
first of all, thank you very much for quick answer.

I got error for "return id(tx20_id).get_wind_cardinal_direction();" 
after changing to "tx23_id" the error was gone.

But unfortunately compiling was not succesful.
See the "Failed" message below:
--------------------------------------------------------------------
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/wind32
Processing wind32 (board: esp-wrover-kit; framework: arduino; platform: espressif32@1.11.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <La Crosse RX23 Library> 1.0.0 #92ce726
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <AsyncTCP-esphome> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling /data/wind32/.pioenvs/wind32/src/main.cpp.o
src/main.cpp:132:3: warning: multi-line comment [-Wcomment]
   //   lambda: !lambda "return id(tx23_id).get_wind_cardinal_direction();\nupdate_interval:\
   ^
src/main.cpp: In lambda function:
src/main.cpp:160:22: error: 'const class esphome::custom::CustomSensorConstructor' has no member named 'get_wind_cardinal_direction'
       return tx23_id.get_wind_cardinal_direction();
                      ^
src/main.cpp:161:24: error: unable to find numeric literal operator 'operator""s'
       update_interval: 2s      
                        ^
src/main.cpp:161:7: warning: label 'update_interval' defined but not used [-Wunused-label]
       update_interval: 2s      
       ^
*** [/data/wind32/.pioenvs/wind32/src/main.cpp.o] Error 1
====== [FAILED] Took 14.00 seconds =====

Any idea how to solve? It looks like that your example was taken from TX20 but does not fit to the above custom sensor prepared by SeByDocKy .

By the way, SeByDocKy was writing that the TX23 Data Pin requires 4.7k pullup, but some ESP32 has LED at GPIO02 which acts as pullup. Another story is VCC for the TX23. Some user report that 3.3V from ESP is ok. Unfortunately my experiance is different. My TX23 Windsensor was unstable with 3.3V. Sometimes ok, but often not ok. Finally it was stable with 4.3V (Diode from 5V to TX23 VCC).

Anyway, any help for above problem is welcome.
Harald

Willi24c, 2020-03-24:
Sorry for bad formatting, sooner or later I may find a way to paste in better way.

Willi24c, 2020-03-24:
Hi,
cut and past from e-mail is somethimes troublesome!
After making correct position of "update_interval: 2s" some 
error message is gone, however the remaining error is:
--------------------------------------------------------
src/main.cpp: In lambda function:
src/main.cpp:160:22: error: 'const class esphome::custom::CustomSensorConstructor' has no member named 'get_wind_cardinal_direction'
       return tx23_id.get_wind_cardinal_direction();
                      ^
*** [/data/wind32/.pioenvs/wind32/src/main.cpp.o] Error 1
---------------------------------------------------------
Once again, thank you very much in advance for any solution.
Harald

TheEggi, 2020-03-24:
The ID should be the one from the wind sensor ... my config looks like the following:

```
  - platform: tx20
    id: tx20_id
    wind_speed:
      name: "Windspeed"
    wind_direction_degrees:
      name: "Wind direction degrees"
    pin: 
      number: GPIO04
      mode: INPUT

text_sensor:
  - platform: template
    name: "Wind direction"
    lambda: |-
      return id(tx20_id).get_wind_cardinal_direction();
    update_interval: 2s
```

closed 2020-12-05 by stale[bot]