esphome/issues #909: Bringing Sonoff Basic back online - WEIRD!

Haringstad, 2019-12-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io / Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff Basic
**Affected component:**

**Description of problem:**
I've been fighting with the WiFi issue for some time now, and I guess I found something that might shed light on the issue (at least for SONOFF Basic). With debug OFF (not enabled at all) and reboot_timeout: 0s, some of my sonoffs will not connect to wifi.
The moment that I turn on a 5s delay, and enable "debug:" in the yaml file, they start connecting!
I'll be in the US soon, and I can send one of the sonoffs to you by USPS if you like, so you can test?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

brandond, 2019-12-09:
What version are you on? Most of this has been addressed in recent releases with the wifi tx power stuff.

Haringstad, 2019-12-09:
@brandond I'm on 1.14.3 right now. 
I assume that you refer to the "power_save_mode: none" option?? Because I've been reading a lot regarding the WiFi issues lately, and maybe I just overlooked?

ryrzy, 2019-12-10:
sonoff basic are so strange units... i have 2 pcs that have problems with wifi when connecting to AC, on serial cable everything is ok xD
i found this: https://github.com/arendst/Tasmota/issues/391#issuecomment-341970844
and ordered this capacitors, will see if it helps.

brandond, 2019-12-10:
@Haringstad it looks like it hasn't been added to the docs yet, but @OttoWinter added an `output_power` setting to the wifi config. See https://github.com/esphome/issues/issues/455#issuecomment-552976310 for the full rationale. Between tx_power and fast_connect settings it seems like most stability/connectivity issues can be addressed.

Haringstad, 2019-12-10:
@brandond Yes, I found some info, and I'm looking at it at this time and yes, it looks like there is life back in some. I hope to bring some back to life in the coming weekend, which are all the way across the ocean ;-) But still, enabling "debug:", made the one that had the biggest issue, work. Still weird, right?

brandond, 2019-12-10:
Yes, odd stuff like adding or removing bits of config to restore functionality was one symptoms of the flash access instability issue that is resolved by reducing the tx power.

closed 2020-08-07 by stale[bot]