esphome/issues #910: esp32 unreachable every 3:21 Hours

seanomat, 2019-12-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

hass.io 32bit on raspberry pi 3b+

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 nodeMCU devkit

**Affected component:**

WiFi

**Description of problem:**
I have two esp32 and about 15 esp8266.
Both esp32's and only those show exactly the same behavior: they lose connection to home Assistant and have the state "unavailable" for a few seconds, before connecting again.
This happens reliably every 3:21 hours.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  device_name: livingroom_sculpture_01
  api_pwd: !secret livingroom_sculpture_01_pwd

esphome:
  name: ${device_name}
  platform: ESP32
  board: esp32dev

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  ap:
    ssid: "${device_name}"
    password: '${api_pwd}'

captive_portal:

logger:
  baud_rate: 0

#web_server:
#  port: 80

api:
  password: '${api_pwd}'

ota:
  password: '${api_pwd}'

switch:
  - platform: restart
    name: "${device_name}_restart"
  - platform: template
    name: "IR Audio Power"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x4040
        command: 0xC03F
  - platform: template
    name: "IR Audio Up"
    icon: "mdi:chevron-up-box-outline"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x4040
        command: 0x20DF
  - platform: template
    name: "IR Audio down"
    icon: "mdi:chevron-down-box-outline"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x4040
        command: 0x30CF
  - platform: template
    name: "IR Audio left"
    icon: "mdi:chevron-left-box-outline"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x4040
        command: 0x926D
  - platform: template
    name: "IR Audio right"
    icon: "mdi:chevron-right-box-outline"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x4040
        command: 0x906F
  - platform: template
    name: "IR Audio enter"
    icon: "mdi:keyboard-return"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x4040
        command: 0x10EF
  - platform: template
    name: "IR Audio Vol up"
    icon: "mdi:volume-plus"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x4040
        command: 0x807F
  - platform: template
    name: "IR Audio Vol down"
    icon: "mdi:volume-minus"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x4040
        command: 0x8877
# optoma ep770
#  - platform: template
#    name: "IR Video Power"
#    turn_on_action:
#      remote_transmitter.transmit_nec:
#        address: 35955
#        command: 33150
#Optoma HD39darbee
  - platform: template
    name: "IR Video Power"
    optimistic: true
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x4CB3
        command: 0x40BF
    turn_off_action:
      remote_transmitter.transmit_nec:
        repeat:
          times: 10
          wait_time: 1s
        address: 0x4CB3
        command: 0x748B
  - platform: template
    name: "IR Video Up"
    icon: "mdi:chevron-up-box-outline"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x4CB3
        command: 0x8877
  - platform: template
    name: "IR Video down"
    icon: "mdi:chevron-down-box-outline"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x4CB3
        command: 0x28D7
  - platform: template
    name: "IR Video left"
    icon: "mdi:chevron-left-box-outline"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x4CB3
        command: 0x08F7
  - platform: template
    name: "IR Video right"
    icon: "mdi:chevron-right-box-outline"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x4CB3
        command: 0x48B7
  - platform: template
    name: "IR Video menu"
    icon: "mdi:menu"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x4CB3
        command: 0x708F
  - platform: template
    name: "IR Video enter"
    icon: "mdi:keyboard-return"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x4CB3
        command: 0xF00F
  - platform: template
    name: "IR Video Mode"
    icon: "mdi:television"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x4CB3
        command: 0xA05F
  - platform: template
    name: "IR Video Lamp"
    icon: "mdi:car-light-high"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x4CB3
        command: 0x22DD

light:
  - platform: rgb
    name: ${device_name}_Light_01
    id: light01
    red: output_red
    green: output_green
    blue: output_blue
    effects:
      - random:
      - strobe:
      - flicker:
      - lambda:
          name: swush
          update_interval: 5s
          lambda: |-
            static int state = 0;
            auto call = id(light01).turn_on();
            call.set_transition_length(5000);
            if (state == 0) {
              call.set_rgb(1.0, 0.0, 0.0);
            } else if (state == 1) {
              call.set_rgb(1.0, 1.0, 0.0);
            } else if (state == 2) {
              call.set_rgb(0.0, 1.0, 0.0);
            } else if (state == 3) {
              call.set_rgb(0.0, 1.0, 1.0);
            } else if (state == 4) {
              call.set_rgb(0.0, 0.0, 1.0);
            } else {
              call.set_rgb(1.0, 0.0, 1.0);
            }
            call.perform();
            state += 1;
            if (state == 6)
              state = 0;
output:
  - platform: ledc
    id: output_red
    pin: GPIO16
  - platform: ledc
    id: output_green
    pin: GPIO17
  - platform: ledc
    id: output_blue
    pin: GPIO18
 
i2c:
  sda: 21
  scl: 22
  scan: false

sensor:
  - platform: uptime
    name: "${device_name}_Uptime"
  - platform: bh1750
    name: "${device_name}_Brightness"
    address: 0x23
    update_interval: 30s
    filters:
      - sliding_window_moving_average:
            window_size: 20
            send_every: 10
      - delta: 0.01
  - platform: bme280
    address: 0x76
    update_interval: 30s  
    temperature:
      name: "${device_name}_Temperature"
      filters:
        - offset: -1.5
        - delta: 0.1
        - sliding_window_moving_average:
            window_size: 20
            send_every: 10
      oversampling: 16x
    pressure:
      name: "${device_name}_Pressure"
      filters:
        - delta: 0.1
        - sliding_window_moving_average:
            window_size: 20
            send_every: 10
    humidity:
      name: "${device_name}_Humidity"
      filters:
        - delta: 0.1
        - sliding_window_moving_average:
            window_size: 20
            send_every: 10

remote_transmitter:
  pin: GPIO26
  # Set to 100% when working with RF signals, and 50% if working with IR leds
  carrier_duty_percent: 50%

```

**Logs (if applicable):**

```
[10:24:43][D][sensor:092]: 'livingroom_sculpture_01_Uptime': Sending state 183275.48438 s with 0 decimals of accuracy ERROR Error while reading incoming messages: Error while receiving data: [Errno 104] Connection reset by peer WARNING Disconnected from API: Error while receiving data: [Errno 104] Connection reset by peer INFO Connecting to livingroom_sculpture_01.local:6053 (192.168.17.28) WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 1 seconds INFO Connecting to livingroom_sculpture_01.local:6053 (192.168.17.28) INFO Successfully connected to livingroom_sculpture_01.local [10:25:25][D][bh1750.sensor:071]: 'livingroom_sculpture_01_Brightness': Got illuminance=72.5lx [10:25:25][D][bme280.sensor:197]: Got temperature=21.5°C pressure=1019.7hPa humidity=42.8%

```

**Additional information and things you've tried:**

drthanwho, 2020-02-09:
I've noticed a similar behaviours as well recently. Although it's unclear if this has always been happening and I only now catched it. But after setting up an automation to notify me for an unavailable state it does seem to happen several times every day. 

However in my case it is NodeMCUs so not ESP32s But it is the same message. And on top of that I noticed it happened simultaneously for more than one ESPHOME node (unsure what that means in terms of troubleshooting though) 
```
[18:54:47][D][sensor:092]: 'Koudouni Uptime': Sending state 105942.45312 s with 0 decimals of accuracy
[18:55:09][D][sensor:092]: 'Koudouni WiFi Signal': Sending state -58.00000 dB with 0 decimals of accuracy
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to entrance.local:6053 (192.168.1.20)
INFO Successfully connected to entrance.local
[18:55:54][D][time:030]: Synchronized time: Sun Feb  9 18:55:54 2020
[18:56:09][D][sensor:092]: 'Koudouni WiFi Signal': Sending state -58.00000 dB with 0 decimals of accuracy```

lancelon, 2020-11-17:
Not sure if I have the same issue or a similar issue but I'm running EspHome on an ESP32-POE using ETHERNET for communication. Sadly my sensors are constantly going unavailable. I'm really worried because I thought using ethernet would be super reliable. Sadly this is not turning out to be the case :-( Any ideas?
[image]

lancelon, 2020-11-17:
hmmm...wondering if it's related to #45 because I DO have both the API and MQTT enabled...? Should I select only one??

drthanwho, 2020-11-17:
Depends. If you are using mqtt why do you also have the api. And have you set up its integration? How often does it disconnect and what is the error in its logs when it disconnects?

If you are not sure this is the same issue (so if yours isn't disconnecting every 3:21 hours, I would open a different issue

closed 2021-03-31 by stale[bot]