esphome/issues #914: missing esphome after install

betyar, 2019-12-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Ubuntu 18.04.3 LTS (32 bit) with python 3.6.9

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos Mini

**Affected component:**

N/A

**Description of problem:**
A fresh install of esphome using Python 3 completes successfully but esphome appears to be missing. All other support files exist at _~/.local/lib/python3.6/site-packages/esphome_ with the exception of esphome itself. Can't find it anywhere.

The command to install was this:

> pip3 install esphome

The end result was this (full details below in the log section):

> Successfully built paho-mqtt ifaddr PyYAML platformio tornado voluptuous esptool tabulate pyaes
> Installing collected packages: pytz, tzlocal, paho-mqtt, colorlog, ifaddr, pyserial, PyYAML, bottle, click, colorama, urllib3, certifi, chardet, idna, requests, semantic-version, tabulate, platformio, tornado, voluptuous, six, ecdsa, pyaes, esptool, setuptools, protobuf, esphome
> Successfully installed PyYAML-5.1.2 bottle-0.12.18 certifi-2019.11.28 chardet-3.0.4 click-7.0 colorama-0.4.3 colorlog-4.0.2 ecdsa-0.14.1 esphome-1.14.3 esptool-2.7 idna-2.8 ifaddr-0.1.6 paho-mqtt-1.4.0 platformio-4.0.3 protobuf-3.10.0 pyaes-1.6.1 pyserial-3.4 pytz-2019.3 requests-2.22.0 semantic-version-2.8.3 setuptools-42.0.2 six-1.13.0 tabulate-0.8.6 tornado-5.1.1 tzlocal-2.0.0 urllib3-1.25.7 voluptuous-0.11.7 

When I tried to create a config file using the wizard I got this:

> john@aspire5315:~/Automation/esphome$ esphome wemos.yaml wizard 
>esphome: command not found

I also tried this:

> john@aspire5315:~/Automation/esphome$ python3 esphome wemos.yaml wizard
> python3: can't open file 'esphome': [Errno 2] No such file or directory
 
I read in Changelog - Version 1.10.0a that I needed to install the development version of platformio, so I did and that also seemed to install OK but I still get the same error.

Other stuff, such as esptool, runs with no problem.

I checked other issues relating to starting esphome the first time but they all had to do with an exception being thrown; in my case esphome itself doesn't exist. 

Someone else had the exact same problem [here](https://everythingsmarthome.co.uk/howto/how-to-install-esphome-or-tasmota-on-a-cheap-smart-wifi-rgb-bulb/?unapproved=110&moderation-hash=f99b358644b54fcbdb14a4790da5b809#comment-110) but the solution was to create a binary file from within hassio. I don't use hassio and want to use esphome only from the command line, so I am looking at a solution for a standalone installation of esphome.

**Problem-relevant YAML-configuration entries:**
```yaml
DIDN'T GET THIS FAR

```

**Logs (if applicable):**

```
john@aspire5315:~/Automation/esphome$ pip3 install esphome
Collecting esphome
 Downloading https://files.pythonhosted.org/packages/77/40/2a1bb03a72b374b4a538785925c82fed27e31445412cbb4fd14228b15dac/esphome-1.14.3-py2.py3-none-any.whl (1.4MB)
   100% |████████████████████████████████| 1.4MB 360kB/s 
Collecting tzlocal==2.0.0 (from esphome)
 Downloading https://files.pythonhosted.org/packages/ef/99/53bd1ac9349262f59c1c421d8fcc2559ae8a5eeffed9202684756b648d33/tzlocal-2.0.0-py2.py3-none-any.whl
Collecting paho-mqtt==1.4.0 (from esphome)
 Downloading https://files.pythonhosted.org/packages/25/63/db25e62979c2a716a74950c9ed658dce431b5cb01fde29eb6cba9489a904/paho-mqtt-1.4.0.tar.gz (88kB)
   100% |████████████████████████████████| 92kB 1.0MB/s 
Collecting colorlog==4.0.2 (from esphome)
 Downloading https://files.pythonhosted.org/packages/68/4d/892728b0c14547224f0ac40884e722a3d00cb54e7a146aea0b3186806c9e/colorlog-4.0.2-py2.py3-none-any.whl
Collecting ifaddr==0.1.6 (from esphome)
 Downloading https://files.pythonhosted.org/packages/9f/54/d92bda685093ebc70e2057abfa83ef1b3fb0ae2b6357262a3e19dfe96bb8/ifaddr-0.1.6.tar.gz
Collecting pyserial==3.4 (from esphome)
 Downloading https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl (193kB)
   100% |████████████████████████████████| 194kB 2.0MB/s 
Collecting PyYAML==5.1.2 (from esphome)
 Downloading https://files.pythonhosted.org/packages/e3/e8/b3212641ee2718d556df0f23f78de8303f068fe29cdaa7a91018849582fe/PyYAML-5.1.2.tar.gz (265kB)
[... 96 lines cut ...]
Collecting pyserial!=3.3,<4,>=3 (from platformio==4.1.1b5)
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Collecting requests<3,>=2.4.0 (from platformio==4.1.1b5)
  Using cached https://files.pythonhosted.org/packages/51/bd/23c926cd341ea6b7dd0b2a00aba99ae0f828be89d72b2190f27c11d4b7fb/requests-2.22.0-py2.py3-none-any.whl
Collecting semantic_version<3,>=2.8.1 (from platformio==4.1.1b5)
  Using cached https://files.pythonhosted.org/packages/36/4a/c9f6e279dd1c7919ec1d63fc4b38e0aedd51f26f7232f0783489cf191247/semantic_version-2.8.3-py2.py3-none-any.whl
Collecting tabulate<1,>=0.8.3 (from platformio==4.1.1b5)
Collecting pyelftools<1,>=0.25 (from platformio==4.1.1b5)
  Downloading https://files.pythonhosted.org/packages/bb/2f/bf41f3c3867d6707fa9b872658bb23088a64d0e522e8979f54c694b8cbe1/pyelftools-0.26-py2.py3-none-any.whl (136kB)
    100% |████████████████████████████████| 143kB 964kB/s 
Collecting marshmallow<3,>=2.20.5 (from platformio==4.1.1b5)
  Downloading https://files.pythonhosted.org/packages/e3/07/c7a12d21619abeedf5fca7381f4cf3fa798a26e8d92ad66bc73f5fc58fc1/marshmallow-2.20.5-py2.py3-none-any.whl (50kB)
    100% |████████████████████████████████| 51kB 2.0MB/s 
Collecting certifi>=2017.4.17 (from requests<3,>=2.4.0->platformio==4.1.1b5)
  Using cached https://files.pythonhosted.org/packages/b9/63/df50cac98ea0d5b006c55a399c3bf1db9da7b5a24de7890bc9cfd5dd9e99/certifi-2019.11.28-py2.py3-none-any.whl
Collecting idna<2.9,>=2.5 (from requests<3,>=2.4.0->platformio==4.1.1b5)
  Using cached https://files.pythonhosted.org/packages/14/2c/cd551d81dbe15200be1cf41cd03869a46fe7226e7450af7a6545bfc474c9/idna-2.8-py2.py3-none-any.whl
Collecting chardet<3.1.0,>=3.0.2 (from requests<3,>=2.4.0->platformio==4.1.1b5)
  Using cached https://files.pythonhosted.org/packages/bc/a9/01ffebfb562e4274b6487b4bb1ddec7ca55ec7510b22e4c51f14098443b8/chardet-3.0.4-py2.py3-none-any.whl
Collecting urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 (from requests<3,>=2.4.0->platformio==4.1.1b5)
  Using cached https://files.pythonhosted.org/packages/b4/40/a9837291310ee1ccc242ceb6ebfd9eb21539649f193a7c8c86ba15b98539/urllib3-1.25.7-py2.py3-none-any.whl
Installing collected packages: bottle, click, colorama, pyserial, certifi, idna, chardet, urllib3, requests, semantic-version, tabulate, pyelftools, marshmallow, platformio
  Running setup.py install for platformio ... done
Successfully installed bottle-0.12.18 certifi-2019.11.28 chardet-3.0.4 click-7.0 colorama-0.4.3 idna-2.8 marshmallow-2.20.5 platformio-4.1.1b5 pyelftools-0.26 pyserial-3.4 requests-2.22.0 semantic-version-2.8.3 tabulate-0.8.6 urllib3-1.25.7
```

OttoWinter, 2019-12-10:
Your python site packages are not in your env's PATH.
May I ask how you installed python? Usuaully ubuntu is quite good with setting PATH accordingly (as long as you use the official package from apt).

You can also run it with the python command with `python3 -m esphome` (note the -m).

betyar, 2019-12-10:
Thanks, that did the trick. I installed Python via Ubuntu's Software Center awhile back. Python2 was never installed. I am surprised about the environment path not being set because other pip3 installs, such as esptool (which I also installed awhile ago) ran without problems so I didn't even think to check if the path was the problem. 

Once again  many thanks for your help.

closed 2019-12-10 by betyar