esphome/issues #918: USB upload failed

Anonym-tsk, 2019-12-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

MacOS / Docker (docker-machine)

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini

**Affected component:**

**Description of problem:**
Upload failed (see log)

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: http
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: <...>
  password: <...>
  fast_connect: on

logger:
  level: VERBOSE
```

**Logs (if applicable):**

```
INFO Reading configuration /config/http.yaml...
INFO Detected timezone 'MSK' with UTC offset 3
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/http
Please wait while upgrading PlatformIO...
PlatformIO has been successfully upgraded to 4.1.0!

********************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
- support us with PlatformIO Plus > https://pioplus.com
********************************************************************************

Processing http (board: d1_mini; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP8266HTTPClient> 1.2
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/http/src/esphome/core/component.cpp.o
Compiling .pioenvs/http/src/main.cpp.o
Linking .pioenvs/http/firmware.elf
Building .pioenvs/http/firmware.bin
Creating BIN file ".pioenvs/http/firmware.bin" using ".pioenvs/http/firmware.elf"
Retrieving maximum program size .pioenvs/http/firmware.elf
Checking size .pioenvs/http/firmware.elf
Unhandled exception in thread started by 
========================= [SUCCESS] Took 6.33 seconds =========================
INFO Successfully compiled program.
INFO Running:  esptool.py --before default_reset --after hard_reset --baud 460800 --chip esp8266 --port /dev/ttyUSB0 write_flash 0x0 /config/http/.pioenvs/http/firmware.bin
ERROR Running command failed: 'str' object has no attribute 'decode'
ERROR Please try running esptool.py --before default_reset --after hard_reset --baud 460800 --chip esp8266 --port /dev/ttyUSB0 write_flash 0x0 /config/http/.pioenvs/http/firmware.bin locally.
INFO Upload with baud rate 460800 failed. Trying again with baud rate 115200.
INFO Running:  esptool.py --before default_reset --after hard_reset --baud 115200 --chip esp8266 --port /dev/ttyUSB0 write_flash 0x0 /config/http/.pioenvs/http/firmware.bin
ERROR Running command failed: 'str' object has no attribute 'decode'
ERROR Please try running esptool.py --before default_reset --after hard_reset --baud 115200 --chip esp8266 --port /dev/ttyUSB0 write_flash 0x0 /config/http/.pioenvs/http/firmware.bin locally.

```

**Additional information and things you've tried:**
Broken in this commit: https://github.com/esphome/esphome/pull/793

brandond, 2019-12-12:
What version of Python are you using?

Anonym-tsk, 2019-12-12:
I use latest dev docker container on x86

savikko, 2019-12-13:
I'm having the same problem, environment Windows 10, tried Python 2.7, 3.5, 3.6, 3.7 and 3.8.
Tried also with `-v` -parameter but there is no additional info AFAIK:
```Processing bletracker (board: lolin32; board_build.partitions: partitions.csv; build_flags: -DESPHOME_LOG_LEVEL=ESPHOME_LOG_LEVEL_DEBUG, -Wno-sign-compare, -Wno-unused-but-set-variable, -Wno-unused-variable, -fno-exceptions; framework: arduino; lib_deps: AsyncTCP-esphome@1.1.1, ESPmDNS, Hash; platform: https://github.com/platformio/platform-espressif32.git#feature/stage; upload_speed: 460800)
------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lolin32.html
PLATFORM: Espressif 32 (Stage) 1.11.0-beta.1 #cff56ff (git+https://github.com/platformio/platform-espressif32.git#feature/stage) > WEMOS LOLIN32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 cec3fca (git+https://github.com/espressif/arduino-esp32.git), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for Hash library in registry
Warning! Library `{'name': 'Hash', 'requirements': None}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'name': 'Hash', 'requirements': None}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1 (C:\Users\Tommi Savikko\Projects\tommihome\esphome\bletracker\.piolibdeps\bletracker\AsyncTCP-esphome_ID6798)
|-- <ESPmDNS> 1.0 (C:\Users\Tommi Savikko\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\libraries\ESPmDNS)
|   |-- <WiFi> 1.0 (C:\Users\Tommi Savikko\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\libraries\WiFi)
|-- <Update> 1.0 (C:\Users\Tommi Savikko\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\libraries\Update)
|-- <WiFi> 1.0 (C:\Users\Tommi Savikko\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\libraries\WiFi)
Building in release mode
Unhandled exception in thread started by```

OttoWinter, 2019-12-15:
Likely due to this line: https://github.com/esphome/esphome/pull/793/files?file-filters%5B%5D=.cfg&file-filters%5B%5D=.lint&file-filters%5B%5D=.py&file-filters%5B%5D=.txt&file-filters%5B%5D=.yml&file-filters%5B%5D=No+extension#diff-37d7e28bda3e5a081916cc93a4420a15R121

esptool.py is not using click, and so data is not passed to sys.stdout with already-encoded utf8 form.

closed 2019-12-31 by glmnet