esphome/issues #920: Upload failed when Installing id=6758 @ 1.2.6

gis-cc, 2019-12-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ubuntu /Esphome installed via pip3
**ESP (ESP32/ESP8266, Board/Sonoff):**

platform: ESP32
board: nodemcu-32s
**Affected component:**

**Description of problem:**

Upload failed (see log)
**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test1
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "tttttt"
  password: "33333333"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test1 Fallback Hotspot"
    password: "22222222"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "aaaaaaaa"

ota:
  password: "bbbbbbbb"

```

**Logs (if applicable):**

```
INFO Reading configuration config/test1.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d config/test1
Processing test1 (board: nodemcu-32s; framework: arduino; platform: espressif32@1.11.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Looking for ESPAsyncWebServer-esphome library in registry
LibraryManager: Installing id=6758 @ 1.2.6
EOFError: Compressed file ended before the end-of-stream marker was reached:
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/ccc/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/ccc/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/ccc/.platformio/platforms/espressif32/builder/main.py", line 209:
    target_elf = env.BuildProgram()
  File "/home/ccc/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/platformio.py", line 122:
    _build_project_deps(env)
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/platformio.py", line 47:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "/home/ccc/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/piolib.py", line 1043:
    project.install_dependencies()
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/piolib.py", line 878:
    lm.install(uri)
  File "/usr/local/lib/python3.6/dist-packages/platformio/managers/lib.py", line 359:
    force=force)
  File "/usr/local/lib/python3.6/dist-packages/platformio/managers/package.py", line 703:
    pkg_dir = self._install_from_piorepo(name, requirements)
  File "/usr/local/lib/python3.6/dist-packages/platformio/managers/lib.py", line 220:
    if app.get_setting("strict_ssl") else dl_data['url'], requirements)
  File "/usr/local/lib/python3.6/dist-packages/platformio/managers/package.py", line 483:
    self.unpack(dlpath, tmp_dir)
  File "/usr/local/lib/python3.6/dist-packages/platformio/managers/package.py", line 213:
    return fu.unpack(dest_dir, with_progress=with_progress)
  File "/usr/local/lib/python3.6/dist-packages/platformio/unpacker.py", line 120:
    items = self._unpacker.get_items()
  File "/usr/local/lib/python3.6/dist-packages/platformio/unpacker.py", line 54:
    return self._afo.getmembers()
  File "/usr/lib/python3.6/tarfile.py", line 1761:
    self._load()        # all members, we first have to
  File "/usr/lib/python3.6/tarfile.py", line 2358:
    tarinfo = self.next()
  File "/usr/lib/python3.6/tarfile.py", line 2289:
    self.fileobj.seek(self.offset - 1)
  File "/usr/lib/python3.6/gzip.py", line 368:
    return self._buffer.seek(offset, whence)
  File "/usr/lib/python3.6/_compression.py", line 143:
    data = self.read(min(io.DEFAULT_BUFFER_SIZE, offset))
  File "/usr/lib/python3.6/gzip.py", line 482:
    raise EOFError("Compressed file ended before the "
========================== [FAILED] Took 0.64 seconds ==========================

```

**Additional information and things you've tried:**

brandond, 2019-12-12:
Sounds like you got a bad download from platformio. Either they are having problems on their end, or something on your side is truncating the file (full disk, network proxy or firewall, etc).

gis-cc, 2019-12-13:
> Sounds like you got a bad download from platformio. Either they are having problems on their end, or something on your side is truncating the file (full disk, network proxy or firewall, etc).

How can I uninstall ”LibraryManager: Installing id=6758 @ 1.2.6“
"sudo pip3 uninstall esphome" or "sudo apt remove python3" both can't completely uninstall them.when I open http://192.168.1.111:6052，it always be the same page
`LibraryManager: Installing id=6758 @ 1.2.6
EOFError: Compressed file ended before the end-of-stream marker was reached:
  File "/usr/local/lib/python3.7/dist-packages/platformio/builder/main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")`

gis-cc, 2019-12-16:
> Sounds like you got a bad download from platformio. Either they are having problems on their end, or something on your side is truncating the file (full disk, network proxy or firewall, etc).

it is ok now, thank you very much .When I use China mobile broadband to access the Internet, there will be errors. When I use China Telecom broadband to access the Internet, there will be no such problem.

closed 2020-07-24 by OttoWinter