esphome/issues #921: New device/Hass.io addon/error 404 when clicking edit

prabbit237, 2019-12-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Ubuntu 18.04/Hass.io/Home Assistant 0.102.3/Docker version 19.03.5/Esphome-Docker addon 1.14.3

**ESP (ESP32/ESP8266, Board/Sonoff):**

Any board

**Affected component:**

**Description of problem:**
When setting up a new device, clicking on **edit** results in a "404 not found" error. If I restart the docker container, it loads the .yaml file fine and I can proceed to edit/compile it with no issues. There's probably some step that I'm overlooking. 

[image]

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

ryrzy, 2019-12-12:
you must not restart hassio, just click other page (main page for example) and go back to Esphome page :)

prabbit237, 2019-12-12:
I was just restarting the ESPHome container in HomeAssistant, not the whole Hass.io system but yeah, switching and changing back does get it to load.

closed 2019-12-12 by prabbit237