esphome/issues #924: "Internal: false" does not work

SimonO93, 2019-12-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

esphome + homeassistant on hassbian

**ESP8266**

**Affected component:**

https://esphome.io/components/binary_sensor/index.html

**Description of problem:**
When setting `internal` in configuration sensor is not reported to home assistant.
It doesn't matter if internal is true or false. The only way to see sensor is to remove `internal` element.
I've tried following options:
```yaml
internal: false
internal: no
internal: False
```
In below example only first sensor will be reported which does not contain internal argument.

I need that to use in substitution to hide some sensors in few devices
 
**Problem-relevant YAML-configuration entries:**
```yaml
binary_sensor:
  - <<: !include ../common/sensor_status.yaml
  - platform: gpio
    pin:
      number: D6
      mode: INPUT_PULLUP
    name: "$upper_devicename Lewe"
    device_class: window
  - platform: gpio
    pin:
      number: D7
      mode: INPUT_PULLUP
    name: "$upper_devicename Prawe"
    device_class: window
    internal: false
  - platform: gpio
    pin:
      number: D7
      mode: INPUT_PULLUP
    name: "$upper_devicename Prawe2"
    device_class: window
    internal: no
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

brandond, 2019-12-13:
What version are you on? I just tested this with both `internal: no` and `internal: false` and it works correctly.

OttoWinter, 2019-12-15:
Also works fine here.

Please add more information:
- What are you communicating with HA with? MQTT/API/... ?
- Attach the relevant lines of the *validated* config file (output of validate command).
- Screenshots of what you see in HA. Also try reloading the page.

closed 2019-12-29 by no-response[bot]