esphome/issues #925: static variable in lambda not initialized

guiguid, 2019-12-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio + ESPHome 1.14.3

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 (Wemos Mini D1)

**Affected component:**

Static variable are not initialised in lambda.

**Description of problem:**
```yaml
    lambda: |-
        static float my_err = 0;
        static float my_err_sum = 0;
        static float my_err_last = 0;
        ESP_LOGD("pid","**1** Temp %.1f/%.1f - E:%0.1f/S:%0.1f/L:%0.1f PID %.1f", id(tempsalon).state, id(tempcons).state, my_err, my_err_sum, my_err_last, my_PID);
```

my_err , my_err_sum, my_err_last must be initialized with '0' on first run

It's not the case :  
`[10:19:39][D][pid:704]: **1** Temp 20.8/20.8 - E:nan/S:nan/L:nan PID nan`

Temporary path : 
use 'isnan()' to correctly initialise static var.
```yaml
        if (isnan(my_err)) { my_err = 0;}
        if (isnan(my_err_sum)) { my_err_sum = 0; }
        if (isnan(my_err_last)) { my_err_last = my_err; }
        ESP_LOGD("pid","**2** Temp %.1f/%.1f - E:%0.1f/S:%0.1f/L:%0.1f PID %.1f", id(tempsalon).state, id(tempcons).state, my_err, my_err_sum, my_err_last, my_PID);
```

`[10:19:39][D][pid:708]: **2** Temp 20.8/20.8 - E:0.0/S:0.0/L:0.0 PID nan`

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: template
    name: "PID Output"
    id: pid_output
    lambda: |-
        static float my_err = 0;
        static float my_err_sum = 0;
        static float my_err_last = 0;
        static float my_PID = 0;
        const float Kp = -50; 
        const float Ki = -1;
        const float Kd = -200;
        ESP_LOGD("pid","**1** Temp %.1f/%.1f - E:%0.1f/S:%0.1f/L:%0.1f PID %.1f", id(tempsalon).state, id(tempcons).state, my_err, my_err_sum, my_err_last, my_PID);
        if (isnan(my_err)) { my_err = 0;}
        if (isnan(my_err_sum)) { my_err_sum = 0; }
        if (isnan(my_err_last)) { my_err_last = my_err; }
        ESP_LOGD("pid","**2** Temp %.1f/%.1f - E:%0.1f/S:%0.1f/L:%0.1f PID %.1f", id(tempsalon).state, id(tempcons).state, my_err, my_err_sum, my_err_last, my_PID);
        my_err = float(id(tempsalon).state) - float(id(tempcons).state);     
        my_err_sum += my_err;
        my_PID = (Kp * my_err) + (Ki * my_err_sum) + (Kd * (my_err-my_err_last));     
        my_err_last = my_err;
        ESP_LOGD("pid","Temp %.1f/%.1f - E:%0.1f/%0.1f/%0.1f PID %.1f", id(tempsalon).state, id(tempcons).state, my_err, my_err_sum, (my_err-my_err_last), my_PID);
        if (my_PID > 100) { my_PID = 100;}
        if (my_PID < 0) { my_PID = 0;}
        return my_PID;

    update_interval: 300s
    unit_of_measurement: "%"
    icon: "mdi:water-percent"
    accuracy_decimals: 0
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2019-12-14:
The code is compiled pretty much as is. So the reason / solution should be on the C++ compiler. 

Did you try with 0.0 instead of just 0

Does this code works on plain C++?

OttoWinter, 2019-12-15:
static variables are initialized correctly (plus this is direct C++, no ESPHome intervention here).

The minimal example you posted above also works fine. Please try without changing the my_err etc variables (like in the last example you posted), then the code works fine.

The reason you're getting NANs is because the *sensor* values are NAN because the output is initialized before the sensor's first reading. If you add NAN to any number, you get NAN.

You need to check if the sensor value isnan.

closed 2019-12-15 by OttoWinter

guiguid, 2019-12-16:
> The code is compiled pretty much as is. So the reason / solution should be on the C++ compiler.
> 
> Did you try with 0.0 instead of just 0
> 
> Does this code works on plain C++?

Hi @glmnet 
I've tryed both, 
```yaml
static float my_err = 0.0;
static float my_err = float(0.0);
```
with same résult.
I think you're good, it's compiler related (i'll check the "c" file to confirm)

> static variables are initialized correctly (plus this is direct C++, no ESPHome intervention here).

I think your good, it's probably compiler related (no ESPHome)
 
> The minimal example you posted above also works fine. Please try without changing the my_err etc variables (like in the last example you posted), then the code works fine.

I've allready try, many, many step to identify that. 
Good to see it's working with your configuration, it show that only happen on "my" Hassio + ESPHome 1.14.3.

> The reason you're getting NANs is because the _sensor_ values are NAN because the output is initialized before the sensor's first reading. If you add NAN to any number, you get NAN.
> You need to check if the sensor value isnan.

I've thank that in the beginning of the debug step, but the first call to ESPD_LOG with "my_err" must show "0".  It's not the case.

glmnet, 2019-12-16:
can you try with `int`?
I've seen a SO question saying `static const float ` will not accept initialization per language specification, here is no `const` though

guiguid, 2019-12-16:
interesting : 

```yaml
        static float my_err_float = float(0.0);
        ESP_LOGD("staticdebug","my_err_float %.f", my_err_float) ;
        
        static int my_err_int = int(0);
        ESP_LOGD("staticdebug","my_err_int %d", my_err_int) ;

        static int my_run = 0;
        ESP_LOGD("staticdebug","my_run %d", my_run) ;
        static float my_err = 0;
        ESP_LOGD("staticdebug","my_err %f", my_err) ;
        static float my_err_sum = 0;
        ESP_LOGD("staticdebug","my_err_sum %f", my_err_sum) ;
        static float my_err_last = 0;
        ESP_LOGD("staticdebug","my_err_last %f", my_err_last) ;
        static float my_PID = 0;
        ESP_LOGD("staticdebug","my_PID %f", my_PID) ;
```

```yaml
[19:21:21][D][staticdebug:679]: my_err_float 0
[19:21:21][D][staticdebug:682]: my_err_int 0
[19:21:21][D][staticdebug:685]: my_run 1
[19:21:21][D][staticdebug:696]: my_err nan
[19:21:21][D][staticdebug:698]: my_err_sum nan
[19:21:21][D][staticdebug:700]: my_err_last nan
[19:21:21][D][staticdebug:702]: my_PID nan
```
No logic ? or strange compiler optimization ?
two bug/tips/ features ? : 
float() ou int() seems to be requierd for static declaration...

**very strange :** 
static int my_run = 0;
my_run is intialized **with 1 !!!**

other without strong explicit typing are 'nan' ....

glmnet, 2019-12-16:
> first loop is'nt logged

I was about to ask you about it. GTK

OttoWinter, 2019-12-17:
> (I've seen we don't have slow PID example, I'll do a PR for it.)

There will be a full component for it. See also https://github.com/esphome/esphome/pull/885

glmnet, 2019-12-17:
@guiguid regarding the log stuff, are you running USB (wired) log, or OTA?

If it's OTA I already know "first loop isn't logged" actually ESP boots quite quickly and program starts running ASAP, however Wifi and OTA etc can take a few seconds to come connected. If it is USB, I'd expect the logger to run there, first loop.

What is a bit confusing is that ESPHome dumps the components configuration when OTA logger connects, so you can have all the info available, this is nice but might trick the user thinking the node just booted up.

Having said this, will it be friendly to add an extra line to the log? e.g. between the config and the normal log events saying: "Running loop 1" or "Node uptime is n seconds" etc. Just to hint the users he's not seeing a freshly booted node.

brandond, 2019-12-17:
You only get an extra config dump when using Native API. With MQTT it's a little more clear where the boot logs actually start. It might actually be nice if there was a message indicating whether the config dump a post-boot or on-demand so that folks know what they're looking at.