esphome/issues #926: Communication with HTU21D failed!

Limych, 2019-12-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker installation

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU ESP32

**Affected component:**

HTU21D

**Description of problem:**
Communication with the HTU21D is not established after OTA firmware. After pressing the Reset button, the sensor works fine for a while. But then again it stops transmitting data to HA.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  device_name: "Kitchen"
  <<: !include secrets.yaml

esphome:
  name: kitchen
  platform: ESP32
  board: nodemcu-32s

status_led:
  pin: GPIO2

wifi:
  ssid: $wifi_ssid
  password: !secret wifi_password
  domain: ".lan"

  manual_ip:
    static_ip: 192.168.1.65
    gateway: 192.168.1.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${device_name} Fallback Hotspot"
    password: !secret ota_password

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret ota_password

ota:
  password: !secret ota_password

web_server:
  port: 80
  auth:
    username: admin
    password: !secret web_server_password

i2c:
  - id: bus_indoor
    sda: 21
    scl: 22
    scan: True
  - id: bus_outdoor
    sda: 19
    scl: 23
[... 332 lines cut ...]
[20:46:52][C][bmp280.sensor:117]:     Oversampling: 16x
[20:46:52][C][ccs811:111]: CCS811
[20:46:52][C][ccs811:112]:   Address: 0x5A
[20:46:52][C][ccs811:113]:   Update Interval: 60.0s
[20:46:52][C][ccs811:114]:   CO2 Sensor 'Kitchen eCO2'
[20:46:52][C][ccs811:114]:     Unit of Measurement: 'ppm'
[20:46:52][C][ccs811:114]:     Accuracy Decimals: 0
[20:46:52][C][ccs811:114]:     Icon: 'mdi:periodic-table-co2'
[20:46:52][C][ccs811:115]:   TVOC Sensor 'Kitchen tVOC'
[20:46:52][C][ccs811:115]:     Unit of Measurement: 'ppb'
[20:46:52][C][ccs811:115]:     Accuracy Decimals: 0
[20:46:52][C][ccs811:115]:     Icon: 'mdi:radiator'
[20:46:52][C][ccs811:119]:   Baseline: NOT SET
[20:46:52][C][captive_portal:169]: Captive Portal:
[20:46:52][C][web_server:122]: Web Server:
[20:46:52][C][web_server:123]:   Address: 192.168.1.65:80
[20:46:52][C][web_server:125]:   Basic authentication enabled
[20:46:52][C][ota:029]: Over-The-Air Updates:
[20:46:52][C][ota:030]:   Address: 192.168.1.65:3232
[20:46:52][C][ota:032]:   Using Password.
[20:46:52][C][api:095]: API Server:
[20:46:52][C][api:096]:   Address: 192.168.1.65:6053
```

**Additional information and things you've tried:**

OttoWinter, 2019-12-15:
That happens if a soft-reset of the HTU21D fails at startup.

Please try with a minimal config and all other devices off the i2c bus.

Limych, 2019-12-15:
I already made a version with automatic soft reset via the reset switch in case of a sensor “freezing”.  This did not help. 

I'll try to do it with a minimal config.

Limych, 2019-12-15:
Minimalistic config:
```yaml
substitutions:
  device_name: "Kitchen"
  <<: !include secrets.yaml

esphome:
  name: kitchen
  platform: ESP32
  board: nodemcu-32s

status_led:
  pin: GPIO2

wifi:
  ssid: $wifi_ssid
  password: !secret wifi_password
  domain: ".lan"

  manual_ip:
    static_ip: 192.168.1.65
    gateway: 192.168.1.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${device_name} Fallback Hotspot"
    password: !secret ota_password

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret ota_password

ota:
  password: !secret ota_password

web_server:
  port: 80
  auth:
    username: admin
    password: !secret web_server_password

i2c:
  - id: bus_indoor
    sda: 21
    scl: 22
    scan: True
  - id: bus_outdoor
    sda: 19
    scl: 23
    scan: True

sensor:
  # Outdoor sensors ####################################################

  - platform: htu21d
    i2c_id: bus_outdoor
    # address: 0x40
    update_interval: 60s
    temperature:
      name: "Outdoor Temperature"
      id: outdoor_temp
    humidity:
      name: "Outdoor Humidity"
      id: outdoor_humi
```

Logs: (same error :( )
```text
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.65 using esphome API
INFO Connecting to 192.168.1.65:6053 (192.168.1.65)
INFO Successfully connected to 192.168.1.65
[23:54:51][I][app:100]: ESPHome version 1.14.3 compiled on Dec 15 2019, 23:53:53
[23:54:51][C][status_led:019]: Status LED:
[23:54:51][C][status_led:020]:   Pin: GPIO2 (Mode: OUTPUT)
[23:54:51][C][wifi:415]: WiFi:
[23:54:51][C][wifi:283]:   SSID: [redacted]
[23:54:51][C][wifi:284]:   IP Address: 192.168.1.65
[23:54:51][C][wifi:286]:   BSSID: [redacted]
[23:54:51][C][wifi:287]:   Hostname: 'kitchen'
[23:54:51][C][wifi:291]:   Signal strength: -50 dB ▂▄▆█
[23:54:51][C][wifi:295]:   Channel: 9
[23:54:51][C][wifi:296]:   Subnet: 255.255.255.0
[23:54:51][C][wifi:297]:   Gateway: 192.168.1.1
[23:54:51][C][wifi:298]:   DNS1: 0.0.0.0
[23:54:51][C][wifi:299]:   DNS2: 0.0.0.0
[23:54:51][C][i2c:028]: I2C Bus:
[23:54:51][C][i2c:029]:   SDA Pin: GPIO21
[23:54:51][C][i2c:030]:   SCL Pin: GPIO22
[23:54:51][C][i2c:031]:   Frequency: 50000 Hz
[23:54:51][I][i2c:033]: Scanning i2c bus for active devices...
[23:54:51][I][i2c:040]: Found i2c device at address 0x40
[23:54:51][I][i2c:040]: Found i2c device at address 0x5A
[23:54:51][I][i2c:040]: Found i2c device at address 0x76
[23:54:51][C][i2c:028]: I2C Bus:
[23:54:51][C][i2c:029]:   SDA Pin: GPIO19
[23:54:51][C][i2c:030]:   SCL Pin: GPIO23
[23:54:51][C][i2c:031]:   Frequency: 50000 Hz
[23:54:51][I][i2c:033]: Scanning i2c bus for active devices...
[23:54:51][I][i2c:040]: Found i2c device at address 0x23
[23:54:51][I][i2c:040]: Found i2c device at address 0x40
[23:54:51][I][i2c:040]: Found i2c device at address 0x77
[23:54:51][C][logger:175]: Logger:
[23:54:51][C][logger:176]:   Level: DEBUG
[23:54:51][C][logger:177]:   Log Baud Rate: 115200
[23:54:51][C][logger:178]:   Hardware UART: UART0
[23:54:51][C][htu21d:027]: HTU21D:
[23:54:51][C][htu21d:028]:   Address: 0x40
[23:54:51][E][htu21d:030]: Communication with HTU21D failed!
[23:54:51][C][htu21d:032]:   Update Interval: 60.0s
[23:54:51][C][htu21d:033]:   Temperature 'Outdoor Temperature'
[23:54:51][C][htu21d:033]:     Unit of Measurement: '°C'
[23:54:51][C][htu21d:033]:     Accuracy Decimals: 1
[23:54:51][C][htu21d:033]:     Icon: 'mdi:thermometer'
[23:54:51][C][htu21d:034]:   Humidity 'Outdoor Humidity'
[23:54:51][C][htu21d:034]:     Unit of Measurement: '%'
[23:54:51][C][htu21d:034]:     Accuracy Decimals: 1
[23:54:51][C][htu21d:034]:     Icon: 'mdi:water-percent'
[23:54:51][C][captive_portal:169]: Captive Portal:
[23:54:51][C][web_server:122]: Web Server:
[23:54:51][C][web_server:123]:   Address: 192.168.1.65:80
[23:54:51][C][web_server:125]:   Basic authentication enabled
[23:54:51][C][ota:029]: Over-The-Air Updates:
[23:54:51][C][ota:030]:   Address: 192.168.1.65:3232
[23:54:51][C][ota:032]:   Using Password.
[23:54:51][C][api:095]: API Server:
[23:54:51][C][api:096]:   Address: 192.168.1.65:6053
```

Limych, 2019-12-16:
The error disappeared when I removed the second i2c bus from the configs. And it returned back when adding another sensor to the same (only in the config) bus. At the same time, when the config returned to the state "one bus + only htc21d", the error did not disappear again. It turns out that, in a strange way, it does not always depend on configs.

Limych, 2019-12-16:
My last (more minimalistic) config:
```yaml
substitutions:
  device_name: "Kitchen"
  <<: !include secrets.yaml

esphome:
  name: kitchen
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: $wifi_ssid
  password: !secret wifi_password
  domain: ".lan"

  manual_ip:
    static_ip: 192.168.1.65
    gateway: 192.168.1.1
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret ota_password

ota:
  password: !secret ota_password

i2c:
  - id: bus_outdoor
    sda: 19
    scl: 23
    scan: True

sensor:
  # Outdoor sensors ####################################################

  - platform: htu21d
    i2c_id: bus_outdoor
    # address: 0x40
    update_interval: 60s
    temperature:
      name: "Outdoor Temperature"
      id: outdoor_temp
    humidity:
      name: "Outdoor Humidity"
      id: outdoor_humi
```

And the same error in a logs:
```text
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.65 using esphome API
INFO Connecting to 192.168.1.65:6053 (192.168.1.65)
INFO Successfully connected to 192.168.1.65
[00:31:46][I][app:100]: ESPHome version 1.14.3 compiled on Dec 16 2019, 00:29:56
[00:31:46][C][wifi:415]: WiFi:
[00:31:46][C][wifi:283]:   SSID: [redacted]
[00:31:46][C][wifi:284]:   IP Address: 192.168.1.65
[00:31:46][C][wifi:286]:   BSSID: [redacted]
[00:31:46][C][wifi:287]:   Hostname: 'kitchen'
[00:31:46][C][wifi:291]:   Signal strength: -50 dB ▂▄▆█
[00:31:46][C][wifi:295]:   Channel: 9
[00:31:46][C][wifi:296]:   Subnet: 255.255.255.0
[00:31:46][C][wifi:297]:   Gateway: 192.168.1.1
[00:31:46][C][wifi:298]:   DNS1: 0.0.0.0
[00:31:46][C][wifi:299]:   DNS2: 0.0.0.0
[00:31:46][C][i2c:028]: I2C Bus:
[00:31:46][C][i2c:029]:   SDA Pin: GPIO19
[00:31:46][C][i2c:030]:   SCL Pin: GPIO23
[00:31:46][C][i2c:031]:   Frequency: 50000 Hz
[00:31:46][I][i2c:033]: Scanning i2c bus for active devices...
[00:31:46][I][i2c:040]: Found i2c device at address 0x23
[00:31:46][I][i2c:040]: Found i2c device at address 0x40
[00:31:46][I][i2c:040]: Found i2c device at address 0x77
[00:31:46][C][logger:175]: Logger:
[00:31:46][C][logger:176]:   Level: DEBUG
[00:31:46][C][logger:177]:   Log Baud Rate: 115200
[00:31:46][C][logger:178]:   Hardware UART: UART0
[00:31:46][C][htu21d:027]: HTU21D:
[00:31:46][C][htu21d:028]:   Address: 0x40
[00:31:46][E][htu21d:030]: Communication with HTU21D failed!
[00:31:46][C][htu21d:032]:   Update Interval: 60.0s
[00:31:46][C][htu21d:033]:   Temperature 'Outdoor Temperature'
[00:31:46][C][htu21d:033]:     Unit of Measurement: '°C'
[00:31:46][C][htu21d:033]:     Accuracy Decimals: 1
[00:31:46][C][htu21d:033]:     Icon: 'mdi:thermometer'
[00:31:46][C][htu21d:034]:   Humidity 'Outdoor Humidity'
[00:31:46][C][htu21d:034]:     Unit of Measurement: '%'
[00:31:46][C][htu21d:034]:     Accuracy Decimals: 1
[00:31:46][C][htu21d:034]:     Icon: 'mdi:water-percent'
[00:31:46][C][ota:029]: Over-The-Air Updates:
[00:31:46][C][ota:030]:   Address: 192.168.1.65:3232
[00:31:46][C][ota:032]:   Using Password.
[00:31:46][C][api:095]: API Server:
[00:31:46][C][api:096]:   Address: 192.168.1.65:6053
```

mkono87, 2020-02-15:
Im having this same problem with the same sensor. Its been working for weeks then just started doing this. Im suspecting the sensor has died??

closed 2020-12-05 by stale[bot]

freynder, 2021-07-07:
I had the same problem with this sensor. I set "scan: False" on the i2c interface and manually added the address:
```
platform: htu21d
id: htu_sensor
address: 0x40
```
which seems to work ok now.

prchal, 2021-07-30:
The same with 2 boards and sesors - both esp32. On 8266 works well.
scan: false, address: 0x40 didn't help.

chrismetcalf, 2021-08-03:
I'm getting this issue too. Brand new ESP32 and sensor. 0.19-dev.

chrismetcalf, 2021-08-03:
```================================================= [SUCCESS] Took 47.06 seconds =================================================
INFO Successfully uploaded program.
INFO Starting log output from /dev/cu.usbserial-0001 with baud rate 115200
[23:30:15][I][logger:170]: Log initialized
[23:30:15][I][app:029]: Running through setup()...
[23:30:15][C][htu21d:016]: Setting up HTU21D...
[23:30:15][W][i2c:070]: Received NACK on transmit of address 0x40
[23:30:15][E][component:092]: Component was marked as failed.
[23:30:15][C][wifi:033]: Setting up WiFi...
[23:30:15][D][wifi:324]: Starting scan...
[23:30:17][D][wifi:339]: Found networks:
... redacted ...
[23:30:17][I][wifi:194]: WiFi Connecting to '[redacted]'...
[23:30:18][I][wifi:457]: WiFi Connected!
[23:30:18][C][wifi:303]:   SSID: '[redacted]'
[23:30:18][C][wifi:304]:   IP Address: 192.168.1.160
[23:30:18][C][wifi:306]:   BSSID: [redacted]
[23:30:18][C][wifi:307]:   Hostname: 'temp-shed'
[23:30:18][C][wifi:311]:   Signal strength: -43 dB ▂▄▆█
[23:30:18][C][wifi:315]:   Channel: 1
[23:30:18][C][wifi:316]:   Subnet: 255.255.255.0
[23:30:18][C][wifi:317]:   Gateway: 192.168.1.1
[23:30:18][C][wifi:318]:   DNS1: 192.168.1.1
[23:30:18][C][wifi:319]:   DNS2: 8.8.8.8
[23:30:18][D][wifi:466]: Disabling AP...
[23:30:18][I][app:059]: setup() finished successfully!
[23:30:18][I][app:105]: ESPHome version 1.19.0-dev compiled on Aug  2 2021, 23:29:20
[23:30:18][C][wifi:443]: WiFi:
[23:30:18][C][wifi:303]:   SSID: '[redacted]'
[23:30:18][C][wifi:304]:   IP Address: 192.168.1.160
[23:30:18][C][wifi:306]:   BSSID: [redacted]
[23:30:18][C][wifi:307]:   Hostname: 'temp-shed'
[23:30:18][C][wifi:311]:   Signal strength: -43 dB ▂▄▆█
[23:30:18][C][wifi:315]:   Channel: 1
[23:30:18][C][wifi:316]:   Subnet: 255.255.255.0
[23:30:18][C][wifi:317]:   Gateway: 192.168.1.1
[23:30:18][C][wifi:318]:   DNS1: 192.168.1.1
[23:30:18][C][wifi:319]:   DNS2: 8.8.8.8
[23:30:18][C][i2c:028]: I2C Bus:
[23:30:18][C][i2c:029]:   SDA Pin: GPIO21
[23:30:18][C][i2c:030]:   SCL Pin: GPIO22
[23:30:18][C][i2c:031]:   Frequency: 50000 Hz
[23:30:18][C][logger:189]: Logger:
[23:30:18][C][logger:190]:   Level: DEBUG
[23:30:18][C][logger:191]:   Log Baud Rate: 115200
[23:30:18][C][logger:192]:   Hardware UART: UART0
[23:30:18][C][htu21d:027]: HTU21D:
[23:30:18][C][htu21d:028]:   Address: 0x40
[23:30:18][E][htu21d:030]: Communication with HTU21D failed!
[23:30:18][C][htu21d:032]:   Update Interval: 30.0s
[23:30:18][C][htu21d:033]:   Temperature 'Shed Temperature'
[23:30:18][C][htu21d:033]:     Device Class: 'temperature'
[23:30:18][C][htu21d:033]:     State Class: 'measurement'
[23:30:18][C][htu21d:033]:     Unit of Measurement: '°C'
[23:30:18][C][htu21d:033]:     Accuracy Decimals: 1
[23:30:18][C][htu21d:034]:   Humidity 'Shed Humidity'
[23:30:18][C][htu21d:034]:     Device Class: 'humidity'
[23:30:18][C][htu21d:034]:     State Class: 'measurement'
[23:30:18][C][htu21d:034]:     Unit of Measurement: '%'
[23:30:18][C][htu21d:034]:     Accuracy Decimals: 1
```

chrismetcalf, 2021-08-03:
Reset it and dump the logs and sometimes it works:
```
[23:33:44][C][i2c:028]: I2C Bus:
[23:33:44][C][i2c:029]:   SDA Pin: GPIO21
[23:33:44][C][i2c:030]:   SCL Pin: GPIO22
[23:33:44][C][i2c:031]:   Frequency: 50000 Hz
[23:33:44][C][logger:189]: Logger:
[23:33:44][C][logger:190]:   Level: DEBUG
[23:33:44][C][logger:191]:   Log Baud Rate: 115200
[23:33:44][C][logger:192]:   Hardware UART: UART0
[23:33:44][C][htu21d:027]: HTU21D:
[23:33:44][C][htu21d:028]:   Address: 0x40
[23:33:44][C][htu21d:032]:   Update Interval: 30.0s
[23:33:44][C][htu21d:033]:   Temperature 'Shed Temperature'
[23:33:44][C][htu21d:033]:     Device Class: 'temperature'
[23:33:44][C][htu21d:033]:     State Class: 'measurement'
[23:33:44][C][htu21d:033]:     Unit of Measurement: '°C'
[23:33:44][C][htu21d:033]:     Accuracy Decimals: 1
[23:33:44][C][htu21d:034]:   Humidity 'Shed Humidity'
[23:33:44][C][htu21d:034]:     Device Class: 'humidity'
[23:33:44][C][htu21d:034]:     State Class: 'measurement'
[23:33:44][C][htu21d:034]:     Unit of Measurement: '%'
[23:33:44][C][htu21d:034]:     Accuracy Decimals: 1
[23:34:08][D][htu21d:052]: Got Temperature=24.5°C Humidity=38.5%
[23:34:08][D][sensor:117]: 'Shed Temperature': Sending state 24.46117 °C with 1 decimals of accuracy
[23:34:08][D][sensor:117]: 'Shed Humidity': Sending state 38.54041 % with 1 decimals of accuracy
```

chrismetcalf, 2021-08-03:
Tried upgrading to 1.20.3 and I get the same error.

Tozzi89, 2021-10-30:
I have the same issue, everything is brand new, and the config is very minimal. I have tried some of the tips in this thread but nothing works. I have managed to get it working only once when I pressed reset button on the esp32.