esphome/issues #927: Globals Array does not work

manju-rn, 2019-12-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32, ESP8266
**Affected component:**

Globals

**Description of problem:**
Initializing a global array requires a definite value. Array should be able to defined when values are being provided

**Problem-relevant YAML-configuration entries:**
```yaml
globals:
  - id: global_remote_code
    type: int[]
    restore_value: no
    initial_value: "{1258, -442}"
```
Working code
```yaml
globals:
  - id: global_remote_code
    type: int[2]
    restore_value: no
    initial_value: "{1258, -442}"
```

**Logs (if applicable):**

```
In file included from src/esphome.h:17:0,
                 from src/main.cpp:3:
src/esphome/components/globals/globals_component.h: In instantiation of 'class esphome::globals::GlobalsComponent<int []>':
src/main.cpp:3287:34:   required from here
src/esphome/components/globals/globals_component.h:47:12: error: 'esphome::globals::GlobalsComponent<T>::value_' has incomplete type
   T value_{};
            ^
src/esphome/components/globals/globals_component.h:47:12: error: invalid use of array with unspecified bounds
src/esphome/components/globals/globals_component.h:48:17: error: 'esphome::globals::GlobalsComponent<T>::prev_value_' has incomplete type
   T prev_value_{};
                 ^
src/esphome/components/globals/globals_component.h:48:17: error: invalid use of array with unspecified bounds
src/main.cpp: In function 'void setup()':
src/main.cpp:3351:73: error: no matching function for call to 'esphome::globals::GlobalsComponent<int []>::GlobalsComponent(<brace-enclosed initializer list>)'
   global_remote_code = new globals::GlobalsComponent<int[]>({1258, -442});
                                                                         ^
src/main.cpp:3351:73: note: candidates are:
In file included from src/esphome.h:17:0,
                 from src/main.cpp:3:
src/esphome/components/globals/globals_component.h:15:12: note: esphome::globals::GlobalsComponent<T>::GlobalsComponent(std::array<typename std::remove_extent<_Up>::type, std::extent<_Tp>::value>) [with T = int []; typename std::remove_extent<_Up>::type = int]
   explicit GlobalsComponent(std::array<typename std::remove_extent<T>::type, std::extent<T>::value> initial_value) {
            ^
src/esphome/components/globals/globals_component.h:15:12: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'std::array<int, 0u>'
src/esphome/components/globals/globals_component.h:14:12: note: esphome::globals::GlobalsComponent<T>::GlobalsComponent(T) [with T = int []]
   explicit GlobalsComponent(T initial_value) : value_(initial_value) {}
            ^
src/esphome/components/globals/globals_component.h:14:12: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'int*'
src/esphome/components/globals/globals_component.h:13:12: note: esphome::globals::GlobalsComponent<T>::GlobalsComponent() [with T = int []]
   explicit GlobalsComponent() = default;
            ^
src/esphome/components/globals/globals_component.h:13:12: note:   candidate expects 0 arguments, 1 provided
src/esphome/components/globals/globals_component.h:10:28: note: constexpr esphome::globals::GlobalsComponent<int []>::GlobalsComponent(const esphome::globals::GlobalsComponent<int []>&)
 template<typename T> class GlobalsComponent : public Component {
                            ^
src/esphome/components/globals/globals_component.h:10:28: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'const esphome::globals::GlobalsComponent<int []>&'
src/esphome/components/globals/globals_component.h:10:28: note: constexpr esphome::globals::GlobalsComponent<int []>::GlobalsComponent(esphome::globals::GlobalsComponent<int []>&&)
src/esphome/components/globals/globals_component.h:10:28: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'esphome::globals::GlobalsComponent<int []>&&'
src/esphome/components/globals/globals_component.h: In instantiation of 'T& esphome::globals::GlobalsComponent<T>::value() [with T = int []]':
src/main.cpp:3287:42:   required from here
src/esphome/components/globals/globals_component.h:19:29: error: 'class esphome::globals::GlobalsComponent<int []>' has no member named 'value_'
   T &value() { return this->value_; }
                             ^
src/esphome/components/globals/globals_component.h: In member function 'T& esphome::globals::GlobalsComponent<T>::value() [with T = int []]':
src/esphome/components/globals/globals_component.h:19:37: warning: control reaches end of non-void function [-Wreturn-type]
   T &value() { return this->value_; }
                                     ^
*** [/data/airconcontrol8266/.pioenvs/airconcontrol8266/src/main.cpp.o] Error 1

```

**Additional information and things you've tried:**

brandond, 2019-12-15:
I'm not sure I understand the issue. Do you expect it to work without specifying the size, or infer it from the initial value?

manju-rn, 2019-12-15:
infer it form the initial value.  I was expecting to initialize in a typical way int i[] = "{1258, -442}";

OttoWinter, 2019-12-15:
No, that is not possible.

The global value is enclosed in an object that handles initialization, recovering values from boot etc. The size of the type needs to be known for that.

There's no good way in C++11 to recover the amount of arguments in a braced initializer list (or at least that I could find).

closed 2019-12-15 by OttoWinter

brandond, 2019-12-16:
You could probably get cute with the Python code and parse the initializer to fill in the array size, but that seems like it might be easy to break. Easier to just treat it like normal C++ and specify the array size manually.