esphome/issues #929: Documentation: Esp-01 deep sleep not possible without some soldering

surak, 2019-12-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

Esp-01
**Affected component:**

deep_sleep

**Description of problem:**

As the Esp-01 is basically an esp8266 stripped down, documentation of those two are basically the same. It must be mentioned, however, that an Esp-01 will *NOT* work with the deep sleep module. It will sleep alright, but it won't wake up. The only way of doing so is with soldering a wire from the cpu to a pin, as here: (https://www.instructables.com/id/Enable-DeepSleep-on-an-ESP8266-01/)

I spent quite some time trying until I found this link and realized it's not possible.

glmnet, 2019-12-17:
This is a board limitation, the connection requirement is documented https://esphome.io/components/deep_sleep.html

> note that on the ESP8266 GPIO16 must be connected to the RST pin so that it will wake up again.

Those esp-01 boards are really limited by io pins

I don't know if adding a special mention to this board is needed, if not this issue should be closed.

surak, 2019-12-17:
> I don't know if adding a special mention to this board is needed,

That's really the issue, since it's not clear at all. I would add a couple words about it.

OttoWinter, 2019-12-17:
As @glmnet said, it is documented.

That's also exactly what the guide you posted does: connect the RST pin to GPIO16.

As you noted, the ESP-01 is just a package for the ESP8266 (ESP8266 is the core chip, ESP-01 is the full module with flash etc.).

surak, 2019-12-17:
It is there if you know what it means. It is not clear for first-time users at all. But it's your project, sorry.

OttoWinter, 2019-12-17:
I agree it's not highlighted enough (making it a warning tag would make it stand out more).

But it's in fact documented.

I'm not sure I understand what the cause is here (please comment which):
- The comment is too small and you didn't read it.
- You didn't understand the comment because you thought ESP-01 is not ESP8266.
- Other

surak, 2019-12-17:
ESP8266 is the SOC, correct? ESP-01 is the whole shebang board, if I understand correctly. 

Given such a board has no easy way to do anything like that (as there are no exposed gpio pins), I would never suspect that it would be basically impossible to do it without amazing soldering skills.

closed 2019-12-17 by OttoWinter

glmnet, 2019-12-17:
In an effort to please everybody and don't clutter the docs, maybe ESPHome could throw an error if board esp-01 is chosen with deep sleep.

If somebody solders a wire, then he can change the board type and be on it's own