esphome/issues #933: Sonoff 4CH Pro external Buttons

alexmc1510, 2019-12-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 Board Sonoff 4CH Pro
**Affected component:**

**Description of problem:**
If I confugure GPIO0 and GPIO10 as PULLUP INPUT for external push buttons, behavior of the module is different from expected.
GPIO0 - Sonoff 4CH Pro does not start (it seems it remains in flash mode).
GPIO10 - Sonoff 4CH Pro resets.
The aim of the configuration is to open and close 1 cover so I need 2 external push buttons.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: dormitorio_principal
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: ""
  password: ""

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ""
    password: ""

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

binary_sensor:
- platform: gpio
  pin:
    number: GPIO0
    mode: INPUT_PULLUP
    inverted: true
  id: pulsador_subir_persiana_dormitorio_principal
  on_press:
    then:
      # logic for cycling through movements: open->stop->close->stop->...
      - lambda: |
          if (id(persiana_dormitorio_principal).current_operation == COVER_OPERATION_IDLE) {
            // Cover is idle, check current state and either open or close cover.
            id(persiana_dormitorio_principal).open();
          } else {
            // Cover is opening/closing. Stop it.
            id(persiana_dormitorio_principal).stop();
          }

- platform: gpio
  pin:
    number: GPIO10
    mode: INPUT_PULLUP
    inverted: true
  id: pulsador_bajar_persiana_dormitorio_principal
  on_press:
    then:
      # logic for cycling through movements: open->stop->close->stop->...
      - lambda: |
          if (id(persiana_dormitorio_principal).current_operation == COVER_OPERATION_IDLE) {
            // Cover is idle, check current state and either open or close cover.
            id(persiana_dormitorio_principal).close();
          } else {
            // Cover is opening/closing. Stop it.
            id(persiana_dormitorio_principal).stop();
          }
- platform: status
  name: "Status Dormitorio Principal"

switch:
  - platform: gpio
    pin: 
      number: GPIO12
    id: subir_persiana_dormitorio_principal
    interlock: [bajar_persiana_dormitorio_principal]
  - platform: gpio
    pin: 
      number: GPIO5
    id: bajar_persiana_dormitorio_principal
    interlock: [subir_persiana_dormitorio_principal]
  - platform: gpio
    pin: 
      number: GPIO4
    id: radiador_dormitorio_principal
  - platform: gpio
    pin: 
      number: GPIO15
    id: luz_dormitorio_principal

cover:
  - platform: time_based
    name: "Persiana Dormitorio Principal"
    id: persiana_dormitorio_principal
    open_action:
      - switch.turn_on: subir_persiana_dormitorio_principal
    open_duration: 15s
    close_action:
      - switch.turn_on: bajar_persiana_dormitorio_principal
    close_duration: 15s
    stop_action:
      - switch.turn_off: subir_persiana_dormitorio_principal
      - switch.turn_off: bajar_persiana_dormitorio_principal
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
I have tried GPIO9 and GPIO14 with success.

OttoWinter, 2019-12-17:
> GPIO0 - Sonoff 4CH Pro does not start (it seems it remains in flash mode).

Flash mode is decided before the user code even runs. GPIO0 works fine on all ESP8266 modules I have.

Please
1. make sure it's really a Sonoff 4CH Pro (and not some other board)
2. Try with a minimal config (just a binary sensor the pin you want to test, no cover etc)

ng-galien, 2019-12-19:
Hi @alexmc1510 
The 4CH pro have special modes for inputs which are not connected directly to the ESP.
There are some hardwired functions like delay, toogle, etc. 
You can configure it with 3 DIP switchs.

Let's try the config here, it works with Tasmota firmware. I have a 4CH pro iworking at home, if you struggle I can check its DIP config and the YAML. 

https://github.com/arendst/Tasmota/wiki/sonoff-4ch-pro

alexmc1510, 2019-12-19:
Hello,

Thanks for your support. I have modified my code with the GPIO's indicated by @ng-galien and implemented it in the following devices, with the following results:

**SONOFF DUAL R2**
[image]
With config:
#gpio_in_0: GPIO0
#gpio_in_1: GPIO9
Result: Works perfectly.

**SONOFF 4CH R2**
[image]
With config:
#gpio_in_0: GPIO0
#gpio_in_1: GPIO9
#gpio_in_2: GPIO10
#gpio_in_3: GPIO14
#gpio_out_0: GPIO12
#gpio_out_1: GPIO5
#gpio_out_2: GPIO4
#gpio_out_3: GPIO15
Result: Works perfectly.

**SONOFF 4CH PRO**
[image]

With config:
#gpio_in_0: GPIO0
#gpio_in_1: GPIO9
#gpio_in_2: GPIO10
#gpio_in_3: GPIO14
#gpio_out_0: GPIO12
#gpio_out_1: GPIO5
#gpio_out_2: GPIO4
#gpio_out_3: GPIO15
And DIP config:
S6: 1
K5: all 1
K6: all 0

Result: Sometimes button push is not recognized. It seems like a type of freezing but at least, reset has dissapeared. I will continue testing it if you consider necessary but from my point of view it's not a problem in the code because same code in other devices works like a charm.

Thank you.

ng-galien, 2019-12-19:
@alexmc1510 
Hi Alex, great if that helps.
I install this device few months ago and I remenber that buttons "does not works" always as expected. I remenber that rely exclusively with wrong DIP settings.
It's not easy to take of the device for the moment, so I'm not able to check the DIP positions, and I didn't write it anywhere... 

Have a look here for DIP config
https://www.itead.cc/wiki/Sonoff_4CH_Pro

```
And DIP config:
S6: 1
K5: all 1
K6: all 0
```

As I remenber I struggle a little bit in this part, because we don't relay know how the original firmware handle it. 
I had tried the 2^3 configs to find the good one with a simple toggle test, keep in mind to turn off the device before any changes ;)

closed 2020-08-07 by stale[bot]