esphome/issues #936: "Auth Expired" when compiled on windows

beikeland, 2019-12-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Does not work:
$ uname -a
MINGW64_NT-10.0 DESKTOP-OV51DUS 2.8.2(0.313/5/3) 2017-07-12 15:35 x86_64 Msys
$ esphome version
Version: 1.14.3

Does work:
$ uname -a
Linux rama 4.15.0-66-generic #75-Ubuntu SMP Tue Oct 1 05:24:09 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
e$ esphome version
WARNING You're using ESPHome with python 2. Support for python 2 is deprecated and will be removed in 1.15.0. Please reinstall ESPHome with python 3.6 or higher.
Version: 1.14.3

**ESP (ESP32/ESP8266, Board/Sonoff):**

Blitzwolf BW-SHP6 (ESP8285)

**Affected component:**

wifi?

**Description of problem:**
After flashing new firmware from fresh windows install device lost wifi connection. Flashing with exactly the same config from linux, using same esphome version, for some reason works.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ${device_name}
  platform: ESP8266
  board: esp8285

wifi:
  ssid: !secret iot_ssid
  password: !secret iot_password
  domain: !secret iot_domain 
  ap:
    ssid: "ESPhome"

logger:

api:
  reboot_timeout: 0s

ota:

time:
  - platform: homeassistant
    id: homeassistant_time
   
binary_sensor:
  - platform: gpio
    id: button
    internal: true
    pin:
      number: GPIO13
    on_press:
      - switch.toggle: relay

output:
  - platform: gpio
    pin: GPIO00
    inverted: true
    id: led

switch:
  - platform: gpio
    pin: GPIO15
    id: relay
    name: '${device_name} Relay'
    on_turn_on:
      - output.turn_on: led
    on_turn_off:
      - output.turn_off: led

status_led:
  pin:
    number: GPIO02
    inverted: true

sensor:
  - platform: uptime
    name: '${device_name} Uptime'
    unit_of_measurement: days
    update_interval: 300s
    filters:
      - multiply: 0.000011574
    internal: true
  - platform: wifi_signal
    name: '${device_name} WiFi Signal'
    update_interval: 300s
    internal: true
  - platform: hlw8012
    sel_pin:
      number: GPIO12
      inverted: true
    cf_pin: GPIO05
    cf1_pin: GPIO14
    current_resistor: ${current_res}
    voltage_divider: ${voltage_div}
    current:
      name: '${device_name} Current'
      unit_of_measurement: 'A'
      accuracy_decimals: 3
      icon: mdi:flash-outline
      internal: true
    voltage:
      name: '${device_name} Voltage'
      unit_of_measurement: 'V'
      icon: mdi:flash-outline
      internal: true
    power:
      name: '${device_name} Power'
      unit_of_measurement: 'W'
      icon: mdi:flash-outline
      internal: true
    change_mode_every: 4
    update_interval: 10s
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
[I][logger:156]: Log initialized
[C][status_led:014]: Setting up Status LED...
[C][ota:364]: There have been 0 suspected unsuccessful boot attempts.
[I][app:028]: Running through setup()...
[D][binary_sensor:034]: 'button': Sending initial state ON
[C][switch.gpio:011]: Setting up GPIO Switch 'blitzwolf04 Relay'...
[D][switch:025]: 'blitzwolf04 Relay' Turning OFF.
[D][switch:045]: 'blitzwolf04 Relay': Sending state OFF
[D][switch:025]: 'blitzwolf04 Relay' Turning OFF.
[C][hlw8012:013]: Setting up HLW8012...
[C][wifi:033]: Setting up WiFi...
[D][wifi:304]: Starting scan...
[D][sensor:092]: 'blitzwolf04 Uptime': Sending state 0.00000 days with 0 decimals of accuracy
[D][wifi:319]: Found networks:
[I][wifi:365]: - 'iotssid' (Dx:xx:xx:xx:xx:x4) ▂▄▆█
[D][wifi:366]:     Channel: 13
[D][wifi:367]:     RSSI: -62 dB
[D][wifi:369]: - 'EAGLE300EE7' (34:21:09:30:0E:E7) ▂▄▆█
[D][wifi:369]: - 'Telenor7533sei' (44:F0:34:ED:61:74) ▂▄▆█
[D][wifi:369]: - 'Mi-Fi' (44:F0:34:F1:09:D4) ▂▄▆█
[I][wifi:193]: WiFi Connecting to 'iotssid'...
[W][wifi_esp8266:337]: Event: Disconnected ssid='iotssid' bssid=Dx:xx:xx:xx:xx:x4 reason='Auth Expired'
[W][wifi:451]: Error while connecting to network.
[W][wifi:488]: Restarting WiFi adapter...
[W][wifi_esp8266:337]: Event: Disconnected ssid='iotssid' bssid=Dx:xx:xx:xx:xx:x4 reason='Authentication Failed'
[I][wifi:193]: WiFi Connecting to 'iotssid'...
[D][sensor:092]: 'blitzwolf04 Power': Sending state 0.00000 W with 1 decimals of accuracy
[W][wifi_esp8266:337]: Event: Disconnected ssid='iotssid' bssid=Dx:xx:xx:xx:xx:x4 reason='Auth Expired'
[W][wifi:451]: Error while connecting to network.
[W][wifi:488]: Restarting WiFi adapter...
[W][wifi_esp8266:337]: Event: Disconnected ssid='iotssid' bssid=Dx:xx:xx:xx:xx:x4 reason='Authentication Failed'
[D][wifi:304]: Starting scan...

```

**Additional information and things you've tried:**
have tried `esptool.py --port COM14 erase_flash` with no change.

similar is refrenced in #409 but downgrading to older esphome version failed, and in any case same version produce working binaries on linux.

removed logger and stripped config to basics with no change in outcome.

tneish, 2020-02-02:
Same issue on windows.
I got it working on Linux only after updating the build environment (see below).

Run
`python -m platformio platform update`
to update.

The first 'platform' below worked, the second (older) didn't.

`python -m platformio platform update --dry-run`

> Platform Espressif 8266
> --------
> Checking espressif8266                   @ 2.3.2          [Up-to-date]
> Checking toolchain-xtensa                @ 2.40802.191122 [Up-to-date]
> Checking framework-arduinoespressif8266  @ 2.20603.191216 [Up-to-date]
> Checking tool-esptool                    @ 1.413.0        [Up-to-date]
> Checking tool-esptoolpy                  @ 1.20800.0      [Up-to-date]
> 
> Platform Espressif 8266
> --------
> Checking espressif8266                   @ 2.2.3          [Detached]
> Checking toolchain-xtensa                @ 2.40802.191122 [Up-to-date]
> Checking framework-arduinoespressif8266  @ 2.20502.0      [Up-to-date]
> Checking tool-esptool                    @ 1.413.0        [Up-to-date]
> Checking tool-esptoolpy                  @ 1.20600.0      [Up-to-date]`
>

Electronlibre2012, 2020-02-16:
Hello,

same problem...

The only way to solved is to reboot the internet Box and my Wifi AP...

But other sensor dont start after that...

I dont understand anything...all my sensor have a static IP (x35 sensors).

Have you found a solution @beikeland  and @tneish please?

tneish, 2020-02-16:
For reference I'm using Linux with:
Python 3.7.2
Created a virtual environment with "python3.7 -m venv"
Once inside the venv ("source bin/activate")..
Update pip ("python -m pip install --upgrade pip")
Install esphome ("pip install esphome")

Some special options in the esphome yaml that may or may not make a difference..

> esphome:
>   name: room1
>   platform: ESP8266
>   board: esp01_1m
>   platformio_options:
>     upload_speed: 57600
>     platform: espressif8266@2.3.2
> 
> wifi:
>   ssid: "..."
>   password: "..."
>   fast_connect: true
>   manual_ip:
>     static_ip: 10.0.4.203
>     gateway: 10.0.4.1
>     subnet: 255.255.255.0

Then compile and flash ("esphome room1.yaml run")

Flashed using serial, then flashed OTA while serial was still connected to make sure wifi OTA was working.

In my experience I suspect there can be bad combinations of esphome, platformio, esp libs.. but I have not looked further into it.

Also I am using Home Assistant and switched to adding the esphome devices by IP address instead of mDNS (e.g. "room1.local") as they appear faster after powering on. There is still a slight delay though. You can ping the IP address or run ("esphome room1.yaml logs") to see if it is up before it shows in Home assistant.

Electronlibre2012, 2020-02-17:
@tneish 

Thanks for the time you bring to reply!

What a complete reply!

I am under HassOs on Rpi4 and i cant apply your solution, but i found another one, more simple way...i have a internet box and a AP, i am used to pair my sensor on the AP (Asus R8000) and i have plenty of gateway and sensor (50) so i pair the new sensor on the box this time and it work directly...

The AP is in "mix mode", there is only a SSID for one 2.4Ghz and two 5Ghz band...

The box is in "separate mode", there is one SSID for 2.4Ghz and one for 5Ghz.

Perhaps i have too much IOT on the AP...its strange because it happens just with 1 Weimos mini Pro and 2 Weimos D1 mini...i have a lot of deconnexions still 3 weeks, sometimes all works, sometimes one dont want to connect anymore, not always the same. So i decided to migrate the 3 "faulty" on the internet box SSID, and now, all works like a charm, together.

Thanks anyway for your help!

martin3000, 2020-07-12:
platformio update

h0jeZvgoxFepBQ2C, 2020-08-15:
I have this issue too on linux :(

duchenpaul, 2020-10-04:
> esp01_1m

Thank you, your config works for me:
For the future people:
Add below as a walkarround, I swear to god I will never upgrade again.
```
  platformio_options:
    platform: espressif8266@2.3.2
```

```yaml
esphome:
  name: $device_name
  platform: ESP8266
  platformio_options:
    platform: espressif8266@2.3.2
  board: $board_model
  build_path: build/$device_name
  # esp8266_restore_from_flash: yes
  on_boot:
    priority: 800.0
    # ...
    then:
      - light.turn_off: rgbw_led
      - display.page.show: $main_page 

substitutions:
  board_model: d1_mini
  device_name: oled_display
  wifi_ssid: !secret wifi_ssid
  wifi_password: !secret wifi_password
  wifi_fast_connect: 'true'
  wifi_reboot_timeout: 120s
  ota_password: !secret ota_password
  api_reboot_timeout: 120s
  main_page: time_page
```

polclota, 2020-12-15:
In my case playing with platform: espressif8266@ did not solve the problem, whereas playing with arduino version did. It is important to mention that all my devices are working fine using default config but only one type of them fail, precisely those ones that use more resources. The higest working verison has been 2.5.2.

More info here:
https://esphome.io/components/esphome.html#esphome-arduino-version

esphome:
  name: $devicename
  platform: $platform
  board: $board
  arduino_version: **2.5.2**

K3iVi, 2021-03-19:
I had the same problem and I was very angry. Only thing that helped was to disable 5ghz wifi on my ubiquiti AP.

andrewerrington, 2021-04-05:
I just started getting into ESPhome. I have the exact same problem.

I am compiling on Windows 10.

Attached to my ESP8266 Wemos D1 Mini Pro are a pushbutton, a PIR, a WS8212 RGB LED, a buzzer, an Si7021 temp/humidity sensor and an SSD1327 1.5" square OLED display. The temperature sensor and the OLED are connected by I2C. Everything works! Thanks to all the devs!

However, when the SSD1327 OLED display is attached then wifi won't work. The OLED display works, but wifi doesn't. If I disconnect the display and reset the ESP8266 (display code is still present, but it fails because the device is not present) then I can connect to wifi.

I have tried two different power supplies. Different distances to the access point. Powering the OLED from 3.3V or from 5V. And two different D1 Mini Pros (an early one from 2018 and a newer one from last year). I think it connected once, but it is rare. It always connects if the OLED is not connected.

Happy to post error messages, but it seems that this is a common problem on ESP8266 and ESP32. I have plenty of other ESP8266 devices running ESPeasy, but no display. They all work.

andrewerrington, 2021-04-05:
Ok! It connected, and I got a log! Including some crashes. I've edited the AP names and MAC addresses. The AP I am connecting to is "MYSSID"
`
curacy
[W][wifi:485]: Error while connecting to network.
[W][wifi:522]: Restarting WiFi adapter...
[D][wifi:324]: Starting scan...
[D][binary_sensor:036]: 'LDK PIR': Sending state ON
[D][adc:056]: 'LDK Brightness': Got voltage=0.73V
[D][sensor:092]: 'LDK Brightness': Sending state 0.73438 V with 2 decimals of accuracy
[D][binary_sensor:036]: 'LDK PIR': Sending state OFF
[D][wifi:339]: Found networks:
[I][wifi:385]: - 'MYSSID' (DE:AD:BE:EF:00:00) ▂▄▆█
[D][wifi:386]:     Channel: 5
[D][wifi:387]:     RSSI: -63 dB
[D][wifi:389]: - 'ANOTHERSSID' (12:34:56:78:9A:BC) ▂▄▆█
[I][wifi:194]: WiFi Connecting to 'MYSSID'...
[I][wifi:116]: Starting fallback AP!
[C][wifi:161]: Setting up AP...
[C][wifi:163]:   AP SSID: 'Ldk Fallback Hotspot'
[C][wifi:164]:   AP Password: 'UXRUd7W8VcqM'
[C][wifi:173]:   IP Address: 192.168.4.1
[W][wifi_esp8266:444]: Event: Disconnected ssid='MYSSID' bssid=DE:AD:BE:EF:00:00 reason='Auth Expired'
[W][wifi:485]: Error while connecting to network.
[I][wifi:194]: WiFi Connecting to 'MYSSID'...
[W][wifi_esp8266:444]: Event: Disconnected ssid='MYSSID' bssid=DE:AD:BE:EF:00:00 reason='Auth Expired'
[W][wifi:485]: Error while connecting to network.
[D][adc:056]: 'LDK Brightness': Got voltage=0.74V
[D][sensor:092]: 'LDK Brightness': Sending state 0.73633 V with 2 decimals of accuracy
[D][wifi:324]: Starting scan...
[D][binary_sensor:036]: 'LDK PIR': Sending state ON
[D][binary_sensor:036]: 'LDK PIR': Sending state OFF
[D][adc:056]: 'LDK Brightness': Got voltage=0.74V
[D][sensor:092]: 'LDK Brightness': Sending state 0.73633 V with 2 decimals of accuracy
[D][wifi:339]: Found networks:
[I][wifi:385]: - 'MYSSID' (DE:AD:BE:EF:00:00) ▂▄▆█
[D][wifi:386]:     Channel: 5
[D][wifi:387]:     RSSI: -63 dB
[D][wifi:389]: - 'ANOTHERSSID' (12:34:56:78:9A:BC) ▂▄▆█
[D][wifi:389]: - 'FOURTHSSID' (00:04:00:04:00:04) ▂▄▆█
[I][wifi:194]: WiFi Connecting to 'MYSSID'...
[D][binary_sensor:036]: 'LDK PIR': Sending state ON
[W][wifi_esp8266:444]: Event: Disconnected ssid='MYSSID' bssid=DE:AD:BE:EF:00:00 reason='Auth Expired'
[W][wifi:485]: Error while connecting to network.
[I][wifi:194]: WiFi Connecting to 'MYSSID'...
[D][binary_sensor:036]: 'LDK PIR': Sending state OFF
[W][wifi_esp8266:444]: Event: Disconnected ssid='MYSSID' bssid=DE:AD:BE:EF:00:00 reason='Auth Expired'
[W][wifi:485]: Error while connecting to network.
[D][adc:056]: 'LDK Brightness': Got voltage=0.74V
[D][sensor:092]: 'LDK Brightness': Sending state 0.73926 V with 2 decimals of accuracy
[D][wifi:324]: Starting scan...
[D][wifi:339]: Found networks:
[I][wifi:385]: - 'MYSSID' (DE:AD:BE:EF:00:00) ▂▄▆█
[D][wifi:386]:     Channel: 5
[D][wifi:387]:     RSSI: -63 dB
[D][wifi:389]: - 'ANOTHERSSID' (12:34:56:78:9A:BC) ▂▄▆█
[D][wifi:389]: - 'THIRDSSID' (00:03:00:03:00:03) ▂▄▆█
[I][wifi:194]: WiFi Connecting to 'MYSSID'...
[W][wifi_esp8266:444]: Event: Disconnected ssid='MYSSID' bssid=DE:AD:BE:EF:00:00 reason='Auth Expired'
[W][wifi:485]: Error while connecting to network.
[I][wifi:194]: WiFi Connecting to 'MYSSID'...
[D][adc:056]: 'LDK Brightness': Got voltage=0.74V
[D][sensor:092]: 'LDK Brightness': Sending state 0.73926 V with 2 decimals of accuracy
[D][binary_sensor:036]: 'LDK PIR': Sending state ON
[I][wifi:457]: WiFi Connected!
[C][wifi:303]:   SSID: 'MYSSID'
[C][wifi:304]:   IP Address: 192.168.1.30
[C][wifi:306]:   BSSID: DE:AD:BE:EF:00:00
[C][wifi:307]:   Hostname: 'ldk'
[C][wifi:311]:   Signal strength: -72 dB ▂▄▆█
[C][wifi:315]:   Channel: 5
[... 1716 lines cut ...]
[D][sensor:092]: 'LDK Brightness': Sending state 0.73242 V with 2 decimals of accuracy
[D][binary_sensor:036]: 'LDK PIR': Sending state ON
[D][binary_sensor:036]: 'LDK PIR': Sending state OFF
[D][adc:056]: 'LDK Brightness': Got voltage=0.73V
[D][sensor:092]: 'LDK Brightness': Sending state 0.73242 V with 2 decimals of accuracy
[D][adc:056]: 'LDK Brightness': Got voltage=0.73V
[D][sensor:092]: 'LDK Brightness': Sending state 0.73047 V with 2 decimals of accuracy
[D][adc:056]: 'LDK Brightness': Got voltage=0.74V
[D][sensor:092]: 'LDK Brightness': Sending state 0.73535 V with 2 decimals of accuracy
[D][adc:056]: 'LDK Brightness': Got voltage=0.74V
[D][sensor:092]: 'LDK Brightness': Sending state 0.73633 V with 2 decimals of accuracy
[D][adc:056]: 'LDK Brightness': Got voltage=0.73V
[D][sensor:092]: 'LDK Brightness': Sending state 0.73242 V with 2 decimals of accuracy
[D][light:265]: 'LDK RGB Light' Setting:
[D][light:274]:   State: ON
[D][light:304]:   Transition Length: 1.0s
[D][light:265]: 'LDK RGB Light' Setting:
[D][light:274]:   State: OFF
[D][light:304]:   Transition Length: 1.0s
[D][htu21d:052]: Got Temperature=25.2°C Humidity=44.7%
[D][sensor:092]: 'LDK Temperature': Sending state 25.15831 °C with 1 decimals of accuracy
[D][sensor:092]: 'LDK Humidity': Sending state 44.72784 % with 1 decimals of accuracy
[D][adc:056]: 'LDK Brightness': Got voltage=0.74V
[D][sensor:092]: 'LDK Brightness': Sending state 0.73535 V with 2 decimals of accuracy
`

andrewerrington, 2021-04-05:
It dropped out again and fails to reconnect. :(

duchenpaul, 2021-04-05:
From the log, l see memory error, you can try  to start with very basic functions like connect WiFi only, to see it it just bcos lack of memory/flash

polclota, 2021-04-05:
I solved similar issues I had by setting the display component the last and wifi the first in the yaml file.

andrewerrington, 2021-04-06:
Thanks for your comments. Moving the display component last has no effect. The best and most obvious effect is to connect and disconnect the display (SSD1327 1.5" OLED on I2C). I will try compiling with options one by one and report back.

mmakaay, 2021-05-23:
While working on getting two SS1306 displays working through a multiplexer on an ESP8266 (D1 Mini), I ran into the same issue. Without the displays configured in the YAML, it worked. With one display it worked. With two displays, I got auth expired messages for the WiFi connection.

The fix for my setup was to increase the I2C frequency, to speed up the transfer of data to the displays. It looks like the transfer was blocking longer than the WiFi code was happy with.

```yaml
i2c:
  frequency: 400kHz
```

The 400 kHz was enough for getting two displays work. You can go up to 800 kHz if that isn't fast enough.

Maybe this solution can help you too.

andrewerrington, 2021-05-23:
I did try that, but the problem persisted. I'll try it again, however, it
shouldn't happen and is probably a bug.

polclota, 2021-08-09:
Setting i2c frequency to 400kHz worked for me. Before knowing about this great tip I had already solved the issue by reducing de quantity of pixels displayed on my SSD1306 128x64 screen while trying to connect to wifi but increasing the frequency allowed me to go back to initial pixel density. I guess it has to do with how busy is the bus in a certain moment and speeding up the process clearly solves it. BTW, forget about my previous post on changing versions as with this solution I can use whatever version I please.

kylegordon, 2021-08-29:
I had similar and removing the status_led component cured my wifi connection issue.

I repeated the process several times... very strange, but effective.

GsynthChris, 2022-02-21:
Hello, I ran into this problem this morning and couldn't find much online about it. To be honest I rolled my eyes when I saw the 400khz frequency suggestion but I'll be damned, that fixed it! LOL

Thanks for the strange yet effective fix!

```
i2c:
  frequency: 400kHz
```