esphome/issues #938: Installation failed - no such image: esphome/esphome-hassio-amd64:1.14.3

windpoint, 2019-12-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

PVE Manager Version pve-manager/6.1-5/9bf06119
Home Assistant 0.103.2
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Affected component:**

**Description of problem:**
Trying to install esphome as hassio addon. Installation failed. Please see system log section for details.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

19-12-20 20:25:32 INFO (SyncWorker_2) [hassio.docker.interface] Pull image esphome/esphome-hassio-amd64 tag 1.14.3.

19-12-20 20:26:45 ERROR (SyncWorker_2) [hassio.docker.interface] Can't install esphome/esphome-hassio-amd64:1.14.3 -> 404 Client Error: Not Found ("no such image: esphome/esphome-hassio-amd64:1.14.3: No such image: esphome/esphome-hassio-amd64:1.14.3").
```

**Additional information and things you've tried:**

bpenglase, 2020-01-05:
I'm getting the same thing. HA 0.103.4, Supervisor 193, HassOS 3.7.
All 3 esphome selections fail (dev/beta/main).

niro1987, 2020-04-27:
same thing. HA 0.108.9, Supervisor 219, HassOS 3.13. Installed on Hyper-V.

danuals, 2020-04-29:
same for pi lassos set up
20-04-29 20:36:05 ERROR (SyncWorker_16) [supervisor.docker.interface] Can't install esphome/esphome-hassio-armv7:1.14.3 -> 404 Client Error: Not Found ("no such image: esphome/esphome-hassio-armv7:1.14.3: No such image: esphome/esphome-hassio-armv7:1.14.3").

SmartHomeFanatic, 2020-06-14:
Any news on that one? I'm getting ERROR (SyncWorker_5) [supervisor.docker.interface] Can't install esphome/esphome-hassio-armv7:1.14.4 -> 404 Client Error: Not Found ("no such image: esphome/esphome-hassio-armv7:1.14.4: No such image: esphome/esphome-hassio-armv7:1.14.4").

Running HassOS 4.10 with Home Assistant 0.111.2

danuals, 2020-06-14:
Didn’t find out why, as my sd card died soon after.  Fresh install and then esphome as one of the first things to install and it went ok.

closed 2020-10-26 by stale[bot]

PMaxx-Gaming, 2021-01-22:
Did anyone ever find a fix for this? I'm having the same issue, running HassOS 2021.1.4

SmartHomeFanatic, 2021-01-22:
I had the same issue. Couldn't update via CLI or in the supervisor dashboard. Strangely update worked when doing it in the "System" tab in the super visor.

PMaxx-Gaming, 2021-01-22:
Not trying to update anything, just trying to get ESPHome to install

SmartHomeFanatic, 2021-01-22:
Oh sorry, replied that to the wrong topic.
Only thing to get rid of the error you have was to reinstall HA completly

PMaxx-Gaming, 2021-01-22:
Brutal...