esphome/issues #942: MQTT state/command topics not working on 'switch' component

jncss, 2019-12-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Command line

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, Wemos D1 mini

**Affected component:**

https://esphome.io/components/switch/index.html

**Description of problem:**
'prova/switch/led/state' and 'prova/switch/led/command' MQTT topics doesn't work

**Problem-relevant YAML-configuration entries:**
```yaml

esphome:
  name: prova
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "..."
  password: "..."

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Prova Fallback Hotspot"

captive_portal:

logger:

ota:

mqtt:
  broker: 10.10.10.203

switch:
  - platform: gpio
    id: led
    pin: D4
    inverted: yes

```

**Logs (if applicable):**

```
0:01:21][C][mqtt:051]: MQTT:
[10:01:21][C][mqtt:053]: Server Address: 10.10.10.203:1883 (10.10.10.203)
[10:01:21][C][mqtt:054]: Username: ''
[10:01:21][C][mqtt:055]: Client ID: 'prova-840d8e8f4c0b'
[10:01:21][C][mqtt:057]: Discovery prefix: 'homeassistant'
[10:01:21][C][mqtt:058]: Discovery retain: YES
[10:01:21][C][mqtt:060]: Topic Prefix: 'prova'
[10:01:21][C][mqtt:062]: Log Topic: 'prova/debug'
[10:01:21][C][mqtt:065]: Availability: 'prova/status'
[10:01:21][C][mqtt.switch:038]: MQTT Switch 'led':
[10:01:21][C][mqtt.switch:039]: State Topic: 'prova/switch/led/state'
[10:01:21][C][mqtt.switch:039]: Command Topic: 'prova/switch/led/command'

```

**Additional information and things you've tried:**

Seems like generated code is not calling MQTTSwitchComponent::setup()
If I add
            mqtt_mqttswitchcomponent->setup();
on generated code, everything is working fine.

Maybe I forgot something on 'yaml' file...

micw, 2020-02-06:
I have the same issue with "light" topics

myoung34, 2020-02-23:
Same here on `binary_sensor`

reimerp, 2020-05-30:
I also tried to add a switch but could not manage to get it working:
the mqtt/state is triggered when the button is toggled by web server, but it seems not listening on mqtt/command

bc547, 2020-06-15:
Same problem here (with v1.14.4). The correct topic names are shown at boot:
```
[19:23:20][C][mqtt.switch:038]: MQTT Switch 'doorlock':
[19:23:20][C][mqtt.switch:039]:   State Topic: 'sonoff_sv_1/switch/doorlock/state'
[19:23:20][C][mqtt.switch:039]:   Command Topic: 'sonoff_sv_1/switch/doorlock/command'
[19:23:20][C][mqtt.binary_sensor:018]: MQTT Binary Sensor 'pushbutton':
[19:23:20][C][mqtt.binary_sensor:019]:   State Topic: 'sonoff_sv_1/binary_sensor/pushbutton/state'
```
But it does not listen (e.g. switch command) or publish (e.g. button state) to those topics.

bc547, 2020-06-15:
> Seems like generated code is not calling MQTTSwitchComponent::setup()
> If I add
> mqtt_mqttswitchcomponent->setup();
> on generated code, everything is working fine.

It doesn't seem to work with me. I placed it right before App.setup().
Where do you put that line exactly?

bc547, 2020-06-16:
Update:  This is documented behavior.  

The entity also needs a name property. If it does not have a name, it is considered 'internal' and the controls are not exposed to the front-end.

From the switch component documentation page:

> internal (Optional, boolean): Mark this component as internal. Internal components will not be exposed to the frontend (like Home Assistant). Only specifying an id without a name will implicitly set this to true.

Once I also specify a name in my config, everything works fine.

glmnet, 2020-06-16:
I’m closing. there is a fr https://github.com/esphome/feature-requests/issues/600 you can upvote or propose other options

closed 2020-06-16 by glmnet