esphome/issues #943: zyaura Sensor reported invalid data, ZyAura CO2': Sending state nan ppm with 0 decimals of accuracy

vizgin, 2019-12-22:
**Operating environment/Installation (Hass.io):**

**ESP (ESP8266):**

**Affected component:**

**Description of problem:**

Missing sensor data zyaura

**Problem-relevant YAML-configuration entries:**
```yaml

esphome:
  name: esp01_1
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "ххх"
  password: "ххх"

logger:
    level: VERY_VERBOSE
api:

ota:

captive_portal:

sensor:
  - platform: zyaura
    clock_pin: D1
    data_pin: D2
    co2:
      name: "ZyAura CO2"
    temperature:
      name: "ZyAura Temperature"
#    humidity:
#      name: "ZyAura Humidity"
    update_interval: 60s

INFO Reading configuration /config/esphome/esp01.yaml...
INFO Configuration is valid!
esphome:
  name: esp01_1
  platform: ESP8266
  board: nodemcuv2
  platformio_options: {}
  arduino_version: espressif8266@2.2.3
  board_flash_mode: dout
  build_path: esp01_1
  libraries: []
  esp8266_restore_from_flash: false
  includes: []
wifi:
  fast_connect: false
  output_power: 20.0
  reboot_timeout: 15min
  power_save_mode: NONE
  domain: .local
  networks:
  - ssid: хххх
    password: хххх
    priority: 0.0
  use_address: esp01_1.local
logger:
  level: VERY_VERBOSE
  hardware_uart: UART0
  baud_rate: 115200
  esp8266_store_log_strings_in_flash: true
  logs: {}
  tx_buffer_size: 512
api:
  port: 6053
  password: ''
  reboot_timeout: 15min
ota:
  safe_mode: true
  port: 8266
  password: ''
sensor:
- platform: zyaura
  clock_pin:
    number: 5
    mode: INPUT
    inverted: false
  data_pin:
    number: 4
    mode: INPUT
    inverted: false
  co2:
    name: ZyAura CO2
    accuracy_decimals: 0
    force_update: false
    unit_of_measurement: ppm
    icon: mdi:periodic-table-co2
  temperature:
    name: ZyAura Temperature
    accuracy_decimals: 1
    force_update: false
    unit_of_measurement: °C
    icon: mdi:thermometer
  update_interval: 60s

```

**Logs (if applicable):**

```
INFO Connecting to esp01_1.local:6053 (192.168.88.231)
INFO Successfully connected to esp01_1.local
[13:40:08][I][app:100]: ESPHome version 1.14.3 compiled on Dec 22 2019, 13:37:23
[13:40:08][C][wifi:415]: WiFi:
[13:40:08][C][wifi:283]:   SSID: [redacted]
[13:40:08][C][wifi:284]:   IP Address: 192.168.88.231
[13:40:08][C][wifi:286]:   BSSID: [redacted]
[13:40:08][C][wifi:287]:   Hostname: 'esp01_1'
[13:40:08][C][wifi:291]:   Signal strength: -66 dB ▂▄▆█
[13:40:08][V][wifi:293]:   Priority: 0.0
[13:40:08][C][wifi:295]:   Channel: 8
[13:40:08][C][wifi:296]:   Subnet: 255.255.255.0
[13:40:08][C][wifi:297]:   Gateway: 192.168.88.1
[13:40:08][C][wifi:298]:   DNS1: 192.168.88.1
[13:40:08][C][wifi:299]:   DNS2: 213.108.208.2
[13:40:08][C][logger:175]: Logger:
[13:40:08][C][logger:176]:   Level: VERY_VERBOSE
[13:40:08][C][logger:177]:   Log Baud Rate: 115200
[13:40:08][C][logger:178]:   Hardware UART: UART0
[13:40:08][C][zyaura:096]: ZyAuraSensor:
[13:40:08][C][zyaura:097]:   Pin Clock: GPIO5 (Mode: INPUT)
[13:40:08][C][zyaura:098]:   Pin Data: GPIO4 (Mode: INPUT)
[13:40:08][C][zyaura:099]:   Update Interval: 60.0s
[13:40:08][C][zyaura:101]:   CO2 'ZyAura CO2'
[13:40:08][C][zyaura:101]:     Unit of Measurement: 'ppm'
[13:40:08][C][zyaura:101]:     Accuracy Decimals: 0
[13:40:08][C][zyaura:101]:     Icon: 'mdi:periodic-table-co2'
[13:40:08][C][zyaura:102]:   Temperature 'ZyAura Temperature'
[13:40:08][C][zyaura:102]:     Unit of Measurement: '°C'
[13:40:08][C][zyaura:102]:     Accuracy Decimals: 1
[13:40:08][C][zyaura:102]:     Icon: 'mdi:thermometer'
[13:40:08][C][ota:029]: Over-The-Air Updates:
[13:40:08][C][ota:030]:   Address: esp01_1.local:8266
[13:40:08][C][api:095]: API Server:
[13:40:08][C][api:096]:   Address: esp01_1.local:6053
[13:40:13][VV][api.service:250]: on_ping_request: PingRequest {}
[13:40:13][VV][api.service:035]: send_ping_response: PingResponse {}
[13:40:18][VV][api.service:250]: on_ping_request: PingRequest {}
[13:40:18][VV][api.service:035]: send_ping_response: PingResponse {}
[13:40:23][VV][api.service:250]: on_ping_request: PingRequest {}
[13:40:23][VV][api.service:035]: send_ping_response: PingResponse {}
[13:40:28][VV][api.service:250]: on_ping_request: PingRequest {}
[13:40:28][VV][api.service:035]: send_ping_response: PingResponse {}
[13:40:33][VV][api.service:250]: on_ping_request: PingRequest {}
[13:40:33][VV][api.service:035]: send_ping_response: PingResponse {}
[13:40:38][VV][api.service:250]: on_ping_request: PingRequest {}
[13:40:38][VV][api.service:035]: send_ping_response: PingResponse {}
[13:40:43][VV][api.service:250]: on_ping_request: PingRequest {}
[13:40:43][VV][api.service:035]: send_ping_response: PingResponse {}
[13:40:46][VV][scheduler:131]: Running interval 'update' with interval=60000 last_execution=182 (now=60182)
[13:40:46][V][sensor:013]: 'ZyAura CO2': Received new state nan
[13:40:46][D][sensor:092]: 'ZyAura CO2': Sending state nan ppm with 0 decimals of accuracy
[13:40:46][W][zyaura:086]: Sensor reported invalid data. Is the update interval too small?
[13:40:46][V][sensor:013]: 'ZyAura Temperature': Received new state nan
[13:40:46][D][sensor:092]: 'ZyAura Temperature': Sending state nan °C with 1 decimals of accuracy
[13:40:46][W][zyaura:086]: Sensor reported invalid data. Is the update interval too small?
[13:40:48][VV][api.service:250]: on_ping_request: PingRequest {}
[13:40:48][VV][api.service:035]: send_ping_response: PingResponse {}
[13:40:53][VV][api.service:250]: on_ping_request: PingRequest {}
[13:40:53][VV][api.service:035]: send_ping_response: PingResponse {}
[13:40:58][VV][api.service:250]: on_ping_request: PingRequest {}
[13:40:58][VV][api.service:035]: send_ping_response: PingResponse {}
[13:41:03][VV][api.service:250]: on_ping_request: PingRequest {}
[13:41:04][VV][api.service:035]: send_ping_response: PingResponse {}
[13:41:10][VV][api.service:250]: on_ping_request: PingRequest {}
[13:41:10][VV][api.service:035]: send_ping_response: PingResponse {}
[13:41:15][VV][api.service:250]: on_ping_request: PingRequest {}
[13:41:15][VV][api.service:035]: send_ping_response: PingResponse {}
[13:41:20][VV][api.service:250]: on_ping_request: PingRequest {}
[13:41:20][VV][api.service:035]: send_ping_response: PingResponse {}
[13:41:25][VV][api.service:250]: on_ping_request: PingRequest {}
[13:41:25][VV][api.service:035]: send_ping_response: PingResponse {}
[13:41:30][VV][api.service:250]: on_ping_request: PingRequest {}
[13:41:30][VV][api.service:035]: send_ping_response: PingResponse {}
[13:41:35][VV][api.service:250]: on_ping_request: PingRequest {}
[13:41:35][VV][api.service:035]: send_ping_response: PingResponse {}
[13:41:40][VV][api.service:250]: on_ping_request: PingRequest {}
[13:41:40][VV][api.service:035]: send_ping_response: PingResponse {}
[13:41:45][VV][api.service:250]: on_ping_request: PingRequest {}
[13:41:45][VV][api.service:035]: send_ping_response: PingResponse {}
[13:41:46][VV][scheduler:131]: Running interval 'update' with interval=60000 last_execution=60182 (now=120183)
[13:41:46][V][sensor:013]: 'ZyAura CO2': Received new state nan
[13:41:46][D][sensor:092]: 'ZyAura CO2': Sending state nan ppm with 0 decimals of accuracy
[13:41:46][W][zyaura:086]: Sensor reported invalid data. Is the update interval too small?
[13:41:46][V][sensor:013]: 'ZyAura Temperature': Received new state nan
[13:41:46][D][sensor:092]: 'ZyAura Temperature': Sending state nan °C with 1 decimals of accuracy
[13:41:46][W][zyaura:086]: Sensor reported invalid data. Is the update interval too small?
[13:41:50][VV][api.service:250]: on_ping_request: PingRequest {}
[13:41:50][VV][api.service:035]: send_ping_response: PingResponse {}
[13:41:50][I][ota:046]: Boot seems successful, resetting boot loop counter.
[13:41:50][VV][preferences:051]: SAVE 32: 0=0x00000000 1=0x0DEFE4E3 (Type=233825507, CRC=0x0DEFE4E3)
[13:41:55][VV][api.service:250]: on_ping_request: PingRequest {}
[13:41:55][VV][api.service:035]: send_ping_response: PingResponse {}
[13:42:00][VV][api.service:250]: on_ping_request: PingRequest {}
[13:42:00][VV][api.service:035]: send_ping_response: PingResponse {}
[13:42:05][VV][api.service:250]: on_ping_request: PingRequest {}
[13:42:05][VV][api.service:035]: send_ping_response: PingResponse {}
[13:42:10][VV][api.service:250]: on_ping_request: PingRequest {}
[13:42:10][VV][api.service:035]: send_ping_response: PingResponse {}
[13:42:15][VV][api.service:250]: on_ping_request: PingRequest {}
[13:42:15][VV][api.service:035]: send_ping_response: PingResponse {}
[13:42:20][VV][api.service:250]: on_ping_request: PingRequest {}
[13:42:20][VV][api.service:035]: send_ping_response: PingResponse {}
[13:42:25][VV][api.service:250]: on_ping_request: PingRequest {}
[13:42:25][VV][api.service:035]: send_ping_response: PingResponse {}
[13:42:30][VV][api.service:250]: on_ping_request: PingRequest {}
[13:42:30][VV][api.service:035]: send_ping_response: PingResponse {}
[13:42:35][VV][api.service:250]: on_ping_request: PingRequest {}
[13:42:35][VV][api.service:035]: send_ping_response: PingResponse {}

```

**Additional information and things you've tried:**

jordes, 2020-02-10:
i have the same problem

jordes, 2020-02-17:
@vizgin hey could you solve it?

glmnet, 2020-02-18:
CC @Anonym-tsk

Anonym-tsk, 2020-02-18:
I see NodeMCU on photo, `esp01_1` in config and pins D1 and D2. Are you sure?

mrsantik, 2020-02-25:
The same connection - the same problem.

vizgin, 2020-03-14:
Yes!
I tchange D1 => D2 and D2 => D1 in config and it start's work!

closed 2020-03-14 by vizgin