esphome/issues #944: Compilation error for sonoff basic on hass.io

Matt2cv, 2019-12-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on Odroid C2
ESPHOME Beta 1.14.3

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
Sonoff Basic

**Affected component:**

https://esphome.io/devices/sonoff_basic.html

**Description of problem:**
When I compile a yaml file for a new ESP8266 device, I've got a Core Dumped error.  
Same error with ESPHOME stable.
I make a test with ESP32 node and I have not got errors.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ventilateur_cuisine
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "xxxxxx"
  password: "xxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ventilateur Cuisine"
    password: "xxxxxx"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
INFO Reading configuration /config/esphome/ventilateur_cuisine.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/ventilateur_cuisine
Processing ventilateur_cuisine (board: esp01_1m; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/ventilateur_cuisine/.pioenvs/ventilateur_cuisine/src/esphome/components/api/api_connection.cpp.o
Compiling /data/ventilateur_cuisine/.pioenvs/ventilateur_cuisine/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/ventilateur_cuisine/.pioenvs/ventilateur_cuisine/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/ventilateur_cuisine/.pioenvs/ventilateur_cuisine/src/esphome/components/api/api_server.cpp.o
Illegal instruction (core dumped)
*** [/data/ventilateur_cuisine/.pioenvs/ventilateur_cuisine/src/esphome/components/api/api_connection.cpp.o] Error 132
Illegal instruction (core dumped)
*** [/data/ventilateur_cuisine/.pioenvs/ventilateur_cuisine/src/esphome/components/api/api_pb2.cpp.o] Error 132
Illegal instruction (core dumped)
*** [/data/ventilateur_cuisine/.pioenvs/ventilateur_cuisine/src/esphome/components/api/api_pb2_service.cpp.o] Error 132
Illegal instruction (core dumped)
*** [/data/ventilateur_cuisine/.pioenvs/ventilateur_cuisine/src/esphome/components/api/api_server.cpp.o] Error 132
========================== [FAILED] Took 6.15 seconds ==========================
```

**Additional information and things you've tried:**

geoffreyheller, 2020-01-03:
I am also running on an ODroid C2 and I am getting the exact same error messages.

zvldz, 2020-03-03:
https://github.com/esphome/issues/issues/805

Matt2cv, 2020-03-04:
I abandoned Esphome. I passed my modules under Tasmota

OttoWinter, 2020-07-24:
Closing as duplicate of https://github.com/esphome/issues/issues/805

closed 2020-07-24 by OttoWinter