esphome/issues #945: Create new device under latest dev file names are converted to special characters.

marc-gist, 2019-12-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip using  ` pip install --upgrade git+https://github.com/esphome/esphome`
Python 3.6.9
`esphome config/ dashboard`

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**Affected component:**

Configure/Make new

**Description of problem:**

When I create a new item, the file and name convert to: 

> b&#39;quinled_quad_2&#39; 

edit file shows

> Edit b'quinled_quad_2'.yaml

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

poldim, 2020-01-06:
Confirming the same thing happens to me.  

##**Steps to reproduce:**

1. Go to the dashboard
2. Click the plus button to add a new device
3. Name the device "esphome_dining_room_dimmer"
4. Fill out the rest of the form to have the device created
5. UI shows the following: 
[image]
6. The filename as created: `b'esphome_dining_room_dimmer'.yaml`
[image]

timsavage, 2020-01-09:
A pull request with the fix is open.

brandond, 2020-01-09:
Fixed by https://github.com/esphome/esphome/pull/941

closed 2020-01-09 by brandond