esphome/issues #946: Bang Bang Climate Controller MQTT state flood

delfer, 2019-12-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Affected component:**

https://esphome.io/components/climate/bang_bang.html

**Description of problem:**
Component sending messages into MQTT topics over 50 times per second!
.../mode/state auto
.../current_temperature/state 19.6
.../target_temperature_low/state 17.0
.../target_temperature_high/state 18.0

When data from sensor updates only 1 time per minute

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esph_rack_heater
  platform: ESP8266
  board: nodemcuv2
wifi:
  ssid: "XXX"
  password: "XXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Dev1 Fallback Hotspot"
    password: "XXX"

  # Optional manual IP
  manual_ip:
    static_ip: 192.168.1.2
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "XXX"

ota:
  password: "XXX"

web_server:
  port: 80

mqtt:
  broker: "192.168.1.3"
  username: XXX
  password: XXX
dallas:
  - pin: GPIO5
sensor:
  - platform: dallas
    index: 0
    name: "Rack temperature"
    id: rack_temp
    filters:
      - delta: 0.1
switch:
  - platform: gpio
    pin: GPIO4
    name: "Rack heater"
    id: heater
climate:
  - platform: bang_bang
    sensor: rack_temp
    default_target_temperature_low: 15 °C
    default_target_temperature_high: 16 °C
    name: "Rack thermostat"
    heat_action :
      - switch.turn_on: heater
    idle_action:
      - switch.turn_off: heater
```

**Logs (if applicable):**

```
[20:02:02][I][app:100]: ESPHome version 1.14.3 compiled on Dec 25 2019, 20:01:31
...
[20:02:02][C][logger:175]: Logger:
[20:02:02][C][logger:176]:   Level: DEBUG
[20:02:02][C][logger:177]:   Log Baud Rate: 115200
[20:02:02][C][logger:178]:   Hardware UART: UART0
[20:02:02][C][bang_bang.climate:149]: Bang Bang Climate 'Rack thermostat'
[20:02:02][C][bang_bang.climate:150]:   Supports HEAT: YES
[20:02:02][C][bang_bang.climate:151]:   Supports COOL: NO
[20:02:02][C][bang_bang.climate:152]:   Supports AWAY mode: NO
[20:02:02][C][bang_bang.climate:153]:   Default Target Temperature Low: 15.0°C
[20:02:02][C][bang_bang.climate:154]:   Default Target Temperature High: 16.0°C
...
0:02:02][C][dallas.sensor:070]: DallasComponent:
[20:02:02][C][dallas.sensor:071]:   Pin: GPIO5 (Mode: INPUT)
[20:02:02][C][dallas.sensor:072]:   Update Interval: 60.0s
[20:02:02][D][dallas.sensor:077]:   Found sensors:
[20:02:02][D][dallas.sensor:080]:     0x5A01191277357828
[20:02:02][C][dallas.sensor:085]:   Device 'Rack temperature'
[20:02:02][C][dallas.sensor:085]:     Unit of Measurement: '°C'
[20:02:02][C][dallas.sensor:085]:     Accuracy Decimals: 1
[20:02:02][C][dallas.sensor:085]:     Icon: 'mdi:thermometer'
[20:02:02][C][dallas.sensor:087]:     Index 0
[20:02:02][C][dallas.sensor:093]:     Address: 0x5A01191277357828
[20:02:02][C][dallas.sensor:094]:     Resolution: 12
[20:02:02][C][dallas.sensor:085]:   Device 'Rack heater input temp'
[20:02:02][C][dallas.sensor:085]:     Unit of Measurement: '°C'
[20:02:02][C][dallas.sensor:085]:     Accuracy Decimals: 1
...
```
(logs truncated)

**Additional information and things you've tried:**
-- nothing --

michalsteyn, 2020-01-23:
I am also experiencing the exact same behavior.
Had to set Climate to Internal to avoid the mqtt flood. But a better solution would be appreciated.

User8283, 2020-02-14:
I observe exactly the same behavior

delfer, 2020-07-24:
Up

OttoWinter, 2020-07-24:
This is because `publish_state_` [here](https://github.com/esphome/esphome/blob/dev/esphome/components/mqtt/mqtt_climate.cpp#L211) publishes on all topics again if publishing to one topic fails (for example due to lack of memory).

Probably the code would need to be updated to only re-publish failed topics.

michalsteyn, 2020-11-19:
This issue seems to have been resolved. I retested this using the latest stable version and no longer experiencing this issue.

closed 2021-03-26 by delfer