esphome/issues #948: REFUSAL to compile if api.platformio.org is down!

erlanger, 2019-12-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip, linux

**ESP (ESP32/ESP8266, Board/Sonoff):**

d1 mini

**Affected component:**

all components

**Description of problem:**
esphome refuses to compile if platformio.org is down

**Problem-relevant YAML-configuration entries:**
```yaml
# not yaml related

```

**Logs (if applicable):**

```sh
$ esphome *.yaml compile
INFO Reading configuration plug_1.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d plug_1
Processing plug_1 (board: esp8285; framework: arduino; platform: espressif8266@2.2.3)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Looking for ESPAsyncTCP-esphome library in registry
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESPAsyncTCP-esphome%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f8a5c49d130>: Failed to establish a new connection: [Errno 111] Connection refused')) (incremented retry: max=5, total=1)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESPAsyncTCP-esphome%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f8a5c49d4c0>: Failed to establish a new connection: [Errno 111] Connection refused')) (incremented retry: max=5, total=2)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESPAsyncTCP-esphome%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f8a5c49d7f0>: Failed to establish a new connection: [Errno 111] Connection refused')) (incremented retry: max=5, total=3)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESPAsyncTCP-esphome%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f8a5c49db20>: Failed to establish a new connection: [Errno 111] Connection refused')) (incremented retry: max=5, total=4)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESPAsyncTCP-esphome%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f8a5c49de50>: Failed to establish a new connection: [Errno 111] Connection refused')) (incremented retry: max=5, total=5)
APIRequestError: [API] Could not connect to PlatformIO API Service. Please try later.:
  File "/home/hassdev/.local/lib/python3.8/site-packages/platformio/builder/main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/hassdev/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/hassdev/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/hassdev/.platformio/platforms/espressif8266/builder/main.py", line 203:
    target_elf = env.BuildProgram()
  File "/home/hassdev/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/home/hassdev/.local/lib/python3.8/site-packages/platformio/builder/tools/platformio.py", line 122:
    _build_project_deps(env)
  File "/home/hassdev/.local/lib/python3.8/site-packages/platformio/builder/tools/platformio.py", line 47:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "/home/hassdev/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/home/hassdev/.local/lib/python3.8/site-packages/platformio/builder/tools/piolib.py", line 1043:
    project.install_dependencies()
  File "/home/hassdev/.local/lib/python3.8/site-packages/platformio/builder/tools/piolib.py", line 878:
    lm.install(uri)
  File "/home/hassdev/.local/lib/python3.8/site-packages/platformio/managers/lib.py", line 346:
    name = "id=%d" % self.search_lib_id(
  File "/home/hassdev/.local/lib/python3.8/site-packages/platformio/managers/lib.py", line 251:
    result = util.get_api_result("/v2/lib/search",
  File "/home/hassdev/.local/lib/python3.8/site-packages/platformio/util.py", line 376:
    raise exception.APIRequestError(
==================================================================================== [FAILED] Took 31.30 seconds ====================================================================================

```

**Additional information and things you've tried:**

ToSa27, 2019-12-26:
Same here. They mentioned the intent to improve the behavior when the api endpoint is down over in the platformio issue so that it would not block compilation but seems like that did not happen:
https://github.com/platformio/platformio-core/issues/1658

ToSa27, 2019-12-26:
Seems to be back online now...

igorantolic, 2019-12-26:
I can compile on my Windows PC (As I did yesterday) but atfer installing EspHome on RPI4 Home Assistant doesnt work

So I am running espHome from http://hassio4.local:8123/hassio/ingress/15ef4d2f_esphome
and it says 

HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Looking for ESPAsyncTCP-esphome library in registry
LibraryManager: Installing id=6757 @ 1.2.2
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /lib/download/6757?version=1.2.2 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb5b91750>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=1)
APIRequestError: [API] Invalid response: b'<html>\r\n<head><title>502 Bad Gateway</title></head>\r\n<body bgcolor="white">\r\n<center><h1>502 Bad Gateway</h1></center>\r\n<hr><center>nginx</center>\r\n</body>\r\n</html>\r\n':

igorantolic, 2019-12-26:
> Seems to be back online now...

Nope HTTP 504 right now on http://api.platformio.org/

erlanger, 2019-12-26:
Ahh.. thanks; this really should be fixed because compiling code should not depend on access to the internet.

brandond, 2019-12-27:
https://github.com/platformio/platformio-core/issues/1659

OttoWinter, 2019-12-31:
Yeah I know, this has happened before unfortunately.

Back then I looked into what we could do to become more independent of that service (by installing all dependencies beforehand for example), but unfortunately some issues in platformio stopped that from working.

There are a few approaches, each of which I found not to work well:
 1. Install from git/https directly (instead of platformio lib registry). Problem is user needs to have git installed and the download+install takes significantly longer because git needs to download the entire git history when cloning a repo (and I don't know of a way to set `--depth=1` in lib_deps).
 2. Installing libraries globally via `platformio lib -g install`. Due to complicated reasons, the "global" install mode for platformio lib is broken - you can download libs globally, but they're not used from platformio.ini files.
 3. Packaging libraries directly with esphome install (in pip wheel) - would make adding libraries even more difficult (for example, adding a large library would essentially be prohibited because of the added size on everybody's installs, not just the people that use it).

ivankravets, 2020-01-01:
Sorry for the issue. We have started work on improving our cloud infrastructure. The new PlatformIO Registry is planned for release in January. The current version of our registry was created in 2014 and had a lot of limits for scaling :(
We also created status page https://status.platformio.org/

Please give us a few weeks to deploy the new registry. We hope it should help to avoid similar issues with outage.

OttoWinter, 2020-01-01:
@ivankravets btw, I fully understand that this is can just happen (especially at the worst times lol). And I'm very grateful you got it working again very quickly!

Still though I'd like to try to at least partially make the local installs more independent of the current registry availability (same would go for any other service, even major pages like github are down sometimes).

Now that we have you in this thread I'd like to ask: How does the `--global` option for `pio lib install` work (or how is it supposed to work). As far as I understand it should install (as in download+unpack) the package in the `~/.platformio` folder. 

Unfortunately though, even if a library is installed globally through `pio lib -g install`, a PIO registry downtime made compiling not possible (at least that was the case last time I checked, and I don't remember exactly how I simulated PIO registry replying with 500s locally). Do you know if that has been fixed in the last year or so? Does it have to do with fetching additional metadata (like library ID)? Could we somehow help with a fix?

ivankravets, 2020-01-01:
See https://docs.platformio.org/en/latest/librarymanager/ldf.html

So, if you install library to a global storage it will be accessible for any PlatformIO project. We don’t recommend using global storage. This is a bad practice which goes from Arduino IDE. To be honest, we would be happy to drop this global storage and it could save us a lot of time providing a support to our community and help them resolve conflicts with different versions. 

I personally hate this idea when Projects use shared global libraries. You do update of one library and automatically break other projects which depend on the previous version. This is the one of reasons why people move from Arduino IDE to PlatformIO. PlatformIO allows them to declare dependencies per build environment. This is isolated operation and guarantees full reproducibility on any OS. 

We highly recommend using `lib_deps` option in PlatformIO.ini file. The new registry will bring huge improvement over packages and their conflicting names. You will be able to specify dependency in a form of owner/library name. So, now, the library name will be unique for its owner. Owners can’t publish 2 libraries with the same name. 

I’ve just checked PlatformIO.ini of this projects and it looks great. Please do not do any changes now. This our task now to make registry stable. The root of problem was in our dedicated server which we used in Ukraine. It was a little bit old. As result, motherboard has stopped working. Now, we build a fully new architecture of how our products will be deployed to customers. We will mirror our data into a few nodes and Amazon AWS will be responsible to balance this. We don’t need a lot of resources. Python requires 50Mb of RAM and a few percents of CPU. The problem is in hardware and connectivity. We want to have backup nodes  in different data centers. If one goes down, others should work. But! If Amazon goes down, we will have a short outage. There is also solution for this case. But users should not be affected. 

These improvements require changes to PlatformIO Core too. We will work on them when new registry will be released.

glmnet, 2020-01-01:
In contrast to installing a single version globally other approaches are just a multi-version local cache, so pio would try to get a library package from the cache first if there then copy it to the local directory as usual. If not it will download and keep a version in the user cache too.

brandond, 2020-01-02:
@ivankravets better infrastructure is great, but it'd be nice if I could take a project completely offline for whatever reason, via command-line flag or .ini file option.

ivankravets, 2020-01-03:
@brandond you can do this even today. If you build project first time with internet (we need to download source code of all dependencies), you can do the same in off-line mode next time.

germanotero, 2020-06-04:
I guess you are deploying a new registry right now. :(

ivankravets, 2020-06-04:
Now, everything works as expected. Try from your mobile phone using mobile network this URL https://api.platformio.org/lib/info/1

Does itwork?

germanotero, 2020-06-04:
hmm, That works, even inside the docker container, but I still get the error about no internet connection

ivankravets, 2020-06-04:
Sorry, we can't help. Please check your network interface and its configuration.

germanotero, 2020-06-04:
it works fine.. actually,I can curl from inside the docker to the URL you gave me and get the Json.

germanotero, 2020-06-04:
And also, it was working early today, so I'm pretty sure platformio guys did some change, they tell me to move to the new version, but ESP home uses 4.0.3, if I move to the last one, then esphome doesn't starts due to dependency issues..

glmnet, 2020-06-04:
Is this https://api.platformio.org/lib/info/6758 url exactly the one failing?

glmnet, 2020-06-04:
Seems to be a (new?) ESPHome issue https://github.com/esphome/issues/issues/1265
I still don't know, they say solution right now is to use ESPHome dev which uses a newer version of platformio, may this actually be the issue @ivankravets ?

glmnet, 2020-06-04:
Theres also another workaround that involves an iptables command to do a redirect which might point the issue more to a docker configuration change

ajfriesen, 2020-08-07:
Had the same issue on hassio supervised installation.

Uninstalling and installing the addon did work for me.

closed 2020-12-19 by stale[bot]