esphome/issues #950: SGP30 Baseline readings do not work

BertrumUK, 2019-12-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini
**Affected component:**

SGP30

**Description of problem:**
Have just started using a [SGP30](https://esphome.io/components/sensor/sgp30.html) air quality sensor but after waiting 12 hours for the baseline reading to add to the config this causes either the eCO2 or the TVOC to stop reporting any values or to show their max readings of 60000 or 65000 respectively.

According to the [datasheet](https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/0_Datasheets/Gas/Sensirion_Gas_Sensors_SGP30_Datasheet.pdf) there should be a baseline figure for both readings whereas this sensor reports and uses only 1 baseline setting.

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: sgp30
    eco2:
      name: "Living Room eCO2"
      accuracy_decimals: 0
      filters:
        - median:
            window_size: 5
            send_every: 6
            send_first_at: 3
    tvoc:
      name: "Living Room  TVOC"
      accuracy_decimals: 0
    address: 0x58
    update_interval: 10s
    #baseline: 0x016D #0x747F #0xCA15 #0x1491 #0x5F4A #0x7AE8
    compensation:
      temperature_source: temperature
      humidity_source: humidity

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

BertrumUK, 2019-12-26:
I use an external DHT sensor to improve the readings but find that with no baseline set I get readings for both sensors only.

panuruj, 2019-12-30:
For the baseline calibration, I found that it's better to leave the sensor outside for the required 12 hours period before reading the baseline value. 

> According to the datasheet there should be a baseline figure for both readings whereas this sensor reports and uses only 1 baseline setting.

According to the code, the first 2 hex digits (most-significant byte) is the eCO2 baseline, and the last 2 hex digit (least-significant byte) is the TVOC baseline.

https://github.com/esphome/esphome/blob/c8ccb06f11e28c035f2d340444cfb3b3e49110ff/esphome/components/sgp30/sgp30.cpp#L112

BertrumUK, 2019-12-30:
Hello, That makes sense having them combined. 

Its not ideal leaving it outdoors for 12 hours in the UK but I guess you could use a previous baseline and get a new one after an hour.

panuruj, 2019-12-30:
I don't know if you saw my last edit on the comment above. The ESPHome implementation appears to be incorrect compared to the SGP30 datasheet and Adafruit SGP30 implementation. Each eCO2 and TVOC baseline should each have 4 hex digits. I will attempt to fix it in the next few day and create a pull request.

closed 2020-01-12 by OttoWinter