esphome/issues #951: Cannot compile YAML with name >19 characters and MQTT enabled

inforsgit, 2019-12-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Windows 10, Mobaxterm to Windows Terminal, ESPHome installed via pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

Confirmed with both ESP32 and ESP8266 (Wemos D1 Mini)

**Affected component:**

MQTT - https://esphome.io/components/mqtt.html

**Description of problem:**
I was getting crashes when trying to compile a new YAML and couldn't work out what was causing it. I rarely use MQTT but was including a custom component that required it so added it to the configuration. I initially thought it was the custom component but on closer inspection the error below occurs even without loading the custom component.

When the `name:` of the ESPHome device has more than 19 characters AND has MQTT enabled in the YAML, the compilation crashes with the error below. On reducing the length of the  `name:` for the device compilation can complete successfully.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: 01234567890123456789
  platform: ESP32
  board: esp32doit-devkit-v1
  board_flash_mode: dio

# Setup the Wi-Fi connection
wifi:
  ssid: <WiFi SSID>
  password: <WiFi Password>

# Enable the Home Assistant API as the communications method
api:

# Enable logging
logger:

# Enable OTA updates
ota:

mqtt:
  broker: 192.168.1.10
  username: <MQTT User>
  password: <MQTT Password>
```

**Logs (if applicable):**

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 1.11.0 > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10004.191002 (1.0.4), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for AsyncTCP-esphome library in registry
Found: https://platformio.org/lib/show/6798/AsyncTCP-esphome
LibraryManager: Installing id=6798 @ 1.1.1
Using cache: C:\Users\thoma\DOCUME~1\MOBAXT~1\home\.platformio\.cache\30\691256ad6e69109d4be2b0d2f9d51a30
AsyncTCP-esphome @ 1.1.1 has been successfully installed!
Looking for Hash library in registry
Warning! Library `{'name': 'Hash', 'requirements': None}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'name': 'Hash', 'requirements': None}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Looking for AsyncMqttClient-esphome library in registry
Found: https://platformio.org/lib/show/6759/AsyncMqttClient-esphome
LibraryManager: Installing id=6759 @ 0.8.4
Using cache: C:\Users\thoma\DOCUME~1\MOBAXT~1\home\.platformio\.cache\0a\430bb68b5512c64937df9b3c4b147f0a
AsyncMqttClient-esphome @ 0.8.4 has been successfully installed!
Installing dependencies
Looking for ESPAsyncTCP-esphome library in registry
[... 51 lines cut ...]
In file included from .piolibdeps/01234567890123456789/ArduinoJson-esphomelib_ID3837/src/ArduinoJson/Deserialization/../Serialization/../Serialization/FloatParts.hpp:9:0,
                 from .piolibdeps/01234567890123456789/ArduinoJson-esphomelib_ID3837/src/ArduinoJson/Deserialization/../Serialization/JsonWriter.hpp:11,
                 from .piolibdeps/01234567890123456789/ArduinoJson-esphomelib_ID3837/src/ArduinoJson/Deserialization/../Serialization/JsonSerializer.hpp:7,
                 from .piolibdeps/01234567890123456789/ArduinoJson-esphomelib_ID3837/src/ArduinoJson/Deserialization/../Serialization/JsonPrintable.hpp:12,
                 from .piolibdeps/01234567890123456789/ArduinoJson-esphomelib_ID3837/src/ArduinoJson/Deserialization/../JsonVariantBase.hpp:11,
                 from .piolibdeps/01234567890123456789/ArduinoJson-esphomelib_ID3837/src/ArduinoJson/Deserialization/../JsonVariant.hpp:13,
                 from .piolibdeps/01234567890123456789/ArduinoJson-esphomelib_ID3837/src/ArduinoJson/Deserialization/../JsonBuffer.hpp:12,
                 from .piolibdeps/01234567890123456789/ArduinoJson-esphomelib_ID3837/src/ArduinoJson/Deserialization/JsonParser.hpp:7,
                 from .piolibdeps/01234567890123456789/ArduinoJson-esphomelib_ID3837/src/ArduinoJson/JsonBufferBase.hpp:7,
                 from .piolibdeps/01234567890123456789/ArduinoJson-esphomelib_ID3837/src/ArduinoJson/DynamicJsonBuffer.hpp:7,
                 from .piolibdeps/01234567890123456789/ArduinoJson-esphomelib_ID3837/src/ArduinoJson.hpp:7,
                 from .piolibdeps/01234567890123456789/ArduinoJson-esphomelib_ID3837/src/ArduinoJson.h:9,
                 from src/esphome/components/json/json_util.h:4,
                 from src\esphome\components\mqtt\mqtt_client.h:7,
                 from src\esphome\components\mqtt\custom_mqtt_device.h:4,
                 from src\esphome\components\mqtt\custom_mqtt_device.cpp:1:
.piolibdeps/01234567890123456789/ArduinoJson-esphomelib_ID3837/src/ArduinoJson/Deserialization/../Serialization/../Serialization/../TypeTraits/FloatTraits.hpp:9:32: fatal error: ../
Configuration.hpp: No such file or directory
compilation terminated.
*** [.pioenvs\01234567890123456789\src\esphome\components\json\json_util.cpp.o] Error 1
*** [.pioenvs\01234567890123456789\src\esphome\components\mqtt\custom_mqtt_device.cpp.o] Error 1
```

**Additional information and things you've tried:**
Tried a variety of different name options, with/without special characters, with/without numbers, with/without capital letters but the only constant is the any name less than 20 characters (i.e. 1 to 19) works but from 20 characters on the error appears.

glmnet, 2020-01-17:
Looks more like a **too long path file name issue**, which is triggering when your file name goes more than 19 characters and the mqtt library has some deep directory/file names, however this will in turn depend from where the yaml config file is.

In case you really want those long node names, try moving your yaml file into a path closer to the root drive.

I really don't know how ESPHome can help here

glmnet, 2020-01-17:
You can also try this https://www.howtogeek.com/266621/how-to-make-windows-10-accept-file-paths-over-260-characters/

closed 2020-08-07 by stale[bot]