esphome/issues #952: Node never connect to wifi when static IP and DHT sensor are in the config

BugerDread, 2019-12-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome v1.14.3 docker image esphome/esphome @ ubuntu 18.04 x64, 8GB ram, Intel core2 cpu
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, nodemcuv2 board, AM2302 connected to gpio12 with pullup resistor to +3.3V
**Affected component:**

https://esphome.io/components/wifi.html
https://esphome.io/components/sensor/dht.html

**Description of problem:**
When there is static IP and AM2302 sensor in yaml config, it compiles, uploads (over USB), but ESP never get connected to WiFi, trying again and again (see log bellow -  reason='Probe Request Unsuccessful'). If I comment out static IP parametters OR the DHT sensor block it connects to WiFi without problems, instantly. 

Not sure if it is DHT sensor or WiFi component problem.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

esphome:
  name: humi01
  platform: ESP8266
  board: nodemcuv2

wifi:
#  fast_connect: True
  ssid: "Garden WiFi"
  password: "somepassword"

# if I comment out these 4 lines it connects without problem
  manual_ip:
    static_ip: 192.168.88.49
    gateway: 192.168.88.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Humi01 Fallback Hotspot"
    password: "0WxDuZdCQYqq"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "otapassword"

ota:
  password: "otapassword"

#also if I comment out this sensor block it connects without problem - does not make sense
sensor:
  - platform: dht
    pin: GPIO12
    temperature:
      name: "Living Room Temperature"
    humidity:
      name: "Living Room Humidity"
    update_interval: 60s
    model: AM2302

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

[logger:156]: Log initialized
[16:20:00][C][ota:364]: There have been 0 suspected unsuccessful boot attempts.
[16:20:00][I][app:028]: Running through setup()...
[16:20:00][C][dht:011]: Setting up DHT...
[16:20:00][C][wifi:033]: Setting up WiFi...
[16:20:00][D][wifi:304]: Starting scan...
[16:20:00][D][dht:048]: Got Temperature=25.4°C Humidity=53.1%
[16:20:00][D][sensor:092]: 'Living Room Temperature': Sending state 25.40000 °C with 1 decimals of accuracy
[16:20:00][D][sensor:092]: 'Living Room Humidity': Sending state 53.10000 % with 0 decimals of accuracy
[16:20:06][D][wifi:319]: Found networks:
[16:20:06][I][wifi:365]: - 'Garden WiFi' (00:14:BF:3C:F4:B3) ▂▄▆█
[16:20:06][D][wifi:366]:     Channel: 1
[16:20:06][D][wifi:367]:     RSSI: -53 dB
[16:20:06][I][wifi:365]: - 'Garden WiFi' (00:26:5A:77:C1:E0) ▂▄▆█
[16:20:06][D][wifi:366]:     Channel: 13
[16:20:06][D][wifi:367]:     RSSI: -71 dB
[16:20:06][I][wifi:365]: - 'Garden WiFi' (CE:2D:E0:04:38:65) ▂▄▆█
[16:20:06][D][wifi:366]:     Channel: 9
[16:20:06][D][wifi:367]:     RSSI: -79 dB
[16:20:06][D][wifi:369]: - 'MikroTik-043865' (CC:2D:E0:04:38:65) ▂▄▆█
[16:20:06][D][wifi:369]: - 'STARNET_Mikulica' (E4:8D:8C:8C:CC:BE) ▂▄▆█
[16:20:06][I][wifi:193]: WiFi Connecting to 'Garden WiFi'...
[16:20:12][W][wifi_esp8266:334]: Event: Disconnected ssid='Garden WiFi' reason='Probe Request Unsuccessful'
[16:20:12][W][wifi:451]: Error while connecting to network.
[16:20:12][W][wifi:488]: Restarting WiFi adapter...
[16:20:12][I][wifi:193]: WiFi Connecting to 'Garden WiFi'...
[16:20:18][W][wifi_esp8266:334]: Event: Disconnected ssid='Garden WiFi' reason='Probe Request Unsuccessful'
[16:20:18][W][wifi:451]: Error while connecting to network.
[16:20:18][W][wifi:488]: Restarting WiFi adapter...
[16:20:23][D][wifi:304]: Starting scan...
[16:20:29][D][wifi:319]: Found networks:
[16:20:29][I][wifi:365]: - 'Garden WiFi' (00:26:5A:77:C1:E0) ▂▄▆█
[16:20:29][D][wifi:366]:     Channel: 13
[16:20:29][D][wifi:367]:     RSSI: -71 dB
[16:20:29][I][wifi:365]: - 'Garden WiFi' (CE:2D:E0:04:38:65) ▂▄▆█
[16:20:29][D][wifi:366]:     Channel: 9
[16:20:29][D][wifi:367]:     RSSI: -81 dB
[16:20:29][I][wifi:365]: - 'Garden WiFi' (00:14:BF:3C:F4:B3) ▂▄▆█
[16:20:29][D][wifi:366]:     Channel: 1
[16:20:29][D][wifi:367]:     RSSI: -54 dB
[16:20:29][D][wifi:369]: - 'MikroTik-043865' (CC:2D:E0:04:38:65) ▂▄▆█
[16:20:29][I][wifi:193]: WiFi Connecting to 'Garden WiFi'...
[16:20:35][W][wifi_esp8266:334]: Event: Disconnected ssid='Garden WiFi' reason='Probe Request Unsuccessful'
[16:20:35][W][wifi:451]: Error while connecting to network.
[16:20:35][W][wifi:488]: Restarting WiFi adapter...
[16:20:36][I][wifi:193]: WiFi Connecting to 'Garden WiFi'...
[16:20:41][W][wifi_esp8266:334]: Event: Disconnected ssid='Garden WiFi' reason='Probe Request Unsuccessful'
[16:20:41][W][wifi:451]: Error while connecting to network.
[16:20:41][W][wifi:488]: Restarting WiFi adapter...

```

**Additional information and things you've tried:**

Checked that there is no other device with conflicting IP on my network, also tried to change the IP to different one

also tested with other components such as switch or IR receiver and had no problems to connect to WiFi using the same board and static IP

tried  fast_connect: True (you cen see it commented out in my yaml), same situation, refuse to connect

also as you may notice from the logs I have 3 APs with the same SSID in my house (all connected to the same network, with same passwords), not sure if it can be related to this but never had a problem with ESPs / any other wifi devices

all passwords in the yaml above were changed :-)

BugerDread, 2020-01-01:
It seems that WiFi problems described in my original post are related to following boards only:

[image]

Please note the marking of the "ESP" chip - its not usuall ESP but chip marked as "ECO Plugs B REV 1.1" - **I suspect it to be some kind of ESP8266 clone - maybe not fully compatible with original = cause of thesse issues?**

As I played with esphome more and more I found that nodes are facing constant WiFi connection problems. 

Interesting facts I observed:
- when .yaml is created, compiled, uploaded via USB into module shown on image above, there is like 50-50 chance that there will be connection problems
- there are 3 main lets say types of "connection problems" (+ no problem possibility = option 4)
  1. scan results with "no network found!" and no connection occurs (but there are WiFi networks available as visible in my original post)
  2. scan finds the network but fails to connect with "Event: Disconnected ssid='Garden WiFi' reason='Probe Request Unsuccessful'" - as in my original post
  3. connection is estabilished but then fails within few seconds (but this was rare)
  4. no problem - connection works

- for the same .yaml the result is always the same, means always working ok OR always suffering THE SAME connection problem (like in my original post - always problem 2 occurs - network is found but fail to connect if there is the static IP and DHT sensor), even when module is 
  - restarted
  - power removed / connected back
  - module is erased using "esptool.py erase_flash" and power removed, reflashed
  - added 1000uF elyt and 22uF ceramic cap to modules 3.3V rail
  - used different power source (even 5V 3Amps), different USB cables
  - I have 4 of these (same) "ECO Plugs" modules, results are exactly same for all of them, every time, repeatably

- with different ESP8266 modules than those with "ECO Plugs" chips I had never connection issues (even with .yaml that I know it does not work on "ECO Plugs") - tested with these four modules (all other free ESPs I found at home)
  - nodemcu bought with soldered ESP12 (metallic cover over ESP chip)
  - WiFi kit 8 (visible ESP chip with usuall marking ESP8266EX)
  - two Amica nodemcu boards purchased without ESP12 modules, with two different ESP12 modules soldered on (one is marked with Espressif logo, second one with Ai-Thinker)

- when .yaml is changed in a way that should not affect wifi (like adding / removing DHT sensor), it may change its "WiFi behaviour" - means randomly choosing one from 1-4 options described above and keeps the behaviour no matter of restarts / reflashes / erases

- these "ECO Plugs" modules works absolutely fine when programmed using Arduino IDE - the WiFi connection always works as excpected, Im using them in other projects like NTP clock, bitcoin ticker, changed the Arduino code "milion times" and cant remember single problem with WiFi, even using ESP libz version 2.4.x, 2.5.x, 2.6.x

- here is example of .yaml which always results in "no network found!" on "ECO Plugs" module:

```
esphome:
  name: obyvak_lustr_testing
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "Garden WiFi"
  password: "xxxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Obyvak Lustr Fallback Hotspot"
    password: "xxxxxxxx"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xxxxxxxxx"

ota:
  password: "xxxxxxxxx"

output:
  - platform: gpio
    pin: GPIO13
    id: light_relay
#    inverted: True

light:
  - platform: binary
    name: "Obyvak lustr"
    output: light_relay
    restore_mode: RESTORE_DEFAULT_ON
    id: lustr

remote_receiver:
  pin:
    number: GPIO14
    inverted: yes
    mode: INPUT_PULLUP
  dump: all

binary_sensor:
  - platform: remote_receiver
    name: "IR onoff"
    sony:
      data: 0x00001EE9
      nbits: 15
    filters:
      - delayed_off: 100ms
    on_press:
      then:
        - light.toggle:
            id: lustr
  - platform: remote_receiver
    name: "IR dimm"
    sony:
      data: 0x00006CE9
      nbits: 15
    filters:
      - delayed_off: 100ms
    on_press:
      then:
        - light.turn_off:
            id: lustr
        - delay: 500ms
        - light.turn_on:
            id: lustr
```
  result log output of this .yaml:

> Log initialized
> [19:14:23][C][ota:364]: There have been 0 suspected unsuccessful boot attempts.
> [19:14:23][I][app:028]: Running through setup()...
> [19:14:23][C][light:097]: Setting up light 'Obyvak lustr'...
> [19:14:23][D][light:264]: 'Obyvak lustr' Setting:
> [19:14:23][D][light:273]:   State: ON
> [19:14:23][C][remote_receiver.esp8266:033]: Setting up Remote Receiver...
> [19:14:23][C][wifi:033]: Setting up WiFi...
> [19:14:23][D][wifi:304]: Starting scan...
> [19:14:29][D][wifi:319]: Found networks:
> [19:14:29][D][wifi:321]:   No network found!
> [19:14:34][D][wifi:304]: Starting scan...
> [19:14:40][D][wifi:319]: Found networks:
> [19:14:40][D][wifi:321]:   No network found!
> [19:14:45][D][wifi:304]: Starting scan...
> [19:14:51][D][wifi:319]: Found networks:
> [19:14:51][D][wifi:321]:   No network found!
> [19:14:56][D][wifi:304]: Starting scan...
> [19:15:02][D][wifi:319]: Found networks:
> [19:15:02][D][wifi:321]:   No network found!
> ... and so on, never connects
> 

Is it really caused by incompatible ESP clone? Cant imagine what else can cause such issues - I dont suspect voltage regulator / flash or usb2uart chip to be cause of these WiFi problems. Any ideas?

brandond, 2020-01-08:
Try the dev branch. There's also an undocumented RF power setting that you might try changing, it's meant to work around poorly laid out boards (like this one) where the antenna design is so poor that the wifi transmission actually interferes with signals to other chips on the board.

See: https://github.com/esphome/issues/issues/455#issuecomment-552976310

BugerDread, 2020-01-09:
@brandond you are right! 

I made with very simple RF shield out of aluminium foil (xmas chocolates rulez), insulated it using paper adhesive tape and put it over ESP chip - even without grouding the aluminium foil the obyvak_lustr_testing.yaml (that one which was not able to find any AP) magically started to see APs as usuall and connected without problem. Tried to restart about 5 times and every time connected. When I removed it and restarted the ESP "no neworks found" again... Tried to restart 5x - 5x no networks. Put the shield back to esp, restarted and connected again - so YES, this is hardware problem of the module and it has sth to do with RF interference. 

Maybe it could be worth to make some note about these problems into esphome docs to warn (especially new) users - I wasted like 2 or 3 days trying to figure out "whats the problem with my yaml that I cant connect to WiFi?"

Here are some photos of my experiment:

[image]
[image]
[image]
[image]

Should I close this issue or is there a way to add warning into esphome docs?

brandond, 2020-01-10:
Wow that's incredible. This must be one of the worst boards I've seen. I honestly don't trust anything without an rf shield on it. It doesn't guarantee I'll work well, but at least it should get on the network without having a candy wrapper taped to it!

I've been meaning to update the docs to cover the tx power setting, I might mention your experiment in there somewhere as well.

BugerDread, 2020-01-10:
You are right, since this experiment will never buy these modules again! Its interesting that I never noticed problems when programming using arduino... Thinking about it deeply I remember one project where esp acted as a simple thermostat with web server and posibility to see status / set the temperature via web browser - there were sometimes problems with availability of the device, but the thermostat (temperature regulation) worked always OK - and I never find out why until I replaced ESP.

BugerDread, 2020-01-10:
Maybe I will try to disconnect the integrated antenna (by cutting the PCB trace) and solder the cable from old notebook wifi antenna onto that place (+ ground), will see if it will work or not... Just for curiosity.

BugerDread, 2020-01-20:
I tried to set "output_power" to 10, 15, 17dB but none worked without rf shield, these these boards are s****y

idofr, 2020-06-02:
I'm not sure that the cheap board is the problem here.
I tried running the WiFiClient exmaple in the ArduinoIDE and it works without a problem... I'm digging a bit deeper now

BugerDread, 2020-06-02:
> I'm not sure that the cheap board is the problem here.
> I tried running the WiFiClient exmaple in the ArduinoIDE and it works without a problem... I'm digging a bit deeper now

I was also able to run my other arduino projects / examples on these shitty boards almost without problems, I suggest you to try some better board first instead of wasting time trying to dig the issue. 

Its not just about fixed IP, I found that almost any change in esphome yaml can cause wifi issues on these boards, also changes that have nothing to do with wifi at all.

idofr, 2020-06-02:
I fear you're right.
This is my 8266 board which didn't really feel like doing anything
https://de.aliexpress.com/item/32813713134.html?spm=a2g0s.12269583.0.0.5dc772d77dL6zO

In the same order i had an esp32 board
https://de.aliexpress.com/item/32864722159.html?spm=a2g0s.9042311.0.0.30bb4c4dpxXdzv

I compiled the same simple code for it and it connected out of the box. Now i can start working with it and see what it can do :)

BugerDread, 2020-06-02:
> I fear you're right.
> This is my 8266 board which didn't really feel like doing anything
> https://de.aliexpress.com/item/32813713134.html?spm=a2g0s.12269583.0.0.5dc772d77dL6zO
>
> In the same order i had [...]

yep, the 8266 board on the first aliexpress link seems the same I bought and found problematic...

I suggest you to buy some with the metal RF shield, like: https://de.aliexpress.com/item/32665100123.html

I ordered these: https://www.aliexpress.com/item/32643142716.html
... but still waiting for delivery

RoadkillUK, 2020-07-14:
Hey folks, I have a few different ESP8266 clones and they are pretty random when it comes to connecting ... all of the above is so true, it's ridiculous, I have even put some tin foil (from a cigarette packet) around my ESP8266 and boom, it connects.  So all is not lost eh? :)

BugerDread, 2020-07-16:
I can confirm that D1 mini from Wavgat store connects normally (https://www.aliexpress.com/item/32643142716.html)

> Hey folks, I have a few different ESP8266 clones and they are pretty random when it comes to connecting ... all of the above is so true, it's ridiculous, I have even put some tin foil (from a cigarette packet) around my ESP8266 and boom, it connects. So all is not lost eh? :)

Do you have picture or link of/to these modules, please? To AVOID buying them ...

Aleksahek, 2020-09-25:
Chip is ESP8285. same problem. how to fix?

BugerDread, 2020-09-29:
> esphome:
> name: ${device_name}
> platform: ESP8266
> board: esp8285
> Chip is ESP8285. same problem. how to fix?

ESP8285 should be almost the same as 8266 but have integrated 1M FLASH on the chip (I just googled this, I never had 8285 in my hands). Therefore I expect that there is almost no possibility that there will be interference with the FLASH (as it is built-in / on the chip) - but its just my expectation which may be wrong ofc.
```
Differences between ESP8285 and ESP8266:
ESP8285 integrates 1MB Flash in DOUT mode. While ESP8266 requires external Flash.
ESP8285 has two additional GPIO pins(GPIO9 and GPIO10).
To compile and download ESP8285 firmware, you need to set SPI MODE to DOUT mode, otherwise it can not work even after successfully download
```
What do you exactly mean by "same problem"?
- that it refuse to connect to WiFi
or
- that adding RF shield over the board resolves the WiFi issue

Please try to desctibe it in details to be able to help you.

closed 2021-02-10 by stale[bot]

lucky62, 2021-08-28:
I have the same problem. Just some more observations.
[image]

My board has the MCU normally marked as ESP8266EX.
When software flashed by Arduino IDE wifi is working.
When flashed ESPhome - firstly **No network found!** 
After 10 resets (safe mode) WiFi is recognized but connection is failing with **Probe Request Unsuccessful**.

```
[I][logger:170]: Log initialized
[C][ota:378]: There have been 10 suspected unsuccessful boot attempts.
[E][ota:383]: Boot loop detected. Proceeding to safe mode.
[I][app:029]: Running through setup()...
[C][wifi:037]: Setting up WiFi...
[D][wifi:380]: Starting scan...
[D][wifi:395]: Found networks:
[I][wifi:441]: - 'detox77' (14:CC:20:A9:31:8E) ▂▄▆█
[D][wifi:442]:     Channel: 1
[D][wifi:443]:     RSSI: -49 dB
[D][wifi:445]: - 'MD142' (14:CC:20:3B:2A:1A) ▂▄▆█
[D][wifi:445]: - 'Duchonova' (C4:71:54:32:F9:B4) ▂▄▆█
[D][wifi:445]: - 'rackova' (00:8C:54:F5:F0:B1) ▂▄▆█
[D][wifi:445]: - 'jarkonet' (AC:9E:17:68:19:A0) ▂▄▆█
[D][wifi:445]: - 'JURAJ_Wifi' (50:46:5D:BC:42:80) ▂▄▆█
[D][wifi:445]: - 'hakhak' (40:4A:03:63:F6:1A) ▂▄▆█
[D][wifi:445]: - 'TP-Link_27C6' (40:3F:8C:6C:27:C6) ▂▄▆█
[D][wifi:445]: - 'ELBE doma' (E8:65:D4:8B:23:20) ▂▄▆█
[I][wifi:250]: WiFi Connecting to 'detox77'...
[W][wifi_esp8266:451]: Event: Disconnected ssid='dtx99' reason='Probe Request Unsuccessful'
[W][wifi:547]: Error while connecting to network.
[W][wifi:585]: Restarting WiFi adapter...
[I][wifi:250]: WiFi Connecting to 'detox77'...
[W][wifi_esp8266:451]: Event: Disconnected ssid='dtx99' reason='Probe Request Unsuccessful'
[W][wifi:547]: Error while connecting to network.
[W][wifi:585]: Restarting WiFi adapter...
[D][wifi:380]: Starting scan...

```

Dinevir, 2021-08-29:
This works like a charm!!!
https://github.com/esphome/issues/issues/1185#issuecomment-629695555

I have three boards HW628, one of them works fine with ESPHOME, another two see the networks but "Probe Request Unsuccessful" on connection. And with this config both works fine:
`esphome:
  platform: ESP8266
  board: nodemcuv2
  arduino_version: "espressif8266@1.8.0"`

notabene00, 2021-11-15:
@Dinevir mate, u made my day, thank you so much
I have that problematic board in my intercom and using platform==v1.8.0 it is working fine