esphome/issues #953: PlatformIO 502

paularmstrong, 2019-12-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip3 / python3

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 WROOM

**Affected component:**

All

**Description of problem:**
APIRequestError 502 Bad Gateway

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: kitchen_cabinets
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "xxx"
  password: "yyy"

logger:

api:

ota:

```

**Logs (if applicable):**

```
INFO Reading configuration lights/cabinet-lights.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d lights/kitchen_cabinets
Processing kitchen_cabinets (board: nodemcu-32s; framework: arduino; platform: espressif32@1.11.0)
--------------------------------------------------------------------------------
CorePackageManager: Installing tool-scons @ ~3.30101.0
tool-scons @ 3.30101.0 has been successfully installed!
Uninstalling tool-scons @ 2.20501.191222:   [OK]
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Looking for AsyncTCP-esphome library in registry
LibraryManager: Installing id=6798 @ 1.1.1
APIRequestError: [API] Invalid response: b'<html>\r\n<head><title>502 Bad Gateway</title></head>\r\n<body bgcolor="white">\r\n<center><h1>502 Bad Gateway</h1></center>\r\n<hr><center>nginx</center>\r\n</body>\r\n</html>\r\n':
  File "$HOME/Library/Python/3.7/lib/python/site-packages/platformio/builder/main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "$HOME/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "$HOME/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "$HOME/.platformio/platforms/espressif32/builder/main.py", line 209:
    target_elf = env.BuildProgram()
  File "$HOME/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "$HOME/Library/Python/3.7/lib/python/site-packages/platformio/builder/tools/platformio.py", line 122:
    _build_project_deps(env)
  File "$HOME/Library/Python/3.7/lib/python/site-packages/platformio/builder/tools/platformio.py", line 47:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "$HOME/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "$HOME/Library/Python/3.7/lib/python/site-packages/platformio/builder/tools/piolib.py", line 1043:
    project.install_dependencies()
  File "$HOME/Library/Python/3.7/lib/python/site-packages/platformio/builder/tools/piolib.py", line 878:
    lm.install(uri)
  File "$HOME/Library/Python/3.7/lib/python/site-packages/platformio/managers/lib.py", line 359:
    force=force)
  File "$HOME/Library/Python/3.7/lib/python/site-packages/platformio/managers/package.py", line 703:
    pkg_dir = self._install_from_piorepo(name, requirements)
  File "$HOME/Library/Python/3.7/lib/python/site-packages/platformio/managers/lib.py", line 215:
    cache_valid="30d")
  File "$HOME/Library/Python/3.7/lib/python/site-packages/platformio/util.py", line 361:
    result = _get_api_result(url, params, data)
  File "$HOME/Library/Python/3.7/lib/python/site-packages/platformio/util.py", line 85:
    return func(*args, **kwargs)
  File "$HOME/Library/Python/3.7/lib/python/site-packages/platformio/util.py", line 337:
    r.text.encode("utf-8"))
========================== [FAILED] Took 2.38 seconds ==========================

```

**Additional information and things you've tried:**

BertrumUK, 2019-12-26:
See [948](https://github.com/esphome/issues/issues/948)

paularmstrong, 2019-12-30:
Will close this to clean up, but would be great if there could be a clearer error.

closed 2019-12-30 by paularmstrong