esphome/issues #954: Waveshare E-Paper 2.13in-ttgo not working

morph027, 2019-12-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Python virtualenv pip installation 1.14.3

**ESP (ESP32/ESP8266, Board/Sonoff):**

[TTGO T5 V2.3 ESP32](https://de.aliexpress.com/item/4000348231634.html?spm=a2g0s.9042311.0.0.3c1a4c4ds7uQQi)

[image]

**Affected component:**

https://esphome.io/components/display/waveshare_epaper.html

**Description of problem:**

Display does not updating or showing anything.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: eink2
  platform: ESP32
  board: wemosbat

# wifi:
# <redacted>

time:
  - platform: sntp
    id: sntp_time

spi:
  clk_pin: 18
  mosi_pin: 23

font:
  - file: 'Roboto-Regular.ttf'
    id: roboto

display:
  - platform: waveshare_epaper
    cs_pin: 5
    dc_pin: 17
    busy_pin: 4
    reset_pin: 16
    model: 2.13in-ttgo
    update_interval: 10s
    lambda: |-
      it.strftime(0, 0, id(roboto), "%Y-%m-%d %H:%M", id(sntp_time).now());
```

**Logs (if applicable):**

```
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[19:56:30][I][logger:156]: Log initialized
[19:56:30][VV][preferences:038]: LOAD 0: valid=YES, 0=0x00000001 1=0x42F4583F (Type=233825507, CRC=0x42F4583F)
[19:56:30][C][ota:364]: There have been 1 suspected unsuccessful boot attempts.
[19:56:30][VV][preferences:051]: SAVE 0: 0=0x00000002 1=0x13D89D5A (Type=233825507, CRC=0x13D89D5A)
[19:56:30][I][app:028]: Running through setup()...
[19:56:30][V][app:029]: Sorting components by setup priority...
[19:56:30][C][spi:020]: Setting up SPI bus...
[19:56:30][VV][spi:093]: Enabling SPI Chip on pin 5...
[19:56:30][VV][spi:015]: Disabling SPI Chip on pin 5...
[19:56:30][VV][spi:093]: Enabling SPI Chip on pin 5...
[19:56:30][VV][spi:015]: Disabling SPI Chip on pin 5...
[19:56:30][VV][spi:093]: Enabling SPI Chip on pin 5...
[19:56:30][VV][spi:015]: Disabling SPI Chip on pin 5...
[19:56:30][VV][spi:093]: Enabling SPI Chip on pin 5...
[... 526 lines cut ...]
[19:57:00][VV][spi:093]: Enabling SPI Chip on pin 5...
[19:57:00][VV][spi:015]: Disabling SPI Chip on pin 5...
[19:57:00][VV][spi:093]: Enabling SPI Chip on pin 5...
[19:57:00][VV][spi:015]: Disabling SPI Chip on pin 5...
[19:57:00][VV][spi:093]: Enabling SPI Chip on pin 5...
[19:57:00][VV][spi:015]: Disabling SPI Chip on pin 5...
[19:57:00][VV][spi:093]: Enabling SPI Chip on pin 5...
[19:57:00][VV][spi:015]: Disabling SPI Chip on pin 5...
[19:57:00][VV][spi:093]: Enabling SPI Chip on pin 5...
[19:57:00][VV][spi:015]: Disabling SPI Chip on pin 5...
[19:57:00][VV][spi:093]: Enabling SPI Chip on pin 5...
[19:57:00][VV][spi:015]: Disabling SPI Chip on pin 5...
[19:57:00][VV][spi:093]: Enabling SPI Chip on pin 5...
[19:57:00][VV][spi:015]: Disabling SPI Chip on pin 5...
[19:57:00][VV][spi:093]: Enabling SPI Chip on pin 5...
[19:57:00][VV][spi:015]: Disabling SPI Chip on pin 5...
[19:57:00][VV][spi:093]: Enabling SPI Chip on pin 5...
[19:57:00][VV][spi:015]: Disabling SPI Chip on pin 5...
[19:57:00][VV][spi:093]: Enabling SPI Chip on pin 5...
[19:57:00][VV][spi:015]: Disabling SPI Chip on pin 5...
[19:57:00][VV][spi:093]: Enabling SPI Chip on pin 5...
[19:57:00][VV][spi:015]: Disabling SPI Chip on pin 5...
```

**Additional information and things you've tried:**

morph027, 2019-12-28:
Model `2.13in-ttgo` got introduced in https://github.com/esphome/esphome/pull/730 by @thomasklingbeil ... tried with `2.13in`, still nothing.

thomasklingbeil, 2019-12-28:
There are different versions of the ePaper display used on the TTGO boards. I haven’t found a way to distinguish between them yet.

However, I have already added another display type; just need to push those changes to github.

thomasklingbeil, 2019-12-28:
Can you please give this version a try: https://github.com/esphome/esphome/pull/928
(https://github.com/thomasklingbeil/esphome/tree/ttgo-epaper-b73)?

morph027, 2019-12-28:
[image]

Wow, thank you ;) :tada: 

I also needed this PR to make it build: https://github.com/esphome/esphome/pull/923

morph027, 2019-12-28:
Should i update docs for your PR?

thomasklingbeil, 2019-12-28:
> Should i update docs for your PR?

Thanks for the offer! I have already created a PR for the docs repo.

raberrio, 2020-01-04:
Hi, i am using this TTGO T5 v2.3 2.13, but i can't flash successfully the module with the ESPHOME bin. I tried flashing back the original bin and it worked fine. I tried with different boards but i have not had success. How are u flashing the TTGO T5?

thanks

morph027, 2020-01-04:
I'm using the git branch from @thomasklingbeil (inside a virtualenv)

```
python3 -m virtualenv -p /usr/bin/python3 /tmp/venv
pip install -U git+https://github.com/thomasklingbeil/esphome@ttgo-epaper-b73`
curl -L https://raw.githubusercontent.com/esphome/esphome/da2afdf508a02646ba4d7ccd6bb57b01e7902c8b/esphome/util.py > /tmp/venv/lib/python3.6/site-packages/esphome/util.py
. /tmp/venv/bin/activate
```

raberrio, 2020-01-04:
My problem is that i can't even flash the basic .bin generated by esphome. After flashing the .bin, device is doing absolutely nothing. I can flash back the original .bin and device work OK, so i dont get what i am doing wrong. I am deeply sorry if this is offtopic but i am desperate atm. 

thanks

raberrio, 2020-01-09:
[image]

raberrio, 2020-01-09:
@morph027 how do you upload an image?

morph027, 2020-01-10:
https://esphome.io/components/display/index.html#images

sgvj, 2020-01-10:
Thanks all, this had me stumped.

lucasjbx, 2020-01-19:
Hi
I am using the same TTGO T5 v2.3 2.13 as @morph027 and the git branch from @thomasklingbeil.
The display is not working, everything else does.
Original firmware work well

### This is my .yaml:
```
esphome:
  name: epaper_1
  platform: ESP32
  board: esp32dev

wifi:
  ssid: 'virus!'
  password: '***'
  manual_ip:
    static_ip: 192.168.1.62
    gateway: 192.168.1.254
    subnet: 255.255.255.0

#api:

logger:

ota:

binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO39
      mode: INPUT_PULLUP
    name: "eink_sw1"
    id: eink_sw1
    filters:
        - delayed_on: 5ms
        - invert: 
    on_click:
      min_length: 50ms
      max_length: 350ms
      then:
        - if:
            condition:
              - light.is_off:
                  id: eink_led
            then:
              - light.turn_on:
                  id: eink_led
                  brightness: 80%
            else:
              - light.turn_off: eink_led
              
              
        
output:
  - platform: ledc
    pin: GPIO19
    id: gpio_19

light:
  - platform: monochromatic
    output: gpio_19
    name: "eink_led"
    id: eink_led    

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23

font:
  - file: 'Roboto-Regular.ttf'
    id: roboto

display:
  - platform: waveshare_epaper
    dc_pin: GPIO17
    cs_pin: GPIO5
    busy_pin: GPIO4
    reset_pin: GPIO16
    rotation: 180°
    model: 2.13in-ttgo
    update_interval: 10s
    id: my_display
    lambda: |-
       it.print(0, 0, id(roboto), "Hello World!");
```

**Log:**

```
Using 'COM4' as serial port.
Showing logs:
[19:36:30]ets Jun  8 2016 00:22:57
[19:36:30]
[19:36:30]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:36:30]configsip: 0, SPIWP:0xee
[19:36:30]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[19:36:30]mode:DIO, clock div:2
[19:36:30]load:0x3fff0018,len:4
[19:36:30]load:0x3fff001c,len:1044
[19:36:30]load:0x40078000,len:8896
[19:36:30]load:0x40080400,len:5828
[19:36:30]entry 0x400806ac
[19:36:30][I][logger:156]: Log initialized
[19:36:30][C][ota:364]: There have been 1 suspected unsuccessful boot attempts.
[19:36:30][I][app:028]: Running through setup()...
[19:36:30][C][spi:020]: Setting up SPI bus...
[19:36:30][C][wifi:033]: Setting up WiFi...
[19:36:31][D][wifi:304]: Starting scan...
[19:36:32][D][wifi:319]: Found networks:
[19:36:32][I][wifi:365]: - 'virus!' (B0:6E:BF:7C:EF:C0) ▂▄▆█
[19:36:32][D][wifi:366]:     Channel: 9
[19:36:32][D][wifi:367]:     RSSI: -64 dB
[19:36:32][D][wifi:369]: - 'dlink' (00:24:01:CA:BF:46) ▂▄▆█
[19:36:32][I][wifi:193]: WiFi Connecting to 'virus!'...
[19:36:33][I][wifi:423]: WiFi Connected!
[19:36:33][C][wifi:283]:   SSID: 'virus!'
[19:36:33][C][wifi:284]:   IP Address: 192.168.1.62
[19:36:33][C][wifi:286]:   BSSID: B0:6E:BF:7C:EF:C0
[19:36:33][C][wifi:287]:   Hostname: 'eink_1'
[19:36:33][C][wifi:291]:   Signal strength: -65 dB ▂▄▆█
[19:36:33][C][wifi:295]:   Channel: 9
[19:36:33][C][wifi:296]:   Subnet: 255.255.255.0
[19:36:33][C][wifi:297]:   Gateway: 192.168.1.1
[19:36:33][C][wifi:298]:   DNS1: 0.0.0.0
[19:36:33][C][wifi:299]:   DNS2: 0.0.0.0
[19:36:33][C][ota:029]: Over-The-Air Updates:
[19:36:33][C][ota:030]:   Address: 192.168.1.62:3232
[19:36:33][C][sntp:017]: Setting up SNTP...
[19:36:33][I][app:058]: setup() finished successfully!
[19:36:33][I][app:101]: ESPHome version 1.15.0-dev compiled on Jan 19 2020, 16:14:11
[19:36:33][C][wifi:415]: WiFi:
[19:36:33][C][wifi:283]:   SSID: 'virus!'
[19:36:33][C][wifi:284]:   IP Address: 192.168.1.62
[19:36:33][C][wifi:286]:   BSSID: B0:6E:BF:7C:EF:C0
[19:36:33][C][wifi:287]:   Hostname: 'eink_1'
[19:36:33][C][wifi:291]:   Signal strength: -62 dB ▂▄▆█
[19:36:33][C][wifi:295]:   Channel: 9
[19:36:33][C][wifi:296]:   Subnet: 255.255.255.0
[19:36:33][C][wifi:297]:   Gateway: 192.168.1.1
[19:36:33][C][wifi:298]:   DNS1: 0.0.0.0
[19:36:33][C][wifi:299]:   DNS2: 0.0.0.0
[19:36:33][C][spi:079]: SPI bus:
[19:36:33][C][spi:080]:   CLK Pin: GPIO18 (Mode: OUTPUT)
[19:36:33][C][spi:082]:   MOSI Pin: GPIO23 (Mode: OUTPUT)
[19:36:33][C][spi:083]:   Using HW SPI: YES
[19:36:33][C][logger:175]: Logger:
[19:36:33][C][logger:176]:   Level: DEBUG
[19:36:33][C][logger:177]:   Log Baud Rate: 115200
[19:36:33][C][logger:178]:   Hardware UART: UART0
[19:36:33][C][waveshare_epaper:183]: Waveshare E-Paper
[19:36:33][C][waveshare_epaper:183]:   Rotations: 0 °
[19:36:33][C][waveshare_epaper:183]:   Dimensions: 128px x 250px
[19:36:33][C][waveshare_epaper:192]:   Model: 2.13in (TTGO)
[19:36:33][C][waveshare_epaper:201]:   Full Update Every: 30
[19:36:33][C][waveshare_epaper:202]:   Reset Pin: GPIO16 (Mode: OUTPUT)
[19:36:33][C][waveshare_epaper:203]:   DC Pin: GPIO17 (Mode: OUTPUT)
[19:36:33][C][waveshare_epaper:204]:   Busy Pin: GPIO4 (Mode: INPUT)
[19:36:33][C][waveshare_epaper:205]:   Update Interval: 10.0s
[19:36:33][C][ota:029]: Over-The-Air Updates:
[19:36:33][C][ota:030]:   Address: 192.168.1.62:3232
[19:36:33][C][sntp:043]: SNTP Time:
[19:36:33][C][sntp:044]:   Server 1: '0.pool.ntp.org'
[19:36:33][C][sntp:045]:   Server 2: '1.pool.ntp.org'
[19:36:33][C][sntp:046]:   Server 3: '2.pool.ntp.org'
[19:36:33][C][sntp:047]:   Timezone: 'CET-1CEST-2,M3.5.0/2,M10.4.0/3'`
```

thomasklingbeil, 2020-01-19:
> Hi
> I am using the same TTGO T5 v2.3 2.13 as @morph027 and the git branch from @thomasklingbeil.
> The display is not working, everything else does.
> Original firmware work well

Have you tried using `2.13in-ttgo-b73` as the model?

lucasjbx, 2020-01-19:
Thank you!, that worked.

sgvj, 2020-01-19:
I'm now having issues with my screen not clearing and getting tons of weird and nasty persistent artifacts. I'm using the same configuration and pins as those above.

thomasklingbeil, 2020-01-20:
> I'm now having issues with my screen not clearing and getting tons of weird and nasty persistent artifacts. I'm using the same configuration and pins as those above.

Does it work properly with the firmware that was on it initially? Which exact display type is selected there?

Other idea: Are you maybe using partial updates? The default is to only do a full update every 30 times. Try setting `full_update_every` to `2` and let's see what happens.

sgvj, 2020-01-21:
Thanks @thomasklingbeil. Somehow I missed 'full' so it had never done that. It is still holding on to some artifacts, I'm having it cycle between black and blank every minute with full updates and an update interval for the next 9h hoping for them to clear.

raberrio, 2020-01-21:
2 consecutive simple updates should clear the artifacts.

mcowger, 2020-02-11:
I'm not sure if I'm missing something, but I'm somehow not seeing those boards as options.

When I specify `2.13in-ttgo` it tells me that board doesn't exist, despite the fact that it should:

```
esphome: [source entry_display.yaml:2]
  name: entry_display
  platform: ESP32

  Could not find board '2.13in-ttgo'.
```

The only ttgo boards listed are:

```
 ttgo-lora32-v1, ttgo-t-beam, ttgo-t-watch, ttgo-t1,
```

thomasklingbeil, 2020-02-11:
Use `esp32dev` as board type and `2.13in-ttgo` as the display type.

mcowger, 2020-02-11:
Aha, gotcha.

Thanks - now to see if I can get it to actually take a flash (but thats a
different issue).

Heiner1967, 2020-02-25:
> Have you tried using `2.13in-ttgo-b73` as the model?

How do I do this? Is it already in the esphome Dev or esphome beta installation? Or do I have to manualy overwrite the files you have modified in my installation?

And is there a way to indentify on my Device if b73 is to be used?

morph027, 2020-02-25:
@Heiner1967 : https://github.com/esphome/issues/issues/954#issuecomment-570821772

Heiner1967, 2020-02-25:
@ Morph027 : Thanks but i tried same command in my Terminal window of ssh addon in Hassio - does not work Python3 seems to be a unknown command 

> @Heiner1967 : [#954 (comment)](https://github.com/esphome/issues/issues/954#issuecomment-570821772)

@ Morph027 : Thanks but i tried same command in my Terminal window of ssh addon in Hassio - does not work Python3 seems to be a unknown command

raberrio, 2020-02-25:
Hi @Heiner1967 , i will share step by step what worked for me (ubuntu server 18 LTS):

Login, then:
```
mkdir ttgot5
cd ttgot5
git clone https://github.com/thomasklingbeil/esphome.git
cd esphome
git checkout ttgo-epaper-b73
pip3 install -e .
sudo docker build -t esphome-dev -f docker/Dockerfile .
sudo docker run --rm -v "${PWD}":/config --net=host -it esphome-dev
```
"esphome-dev" is the name of your container for this custom esphome. You can change the name but you cannot repeat names.

good luck

Heiner1967, 2020-03-07:
Hi, firs not work for me, I have no Ubuntu, I use hassio on a PI. I only can install either esphome or the dev or the beta version of it. Nowhere b73 is yet included, how much longer I need to wait to find this solution? Or better moving away from Hassio?

Heiner1967, 2020-03-10:
Hi, I got it.
I installed the esphome Dev version in Hassio and chnaged the configurationline inside to:
               esphome_version: 'thomasklingbeil:ttgo-epaper-b73'
Now  I could compile based on your Version, and it works. 
Yipeeh

Many Thanks for your development, 
I hope it comes soon into the next official Version of ESPHome

closed 2020-03-12 by brandond