esphome/issues #955: cannot migrate to esphome from tasmota 8+

mrpraline, 2019-12-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io raspberry pi 3
**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff S20
**Affected component:**

https://esphome.io/devices/sonoff_s20.html?highlight=s20

**Description of problem:**
If you have tasmota 8+ installed on the S20 device, migrating to esphome by uploding the binary does not work anymore, you have to use a version 7.x or older. Tasmota now apparently does a check on the binary before installing. You either have to downgrade to rev. 7.2 first (if possible), or flash by hardware UART. I suspect this will be true for any Tasmota device. 
 
This is obviuosly not a problem of esphome, but the documentation should make this clear to potential users that want to migrate.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

bingobob, 2020-01-01:
hi, same for me...  trying to convert my sonoff basics from tasmota to ESPHOME OTA, they are running the latest tasmota 8.1.0... i create an esp binary and try to flash it via the tasmota web interface OTA, unfortunately i get "upload failed" "not compatible"... am keen to convert if poss?
this [migrate](https://esphome.io/guides/migrate_sonoff_tasmota.html) makes it look simple but compatibility problem for me. Reflashing by wire will be tricky as they're all installed, some in the roof !

would be great to find a solution?

john-arvid, 2020-01-03:
Had the same problem. The solution for me was to upgrade (downgrade) to tasmota-minimal v7.2. Then I would upgrade to the esphome firmware that I had compiled and downloaded.

(I had to use the tasmota-minimal since the normal tasmota was to big for OTA to handle)

nwesoccer, 2020-01-06:
Same issue, glad to see there will be a coded work around.  For now running `Setoption78 1` in the tasmota console disables capability check and allows for the update to esphome.

Phontana, 2020-01-12:
`Setoption78 1` did not work for me, downgrading to 7.2 did. Thanks!

Thunderbiscuits, 2020-01-16:
> `Setoption78 1` did not work for me, downgrading to 7.2 did. Thanks!

Type it in the console, press enter, reboot/restart device (no need to power cycle), and it worked for me!

Ryan-Hobbies, 2020-01-17:
Thanks - Setoption78 worked for me.  Easy fix.

Gumboot13, 2020-01-21:
Cheers, just downgraded to 7.2 and it worked.

iondulgheru, 2020-03-10:
```Setoption78 1 ``` worked for me

shafr, 2020-04-29:
For me no restart was necessary. 
Just SetOption78 1 and then upload new firmware

poldim, 2020-05-01:
My process:
- Restart didn't work
- `SetOption78 1` then flash didn't work
- `SetOption78 1` then restart then flash didn't work
- Downgrade to [7.2 minimal](https://github.com/arendst/Tasmota/releases/download/v7.2.0/tasmota-minimal.bin) then flash finally worked!

closed 2020-07-15 by OttoWinter

myfingerhurt, 2021-03-28:
Here is what I did.
Sonoff pow R1

- 8.4.1 Full
    -  8.5.1 minimal(OK) 
       -  9.1    minimal(OK) 
            - 9.3.1 Full first try( not compatible)
            - go 9.1 minimal console run `Setoption78 1` even without rebooting.
            - 9.3.1 Full  second try     ( OK)

feutl, 2021-06-24:
I know this is an old issue, but I ran into this today.
Having 9.4 and 9.5 version of tasmota and I am unable to flash the esp binary to it, even with `setoption78 1` being set.
I do not get an error, the upload starts and runs into a connection timeout. After that, tasmota is back.

Any ideas why this happens and how to fix ?

shafr, 2021-06-24:
whats the os you are flashing from? windows / linux / mac? 
are you using browser / mqtt upload?

feutl, 2021-06-24:
Sorry
Windows 10
Firefox
All updated to the latest version.

Gumboot13, 2021-06-24:
> Sorry
> Windows 10
> Firefox / Chrome
> All updated to the latest version.
> 
> Error: ERR_CONNECTION_RESET

Downgrade to 7.2 minimal and try to flash esphome

feutl, 2021-06-24:
@Gumboot13 this is an option, butvmore a workaround instead of a solution.

I would like to help figuring out, why it does notvqork the way it is intendend and documented.
If more people should use esphome, such things should work if stated this way.

shafr, 2021-06-25:
probably minimal flashing first is good solution. i would also recommend
doing 'Reset 6' from console and trying again.

feutl, 2021-06-25:
Ok I found a much easier solution, which worked on all my devices (Shelly 1) running Tasmota 9.x
After I have downloaded the binary, I gzipped it and was able to upload it to different devices without any issues, not extra configuration needed, no reset, not what so ever.
Just upload the gzipped binary via the web interface and *bumm* done.

Perhaps this should be tested by others and documentation updated
https://esphome.io/guides/migrate_sonoff_tasmota.html

shafr, 2021-06-25:
that's from Tasmota 8.2 or 8.4

feutl, 2021-06-25:
the ticket as well as the article reference 8+ which implies 8 or 9 or 10 ....
otherwise it would be referenced as 8.x

therefor I assume that perhaps even with 8.x compressing the bin to a gz could have solved the problem.
For 9 it does, anyhow what is the process to get this tested and the documentation updated if this is a feasable solution?

Didel, 2021-07-04:
I can confirm @feutl 's solution seems to work fine!

I was trying to flash a Sonoff SV that already had Tasmota 9.5 flashed to it. Using the firmware upload from the Tasmota Web interface didn't work, not even after applying `setoption78 1` and restarting the device.

However, running a simple `gzip [esphome-firmware-filename].bin` (from the MacOS command line) and uploading the gzipped ESPHome firmware file in the Tasmota web interface did the trick (although the `setoption78` was still enabled). The file flashed successfully and ESPHome was running as expected!

feutl, 2021-07-05:
@Didel this sounds great, I opened another ticket to adjust documentation at
https://github.com/esphome/issues/issues/2201
backing this could help updating the documentation.

Didel, 2021-07-07:
Just to provide a bit more information: I was able to upgrade 2 other Sonoff Basic devices from Tasmota 9.5.0 to ESPHome using the GZip method. This time, however, I was able to confirm that `setOption78` was actually *turned off*before uploading the gzipped .bin file. That means that for the gzip method, you don’t have to enable `setOption78` first!