esphome/issues #957: restore_state on template switch does not initialize GPIO output

JohnJaspersen, 2019-12-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip installed ESPhome version 1.14.3 on XUbuntu 18.04
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, board: esp01_1m (TUYA device)

**Affected component:**

Template Switch

**Description of problem:**
In the yaml file provided, I try to restore the state of the outputs on power-on. There are 2 outputs: a GPIO output, which drives a relais, and a PWM output, that drives a LED. On power-up, only the LED restores to the previous state, the relais is always off  

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
#GENERAL
esphome:
  name: ${device_name}
  platform: ESP8266
  board: esp01_1m
  esp8266_restore_from_flash: yes
#  on_boot:

<<: !include ../../common/general_setup.yaml

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
      inverted: True
    id: button
    on_press:
      then:
        - switch.toggle: relais_switch

#OUTPUTS
output:
  - id: relais_output
    platform: gpio
    pin: GPIO12

  - id: wifi_led_output
    platform: esp8266_pwm
    inverted: yes 
    pin: GPIO4
    #max_power: 0.5
    min_power: 0.02

switch:
  - id: relais_switch
    name: "output"
    platform: template
    optimistic: yes
    restore_state: yes
    turn_off_action:
      - output.turn_off: wifi_led_output
      - output.turn_off: relais_output
    turn_on_action:
      - output.turn_on: wifi_led_output
      - output.turn_on: relais_output
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
I tried swapping the lines that turn on the LED and the relais in the switch, I removed the line that outputs the LED, I tried to initialize the relais on_boot to either on or off, but all that did not help.

When pressing the button (the binairy sensor) both outputs turn on and off in pair perfectly.

I added  logger.log line in the turn_on_action and turn_off_action. They did not give output on reboot. When removing the line restore_state: yes from the template switch, the LED did also not restore its state, as expected.

glmnet, 2019-12-31:
I'm not 100% sure, but I believe what ESPHome does here is to keep the state of 3 things:
* the switch, as you'd expect
* the two outputs independently.

So all states should be restored for you to have a expected behavior.
When it restores the switch state, it will not trigger the `turn_*_action`'s as there is really no action happening, the state is treated as an initial state, so if you add logging there, you'll see nothing at boot, no matter the switch restored state.

The guilty guy here points to the gpio output.
I'll take a look at this, as I'd like to have a deeper understanding of how all state is kept on power offs / reboots

glmnet, 2019-12-31:
I't might be related to some timing in the setup / startup, I'd try adding a delay:

````yaml
turn_on_action:
      - output.turn_on: wifi_led_output
      - delay: 1s
      - output.turn_on: relais_output
```` 

Not saying as a final solution but to figure out what might be wrong.

glmnet, 2019-12-31:
Well, figured it out, the issue is as suspected the setup priority order, both gpio output and template switch use the same `setup_priority::HARDWARE` and with this sample the switch is setup first, writes the ON to the output but then later the output setups itself and turns off.
This can easily be fixed by delaying the setup of the template switch [here](https://github.com/esphome/esphome/blob/05f9dede702f959892493bed9d8ae75797710587/esphome/components/template/switch/template_switch.cpp#L39), changing to `setup_priority::DATA`
Let's see what @OttoWinter has to say

As a temporary work-around the previous code fixes it, you can change delay to lower time if needed, e.g. `10ms`

closed 2020-12-05 by stale[bot]