esphome/issues #959: Xiaomi CGG1 - same model, another firmware (again!)

mario-tux, 2019-12-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

mario-tux, 2019-12-29:
The small fix in the PR works with all my 5 CGG1 sensors.

eipporko, 2019-12-29:
I have 4 CGG1 sensors and my problem is that the frequency of the messages is too low, I only have messages each 40 minutes ... Could be this being related with your issue?

mario-tux, 2019-12-30:
> I have 4 CGG1 sensors and my problem is that the frequency of the messages is too low, I only have messages each 40 minutes ... Could be this being related with your issue?

I don't think it is related: you should not receive any message if the sub-model was not supported.

It could be related to distance and/or wifi-interference: try to change the wifi channel or move the sensors. 
Changing the BLE scan window helped me a lot:
```
esp32_ble_tracker:
  scan_parameters:
    window: 300ms
```

eipporko, 2019-12-30:
I tried with that config (window of 300ms) and now works better =) The only problem is that now is disconnecting from the network continuously... I'm gonna try to decrease the value until find a correct one. Thanks !

mario-tux, 2019-12-31:
@eipporko You mean that your ESP disconnect from the wifi connection, right? I'm not facing this in my case because it's an ESP32 unit with ethernet connection. I
See https://github.com/esphome/issues/issues/735 also.

eipporko, 2019-12-31:
That's right, is a problem related with the wifi connection. 
Now I'm trying with this parameters and seems to be more stable
```
esp32_ble_tracker:
  scan_parameters:
    window: 200ms
```
The frequency of the messages improved compared with the default window, but is still disconnecting from the wifi network. 
Even with those problems, it works =)

Alex9779, 2020-01-02:
Seems unfortunately the Chinese change the firmeware every now and then...

In my system I have a window of 120ms now which seems to be a compromise between getting messages from all devices in reasonable time frames and stability, I see some disconnects of all ESP32 devices with BLETracker over the day but just for one minute, sometimes the are on longer than a day before a disconnect occurs. Even with the default values I get those disconnection events. But I also found that 30/40 for window/interval got me the same stability on one device but a lot problems with OTA, see #869.

closed 2020-01-14 by OttoWinter