esphome/issues #961: Binary sensor sometimes does not see state changes

deece, 2019-12-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

https://esphome.io/components/binary_sensor/gpio.html

**Description of problem:**
I have a Sonoff mini on every light circuit in my house. They all have local toggling via GPIO4.

Occasionally, the bathroom and ensuite modules get themselves into a state where local switch toggles do not turn off the relay. Home assistant does not see these state transistions either.

This can be fixed by restarting the module, or by changing the relay state of off via Home Assistant.

Both of these circuits have both a light and a small fan connected. I don't see this problem on other circuits that only have a light.

This problem persists on ESPHome 1.14.3 and earlier versions.

**Problem-relevant YAML-configuration entries:**
```
esphome:
  name: bathroom_light
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "redacted"
  password: "redacted"
  fast_connect: true
  domain: .theloft

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "redacted"

ota:
  password: "redacted"

status_led:
  pin:
    number: GPIO13
    inverted: true

binary_sensor:
  - platform: gpio
    pin: GPIO00
    id: reset
    internal: true
    filters:
      - invert:
      - delayed_off: 10ms
    on_press:
      - switch.toggle:
          id: relay_1

  - platform: gpio
    name: "Bathroom Light Switch"
    pin: GPIO04
    id: switch_1
    on_press:
      then:
        - switch.toggle:
            id: relay_1
    on_release:
      then:
        - switch.toggle:
            id: relay_1

switch:
  - platform: gpio
    name: "Bathroom Light"
    icon: "mdi: lightbulb_outline"
    pin: GPIO12
    id: relay_1
    restore_mode: restore_default_off
```

**Logs (if applicable):**

**Additional information and things you've tried:**

cvdabbeele, 2019-12-30:
Hi,
I see the same issue on an ESP32cam

**_Additional information and things you've tried:_**

- I do see the state change events passing-by in the OTA logs (as indicated below), but Home-Assistant sometimes misses one and leaves the sensor in the wrong state
- I tried with a more powerful power supply and 
- I also tried with a reduced max_framerate to 1fps (in an attempt to reduce the noise)

**_Versions:_**
ESPHome 1.14.3 on Hass.io (Home Assistant 0.103.5)

**_Logs (if applicable):_**
``

[22:48:57][D][esp32_camera:159]:Got Image: len=15319
[22:48:57][D][binary_sensor:036]: 'westcam2_pir': Sending state ON
[22:48:57][D][esp32_camera:159]: Got Image: len=16460
[22:48:58][D][esp32_camera:159]: Got Image: len=16676
.../... 
[22:49:02][D][binary_sensor:036]: 'westcam2_pir': Sending state OFF
.../... 
[22:49:02][D][esp32_camera:159]: Got Image: len=18219
[22:49:02][D][esp32_camera:159]: Got Image: len=18195
[22:49:03][D][esp32_camera:159]: Got Image: len=18206
.../... 
[22:49:09][D][binary_sensor:036]: 'westcam2_pir': Sending state ON
[22:49:10][D][esp32_camera:159]: Got Image: len=15445
[22:49:10][D][esp32_camera:159]: Got Image: len=14725
``

cvdabbeele, 2020-02-03:
continuing on the above...
In the trace in the ESPhome, for the ESP32cam, I see a line mentioning the state change of a sensor that is connected to the ESP32cam
But in the States screen on the Hassio, the state of that sensor  is not changed

Anyone any tips on what I can try to get a better visibility on the issue?
thanks,
chrisV

deece, 2020-02-12:
It turns out I had a wiring fault that effectively connected the outputs of multiple sonoff minis together, so turning on one would also drive the output of the other on, without activating the relay.

closed 2020-02-12 by deece

cvdabbeele, 2020-02-12:
Please do not close this case, I have uploaded a more detailed description and traces of the logs
This is a real issue
chrisV

deece, 2020-02-12:
It's worth opening a different issue, as you have different symptoms and a different root cause.

cvdabbeele, 2020-02-13:
ok, done, thanks
ChrisV