esphome/issues #962: Web Server Events feed only outputs first 7 states

rudshep, 2019-12-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip on Windows 10

**ESP (ESP32/ESP8266, Board/Sonoff):**

esp01_1m

**Affected component:**

https://esphome.io/components/web_server.html

**Description of problem:**
The Web Server dashboard shows all the sensors in the config but when there are more than 7,
some of the components don't show state in the UI anymore.

The names are there, but the State cell is blank.

It looks like the /events feed only outputs the first 7 states and Sensors take precedence.
In other words, if you add 7 Sensors, Switches and TextSensors and further Sensors get "pushed out".

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: demo_web_server_bug
  comment: Demo the web server bug
  platform: ESP8266
  board: esp01_1m

api:
logger:
ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

web_server:
  port: 80

time:
  - platform: sntp
    id: sntp_time

switch:
  - platform: template
    id: switch1
    name: "Switch 1"

  - platform: template
    id: switch2
    name: "Switch 2"

  - platform: template
    id: switch3
    name: "Switch 3"

text_sensor:
  - platform: wifi_info
    ip_address:
      id: wifi_ip
      name: "IP Address"
    ssid:
      id: wifi_ssid
      name: "SSID"
    bssid:
      id: wifi_bssid
      name: "BSSID"
   
sensor:
  - platform: wifi_signal
    id: wifi_sig
    name: "RSSI"
    update_interval: 60s
  
  # Below are a few dummy sensors to fill up the component count.
  # Notice that when dummy1 is uncommented and included, BSSID no longer shows a value.
  # When dummy2 is included, SSID shows blank.
  # When dummy3 is included, IP Address shows blank.
  # When dummy4 is included, Switch3 no longer shows a state.
  #
  # So, it looks like the Events feed in the web server only outputs the states of the first 7 components and
  # Sensors take precendence over everything else.

  # Uncomment these out one by one to see the others being "pushed out" of the Events feed.
  #- platform: template
  #  id: dummy1
  #  name: "Dummy Sensor 1"

  #- platform: template
  #  id: dummy2
  #  name: "Dummy Sensor 2"

  #- platform: template
  #  id: dummy3
  #  name: "Dummy Sensor 3"

  #- platform: template
  #  id: dummy4
  #  name: "Dummy Sensor 4"
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
This started with a UDP component I am building that casts the state of one sensor over UDP to another node (like a peer-to-peer scenario). I use the web server to monitor the state of the sensors on the various nodes. At first I thought the JSON payload in a TextSensor was breaking the web server (it actually breaks the API client but that's another bug report) but after eliminating all of it, even the most basic of Template sensors will cause the issue as you can see in the YAML example. 

My C++ knowledge is limited but I believe it could be something to do with how the WebServer::setup() function enumerates the components.

jperquin, 2020-01-02:
I am having the same issue.
[image]

nkitanov, 2020-01-04:
Same here - I see the name after first 7, but no state and action. Just noticed it after adding some new sensors. Running on ESP8266 Wemos D1 mini.

jperquin, 2020-01-26:
This only appears to be happening on esp01_1m boards. On a btle_esp32_2 board all entities iterate completely..

rudshep, 2020-01-29:
Just tested on NodeMCU ESP8266 board with 4M and same problem.

JamesT42, 2020-03-02:
I also have this issue on my smartplugs
platform: ESP8266
board: esp8285
[image]

rudshep, 2020-03-07:
Hi,

I found a workaround... kind of. So it looks like either some dependency with ArduinoJSON and/or ESP Async Server. It could also be either API or Logger interferes with the web server but I can't test that because the device is out of space after bringing in the dependencies below.

Steps to "fix" web server:

1. Delete the entire build folder (e.g. demo_web_server_bug). This is important to clear whatever library is clashing.
2. Run with minimal config below, with API and Logger *disabled*.

`esphome:
  name: demo_web_server_bug
  comment: Demo the web server bug
  platform: ESP8266
  #board: nodemcuv2
  board: esp01_1m

ota:
# api:
# logger:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

web_server:
  port: 80

time:
  - platform: sntp
    id: sntp_time

switch:
  - platform: restart
    name: "Restart"

  - platform: template
    id: switch1
    name: "Switch 1"

  - platform: template
    id: switch2
    name: "Switch 2"

  - platform: template
    id: switch3
    name: "Switch 3"

text_sensor:
  - platform: wifi_info
    ip_address:
      id: wifi_ip
      name: "IP Address"
    ssid:
      id: wifi_ssid
      name: "SSID"
    bssid:
      id: wifi_bssid
      name: "BSSID"
   
sensor:
  - platform: wifi_signal
    id: wifi_sig
    name: "RSSI"
    update_interval: 60s
  
  # Below are a few dummy sensors to fill up the component count.
  # Notice that when dummy1 is uncommented and included, BSSID no longer shows a value.
  # When dummy2 is included, SSID shows blank.
  # When dummy3 is included, IP Address shows blank.
  # When dummy4 is included, Switch3 no longer shows a state.
  #
  # So, it looks like the Events feed in the web server only outputs the states of the first 7 components and
  # Sensors take precendence over everything else.

  # Uncomment these out one by one to see the others being "pushed out" of the Events feed.
  - platform: template
    id: dummy1
    name: "Dummy Sensor 1"

  - platform: template
    id: dummy2
    name: "Dummy Sensor 2"

  - platform: template
    id: dummy3
    name: "Dummy Sensor 3"

  - platform: template
    id: dummy4
    name: "Dummy Sensor 4"`

At this point the Web Server exhibits the problem... Some elements are not displayed.

3. Now bring in the following 2 dependencies. Leave API and Logger disabled.

`esphome:
  name: demo_web_server_bug
  comment: Demo the web server bug
  platform: ESP8266
  #board: nodemcuv2
  board: esp01_1m
  platformio_options:
    lib_deps: 
      - ArduinoJson-esphomelib
      - "ESP Async WebServer"
ota:
# api:
# logger:`

Now the web server shows all the elements and buttons as it should.

4. Bring in API and Logger... I couldn't because it is now too big for a 1M device and says "not enough space". From about 39.2% to 48.7%.

I just noticed from the build log that step 1 uses ESPAsyncWebServer-esphome @ 1.2.6 while step 3 brings in ESPAsyncWebServer-esphome @ 1.2.3 . Looks like 1.2.3 worked in the past - but was too big and 1.2.6 is smaller but has this bug.

Guess the only thing to do is dig into the differences between 1.2.3 and 1.2.6 and see where things went wonky.

glmnet, 2020-03-07:
These libraries are a fork from originals for esphome. So if the main developer makes changes then it does not break ESPHome 

I don’t understand what version 1.2.6 might have a a original lib latest version seems to be 1.2.3

rradar, 2020-03-20:
https://github.com/esphome/issues/issues/806

glmnet, 2020-04-14:
duplicate of #806

closed 2020-04-14 by glmnet