esphome/issues #963: Time outs when controlling light.group of multiple monochromatic lights

poldim, 2020-01-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

- ESPHome in docker, running latest build via `esphome/esphome:dev`
- HA in docker, running latest build via `homeassistant/home-assistant:stable`

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 on Wemos D1 -> PCA9685 -> MOSFET -> LED

**Affected component:**

Monochromatic Light: https://esphome.io/components/light/monochromatic.html
HA Light Group: https://www.home-assistant.io/integrations/light.group/

**Description of problem:**
Crashing / Resetting / Timing out when adjusting light group.  I'm trying to create an automation that dims all 8 channels but having to slow down the delay to 10+ seconds in order to prevent failure.

**Problem-relevant YAML-configuration entries:**
```yaml
i2c:
  sda: D4
  scl: D3
  id: bus_a
  frequency: 500kHz

##########################################
######    SERVO PWM 16 CH OUTPUT    ######
##########################################
pca9685:
  frequency: 1525
output:
- platform: pca9685
  id: 'pca9685_output4'
  channel: 4
- platform: pca9685
  id: 'pca9685_output5'
  channel: 5
- platform: pca9685
  id: 'pca9685_output6'
  channel: 6
- platform: pca9685
  id: 'pca9685_output7'
  channel: 7
- platform: pca9685
  id: 'pca9685_output8'
  channel: 8
- platform: pca9685
  id: 'pca9685_output9'
  channel: 9
- platform: pca9685
  id: 'pca9685_output10'
  channel: 10
- platform: pca9685
  id: 'pca9685_output11'
  channel: 11
light:
- platform: monochromatic
  output: pca9685_output4
  name: "Dining Inner 1"
  default_transition_length: 1s
  restore_mode: RESTORE_DEFAULT_OFF
- platform: monochromatic
  output: pca9685_output5
  name: "Dining Inner 2"
  default_transition_length: 1s
  restore_mode: RESTORE_DEFAULT_OFF
- platform: monochromatic
  output: pca9685_output6
  name: "Dining Inner 3"
  default_transition_length: 1s
  restore_mode: RESTORE_DEFAULT_OFF
- platform: monochromatic
  output: pca9685_output7
  name: "Dining Inner 4"
  default_transition_length: 1s
  restore_mode: RESTORE_DEFAULT_OFF
- platform: monochromatic
  output: pca9685_output8
  name: "Dining Outer 1"
  default_transition_length: 1s
  restore_mode: RESTORE_DEFAULT_OFF
- platform: monochromatic
  output: pca9685_output9
  name: "Dining Outer 2"
  default_transition_length: 1s
  restore_mode: RESTORE_DEFAULT_OFF
- platform: monochromatic
  output: pca9685_output10
  name: "Dining Outer 3"
  default_transition_length: 1s
  restore_mode: RESTORE_DEFAULT_OFF
- platform: monochromatic
  output: pca9685_output11
  name: "Dining Outer 4"
  default_transition_length: 1s
  restore_mode: RESTORE_DEFAULT_OFF

```
**HA Light Group:**
```
light:
  - platform: group
    name: Dining Room
    entities:
      - light.dining_inner_1
      - light.dining_inner_2
      - light.dining_inner_3
      - light.dining_inner_4
      - light.dining_outer_1
      - light.dining_outer_2
      - light.dining_outer_3
      - light.dining_outer_4
```

**Logs (if applicable):**

```
21:03:54][D][light:264]: 'Dining Outer 3' Setting:
[21:03:54][D][light:277]:   Brightness: 34%
[21:03:54][D][light:303]:   Transition Length: 1.0s
[21:03:54][D][light:264]: 'Dining Inner 1' Setting:
[21:03:54][D][light:277]:   Brightness: 34%
[21:03:54][D][light:303]:   Transition Length: 1.0s
[21:03:54][D][light:264]: 'Dining Inner 2' Setting:
[21:03:54][D][light:277]:   Brightness: 34%
[21:03:54][D][light:303]:   Transition Length: 1.0s
[21:03:54][D][light:264]: 'Dining Inner 4' Setting:
[21:03:54][D][light:277]:   Brightness: 34%
[21:03:54][D][light:303]:   Transition Length: 1.0s
[21:03:54][D][light:264]: 'Dining Outer 4' Setting:
[21:03:54][D][light:277]:   Brightness: 34%
[21:03:54][D][light:303]:   Transition Length: 1.0s
[21:03:54][D][light:264]: 'Dining Outer 2' Setting:
[21:03:54][D][light:277]:   Brightness: 34%
[21:03:54][D][light:303]:   Transition Length: 1.0s
[21:03:54](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[21:03:54][D][light:277]:   Brightness: 34%
[21:03:59][V][api.connection:691]: Error: Disconnecting Home Assistant 0.103.5 (10.30.30.100)
[21:03:59][D][api:067]: Disconnecting Home Assistant 0.103.5 (10.30.30.100)
[21:04:03][V][api.connection:586]: Hello from client: 'Home Assistant 0.103.5 (10.30.30.100)'
[21:04:03][D][api.connection:602]: Client 'Home Assistant 0.103.5 (10.30.30.100)' connected successfully!
[21:04:04][D][light:264]: 'LVRC 3' Setting:
[21:04:04][D][light:303]:   Transition Length: 1.0s
```

```
[20:52:25](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
ERROR Error while reading incoming messages: Error while receiving data: [Errno 9] Bad file descriptor
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to esphome_lvrc_controller.local:6053 (10.30.30.162)
INFO Successfully connected to esphome_lvrc_controller.local
[20:52:35][V][api.connection:586]: Hello from client: 'Home Assistant 0.103.5 (10.30.30.100)'
[20:52:35][D][api.connection:602]: Client 'Home Assistant 0.103.5 (10.30.30.100)' connected successfully!

```
**Additional information and things you've tried:**

- When I saw the component loop cycle messages, I set the frequency of the I2C bus and PCA9685 to max frequency.

brandond, 2020-01-04:
There are several other open issues relating to the native API getting overloaded and dropping connections when to much traffic is generated in a short period of time. Not sure when Otto will have time to take a look at it, but MQTT does not seem to have the same issue.

poldim, 2020-01-06:
> There are several other open issues relating to the native API getting overloaded and dropping connections when to much traffic is generated in a short period of time. Not sure when Otto will have time to take a look at it, but MQTT does not seem to have the same issue.

Good to know, thanks!

closed 2020-08-07 by stale[bot]

poldim, 2020-08-14:
@brandond I see this got auto closed but wondering if you've seen/heard of any developments on this.  I searched for API overloading but didn't find much...

glmnet, 2020-08-16:
Maybe related https://github.com/esphome/esphome/pull/1026