esphome/issues #964: Servo starts moving on startup

vks007, 2020-01-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio on Raspbian Buster , Using ESPHome (1.14.3) as an add on

**ESP (ESP32/ESP8266, Board/Sonoff):** 

Wemos - ESP8266
**Affected component:**

https://esphome.io/components/servo.html 

**Description of problem:**
I am trying to run a servo using ESPHome, while the servo is working fine and I am able to control it well, it starts rotating as soon as the ESP boots up. I dont know why is it behaving so. I have even detached the servo on_boot but that does not have any affect on this behavior. 
I can be sure that there is nothing wrong with the servo or the Wemos as I am able to run the servo well using normal Arduino code. I have been running the same using my own code for long and wanted to switch to ESPHome for the curtain project I am working on. 
In the code below I am using one button to flip the direction of the servo and another to stop the servo. Both work well except that on start up the motor starts spinning at a slow speed, I have to flip the stop button to stop that. Things are fine thereafter.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  static_ip: XXX.XXX.XX.XX
  status_gpio: '2'
  btn_up: 'D2'
  btn_stop: 'D1'

esphome:
  name: curtain_test
  platform: ESP8266
  board: d1_mini
  on_boot:
    lambda: |-
      {
        id(curtain_test_servo).detach();
        ESP_LOGI("main","detached servo:");
      }

#    - servo.detach: curtain_test_servo

wifi:
  ssid: !secret primary_ssid
  password: !secret primary_ssid_pswd
  manual_ip:
    static_ip: $static_ip
    gateway: !secret default_gateway
    subnet: !secret subnet_mask

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
#  password: !secret ota_pswd
  
<<: !include include/wifiinfo.yaml

servo:
  - id: curtain_test_servo
    output: pwm_output
    restore: false

# Example output platform
# On ESP32, use ledc output
output:
  - platform: esp8266_pwm
    id: pwm_output
    pin: D0
    frequency: 50 Hz

binary_sensor:
  - platform: gpio
    pin:
      number: $btn_up
      mode: INPUT_PULLUP
    name: "Up"
    id: "curtain_up"
    on_press:
      then:
      - servo.write:
          id: curtain_test_servo
          level: -100.0%
    on_release:
      then:
      - servo.write:
          id: curtain_test_servo
          level: 100.0%

  - platform: gpio
    pin:
      number: $btn_stop
      mode: INPUT_PULLUP
    name: "Stop"
    id: "curtain_stop"
    on_state:
      then:
        lambda: |-
            id(curtain_test_servo).detach();

status_led:
  pin:
    number: $status_gpio
    inverted: yes
    
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
The only difference I see between my Arduino code and ESPHome config is that in Arduino code I detach the servo when I stop it and attach it when I have to move it. In ESPHome one can only detach it and attaching seems to be automatic. Not sure if that's the reason. 
For reference snippets of my Ardunio code:

```
Servo myservo;

void prepCurtainMove()
{
    servoRunning = true;
    //save the current state so that if it crashes before stopping cleanly, we know that it crashed in between running
    state curr_state = {.steps = currentStepPosition, .isRunning = servoRunning, .maxSteps = curtainMaxSteps};
    saveCurrentState(curr_state);
    if(!myservo.attached())
      myservo.attach(MOTOR_PIN);
    lastEncoderTime = millis();//initialize the last encoder time as we're starting to move now.
}

void curtainStop()
{
    if(myservo.attached())
      myservo.detach();
    servoRunning = false;
    delay(1000);//wait for a second for the motor to stop completely
    updateEncoder();//update the final stopped encoder position 
    state curr_state = {.steps = currentStepPosition, .isRunning = servoRunning, .maxSteps = curtainMaxSteps};
    saveCurrentState(curr_state);
    setLED(0);//turn the LED OFF
    DPRINTLN("Curtain stopped at position:");DPRINTLN(String(currentStepPosition));
}

void curtainUp()
{
  if(!btnTop)
  {
    if((currentStepPosition > 0) || crashRecovery)// TO DO do direction check here
    {
      curtainDirectionUp = true;
      prepCurtainMove();
      myservo.write(180);
      lastEncoderTime = millis();//initialize the last encoder time as we're starting to move now.
      DPRINTLN("Curtain is moving up");
    }
    else
      DPRINTLN("Curtain up command ignored as curtain is at the top");
  }
  else
      DPRINTLN("Curtain up command ignored as curtain is at the extreme top");
}
  
void curtainDn()
{
  if((currentStepPosition < curtainMaxSteps) || crashRecovery) // TO DO do direction check here
  {
    curtainDirectionUp = false;
    prepCurtainMove();
    myservo.write(0);
    lastEncoderTime = millis();//initialize the last encoder time as we're starting to move now.
    DPRINTLN("Curtain is moving down");
  }
  else
    DPRINTLN("Curtain down command ignored as curtain is at the bottom");
}

```

glmnet, 2020-01-03:
Well, for what I see on the code this is expected behavior, actually the [code tells the servo to center itself](https://github.com/esphome/esphome/blob/05f9dede702f959892493bed9d8ae75797710587/esphome/components/servo/servo.h#L43) (go to position 0)
I don't know if this is expected behavior though, this is not documented, @OttoWinter is this intended, shouldn't the servo start detached?

if you want to override the initial position you certainly can:

````yaml

  on_boot:
    priority: 400
    then:
      servo.write:
        id: curtain_test_servo
        level: -90.0 %
```` 

there will still be a slight movement on the servo though

vks007, 2020-01-03:
Thanks for the response @glmnet but I cant afford any movement in the servo as I measure the rotation via an encoder to determine the position of the curtain. Also, forgot to mention that this is a continuous rotation servo , so this will never reach position 0 (or any position). 
From my viewpoint, the servoi would only move to 0 if we include restore property for the servo which by default is false.
One other thing is that the servo movement on start up is continuous , slow and a bit jerky. Once I press the btn_up , the movement becomes smooth and faster, so there is definitely something different on start up. @OttoWinter , any thoughts?

glmnet, 2020-01-03:
Yes, all makes sense to me, do the continuous rotation servo have any pot to configure the "center" you can try to match it with the ESPHome's 0 and then it wont turn at the beginning.

So I see two options here:
a) add a `start_position`, or `start_detached` setting
b) remove the initialization to center

I like b better as it's cleaner and may be more user-expected, but that would be a bit of a breaking-change

I'm happy to send a PR for any solution let's see what @OttoWinter has to add

glmnet, 2020-01-03:
By the way, we might have a double-bug here, as restore will not work to restore to the detached state. I will restore by activating the servo to the last non-detached value.

vks007, 2020-01-03:
@glmnet , there is no start position for me on the servo. As it's a continuous rotation one, it has no zero point. 
Where did you get the `start_position` setting? I don't see it anywhere on ESPHome docs or even in the code? Further, that's what I tried to do explicitly in the `on_boot` section, but that didn't help.

ezraholm50, 2020-03-27:
Did you resolve this? I'm facing the exact same problem. Altough i dont go witn an encoder, just measure the time for up and down and calibrate(by hand) it when its off.

vks007, 2020-03-30:
No, I wasn't able to resolve it. I plan to move away from esphome for this project given this limitation.

glmnet, 2020-03-30:
Since @OttoWinter is less active than usual I will take over this one. I believe there must be some way to start properly with the servo detached.

glmnet, 2020-03-30:
> @glmnet , there is no start position for me on the servo. As it's a continuous rotation one, it has no zero point.
> Where did you get the `start_position` setting? I don't see it anywhere on ESPHome [...]

Sorry did not answer this before...

All servos has a zero position. This is the 1500uS center position. There should be a value you give to it so it stops rotating, in your case this value is not 0 so when the servo is centered it actually rotates 
So there must be a value you can write to servo so it stops rotating (not only stop it by detaching it)

vks007, 2020-03-31:
Thanks for responding @glmnet . I am actually using my servo for moving a curtain up and down and I store the last stopped position in the ESP memory. I also have end stops to take care of situations where power goes off midway during operation. 
While I don't fully understand the benefit of what you're proposing but my issues is that the servo moves even before I can do anything in the code. As you can see even before the detach gets executed the servo moves (I think so).

glmnet, 2020-04-03:
there are two ways of stoping a rotation servo to stop it:
a) detach it, as you already know
b) send the center value, this is usually a 1500uS signal, this is not rotate to one side, not rotate to the other side, the center.

Now this center 0 is not aligned with your servo center so it rotates when ESPHome starts up.

I'm gonna modify servo code to start detached anyway... stay tuned

glmnet, 2020-04-03:
It turned out to be just a simple bug I did not see on my first review of the code, check if this works better for you at your earliest convenience.

Thanks

vks007, 2020-04-03:
Thanks @glmnet , I'll try this out over this weekend.
Thanks again!

closed 2020-04-05 by glmnet

vks007, 2020-04-12:
> Thanks @glmnet , I'll try this out over this weekend.
> Thanks again!

I was able to get it working well by your fix. Thanks @glmnet for your help!
For others who may want to try it, I compiled a new image of esphome with the fix glmnet suggested and then uploaded the code to wemos esp. Not the servo does not move on startup. I also didnt have to put any code under on_boot section.
Final code for reference:
`substitutions:
  base_topic: 'home/study/curtain'
  static_ip: 192.168.1.67
  status_gpio: 'D3'
  btn_up: 'D2'
  btn_stop: 'D1'
  servo_gpio: 'D0'

esphome:
  name: study_curtain
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret primary_ssid
  password: !secret primary_ssid_pswd
  manual_ip:
    static_ip: $static_ip
    gateway: !secret default_gateway
    subnet: !secret subnet_mask
# use this entry to force an IP when changing the IP using manual_ip. Once program uploads, comment this line
#  use_address: 192.168.1.12

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
#  password: !secret ota_pswd
  
mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_uname
  password: !secret mqtt_pswd
  id: mqtt_client
  topic_prefix: $base_topic

<<: !include include/wifiinfo.yaml

servo:
  - id: study_curtain_servo
    output: pwm_output

# Example output platform
# On ESP32, use ledc output
output:
  - platform: esp8266_pwm
    id: pwm_output
    pin: $servo_gpio
    frequency: 50 Hz

switch:
  - platform: restart
    name: "Reset Study Curtain"
    id: study_curtain

binary_sensor:
  - platform: gpio
    pin:
      number: $btn_up
      mode: INPUT_PULLUP
    name: "Up"
    id: "curtain_up"
    on_press:
      then:
      - servo.write:
          id: study_curtain_servo
          level: -100.0%

  - platform: gpio
    pin:
      number: $btn_stop
      mode: INPUT_PULLUP
    name: "Stop"
    id: "curtain_stop"
    on_press:
      then:
      - servo.detach: study_curtain_servo

status_led:
  pin:
    number: $status_gpio
    inverted: yes
    `