esphome/issues #966: Status binary sensor is unavailable when MQTT availability reporting is disabled

ofalvai, 2020-01-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
pip, macOS

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32

**Affected component:**
https://esphome.io/components/binary_sensor/status.html
https://esphome.io/components/mqtt.html

**Description of problem:**
I'm trying to use the status binary sensor through MQTT, with Home Assistant discovery. However, I don't want every entity to be marked as unavailable when the devices goes to sleep, so I disabled the birth and last will messages as per the documentation. I just want one sensor to tell me the availability, but with this configuration the status sensor is always "Disconnected". I think the problem is the discovery MQTT message and its empty payload attributes:
```
{
    "device_class": "connectivity",
    "payload_on": "",
    "payload_off": "",
    "name": "Mailbox camera status",
    "state_topic": "mailbox_camera/status",
    "availability_topic": "mailbox_camera/status",
    "unique_id": "ESPbinary_sensormailbox_camera_status",
    "device": {...}
}
```

**Problem-relevant YAML-configuration entries:**
```yaml
mqtt:
  broker: 192.168.100.104
  username: ...
  password: ...
  birth_message:
  will_message:

binary_sensor:
  - platform: status
    name: "Mailbox camera status"
```

**Additional information and things you've tried:**
When the birth and last will messages are re-enabled the status sensor works again.

BugerDread, 2020-01-05:
I tried to make DHT humidity sensor which enters deepsleep as soon adata are sent and I forced the status to be always (even the battery dies) online using:
```
mqtt:
  broker: 192.168.88.226
  port: "8884"
  username: "1234"
  password: "5678"
  birth_message:
    topic: nodesensormq/status
    payload: online
  will_message:
    topic: nodesensormq/status
    payload: online
```
nodesensormq is name of the device as defined in "esphome:" section

alternatively you may try to use:
```
mqtt:
  broker: 192.168.88.226
  port: "8884"
  username: "1234"
  password: "5678"
  keepalive: 90s
```
this cause that mosquitto (my mqtt broker) send the last will (= offline) message after 90second it cant ping the device, but my device is configured to deepsleep 60s, therefore is reported online until battery dies and 90s pass

closed 2020-12-06 by stale[bot]

N-Schaef, 2021-05-11:
I just encountered the same issue with a very similar config.

@ofalvai did you ever find a solution?

ofalvai, 2021-05-11:
@N-Schaef not really. I'm not using the deep sleep component anymore, so I don't know what's the current status of this issue.