esphome/issues #967: Endstop Cover showing wrong state

larsenglund, 2020-01-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io (Home Assistant 0.103.0, HassOS 2.12, ESPHome 1.14.3) on raspberry pi 3

**ESP (ESP32/ESP8266, Board/Sonoff):**

platform: ESP32, board: esp32dev

**Affected component:**

https://esphome.io/components/cover/endstop.html

**Description of problem:**

I have a endstop cover with two limit switches. When I power on the ESP with the door closed the limit switch for door closed is on but the cover entity in the gui shows the open button as grayed-out and the stop and close button as available.

[image]

I would expect it to be the opposite.

Also the gui does not update when I manually open or close the door via the physical switches on the garage door opener. I would expect only stop to be active when none of the endstops are on, and only open or close when the opposite endstop is on.

**Problem-relevant YAML-configuration entries:**
```yaml
switch:
  - platform: gpio
    pin: 
      number: GPIO14
      inverted: True
    id: door_open
    interlock: [door_close]
    restore_mode: ALWAYS_OFF
    name: Garage Door Open
    
  - platform: gpio
    pin: 
      number: GPIO0
      inverted: True
    id: door_close
    interlock: [door_open]
    restore_mode: ALWAYS_OFF
    name: Garage Door Close

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO2
      mode: INPUT_PULLUP
      inverted: True
    name: Garage Door Open Limit Switch
    id: open_endstop_binary_sensor
    filters:
      - delayed_on_off: 100ms
      
  - platform: gpio
    pin:
      number: GPIO12
      mode: INPUT_PULLUP
      inverted: True
    name: Garage Door Closed Limit Switch
    id: close_endstop_binary_sensor
    filters:
      - delayed_on_off: 100ms

cover:
  - platform: endstop
    device_class: garage
    name: "Garage Door"

    open_action:
      - switch.turn_on: door_open
      - delay: 500ms
      - switch.turn_off: door_open
    open_duration: 9s
    open_endstop: open_endstop_binary_sensor

    close_action:
      - switch.turn_on: door_close
      - delay: 500ms
      - switch.turn_off: door_close
    close_duration: 9s
    close_endstop: close_endstop_binary_sensor
    
    max_duration: 15s

    stop_action:
      - switch.turn_off: door_open
      - switch.turn_off: door_close
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

ZARichardWright, 2020-01-15:
Hi I have am identical set up and have the same issue, i have tried to rectify this by adding the following to each of the end stops 
```
  on_state:
    if:
      condition:
        binary_sensor.is_on: "close_endstop_binary_sensor"
      then:
        - lambda: id(mcover).publish_state(COVER_CLOSED);
```
How ever this does not update the state and leave the position in the position it was previously set via esphome.

I then tried adding the following to the lamda
```yaml
auto call = id(cover_1).make_call();
// set attributes
call.set_position(0);
call.perform();
```

this how ever caused the associated switch to be triggered if the stores position is not same as the ones i set it to.

I will see if i can not get a template cover to give me the desired results

davidsat, 2020-01-16:
Hi,
I have the same issue. My code is pretty much identical to those above. I also see that after a while (I should measure the exact time), the garage door triggers and open/close again. Is there some kind of retry built into the endstop cover?

ZARichardWright, 2020-01-16:
Hi

I think the issue is when the code trys to set the position to 0 and was previously 100 or vice versa, then esp home triggers the relay (x seconds) after door closed and contact has been denounced. This does atleast ensure the cover stare is always correct but has the undesirable effect of half( opening or closing) door.  If you just set the state property it does not seem to update for some reason. 

Cheers.

svh1985, 2020-07-01:
Same issue here, it also does not update the state when externally controlled.

closed 2020-11-12 by stale[bot]