esphome/issues #968: PMS7003 on software UART causes WDT on esp8266 (wemos d1 and nodemcu)

nkitanov, 2020-01-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

hassio
**ESP (ESP32/ESP8266, Board/Sonoff):**

esp8266 on Wemos D1 and Nodemcu
**Affected component:**

https://esphome.io/components/sensor/pmsx003.html

**Description of problem:**
When running on software UART PMSX7003 sensor causes WDT on random interval about 5-20 min. 

**Problem-relevant YAML-configuration entries:**
```yaml
uart:
  - rx_pin: D3
    baud_rate: 9600
    id: pms

sensor:
    - platform: pmsx003
      type: PMSX003
      uart_id: pms
      pm_1_0:
        name: "PM 1 Concentration"
        id: pm1
      pm_2_5:
        name: "PM 2.5 Concentration"
        id: pm2_5
      pm_10_0:
        name: "PM 10 Concentration"
        id: pm10
```

**Logs (if applicable):**

```
[01:53:01][D][debug:023]: ESPHome version 1.14.3
[01:53:01][D][debug:025]: Free Heap Size: 24952 bytes
[01:53:01][D][debug:053]: Flash Chip: Size=4096kB Speed=40MHz Mode=DOUT
[01:53:01][D][debug:190]: Chip ID: 0x0053D585
[01:53:01][D][debug:191]: SDK Version: 2.2.1(cfd48f3)
[01:53:01][D][debug:192]: Core Version: 2_5_2
[01:53:01][D][debug:193]: Boot Version=31 Mode=1
[01:53:01][D][debug:194]: CPU Frequency: 80
[01:53:01][D][debug:195]: Flash Chip ID=0x00164020
[01:53:01][D][debug:196]: Reset Reason: Hardware Watchdog
[01:53:01][D][debug:197]: Reset Info: Fatal exception:4 flag:1 (WDT) epc1:0x401033d9 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
```

**Additional information and things you've tried:**
To rule out hardware issues I tried two different PMS7003 sensors and also on nodemcu and different pin there. All the same - got WDT frequently. 

Also tried running PMS sensor on the RX hardware uart pin which work after disabling UART logging. No issues then - it works days without WDT, so I guess it's linked to the software bit banging of the uart data. PMS7003 is constantly sending data over the uart and maybe it's not stable when it's not using dedicated hardware.

akomelj, 2020-03-19:
I am having the same issue. PMSA003 & software UART. Been testing with default and latest Arduino framework version and different platformio platform versions without any luck.

akomelj, 2020-03-21:
> I am having the same issue. PMSA003 & software UART. Been testing with default and latest Arduino framework version and different platformio platform versions without any luck.

A small update - I changed ESPHome log level to INFO (to avoid any log message publishing during PM readings) and changed PMSA003 operation mode to 1 minute of readings, 10 minutes off, 1 minute of readings (via setting PMS sensor SET PIN low/high) and added debounce filter to PM sensor readings to publish just the last sensor values after that whole minute of readings (sensor outputs stabilise after roughly 20 seconds but I wanted to err on the safe side).

So far, 35 hours uptime. :-D

nkitanov, 2020-03-27:
Yes, that's an option but I need the data to be displayed constantly. That's why I use the hardware pin and there are no issues. However on a software UART I connected MH-Z19 CO2 sensor to update on 30s interval and I get WDT from time to time. Definitely the issue is coming from software uart library...

akomelj, 2020-03-27:
Hi, yes, of course - there's ton of reports about issues with ESP8266 software serial and WDT resets. I'm just testing various "hacks" because I soldered and hot-glued my wires to ESP and am too lazy to do it again. :-)

Regarding constant updates - I've been running PMSA003 in continuous sampling mode for a few days now without any resets. I'm obviously not arguing that DEBUG logging causes the resets; it just considerably lowers the possibility of lock-ups in my particular case.

closed 2020-08-08 by stale[bot]