esphome/issues #970: OTA flashback error ESP8266

Rustam2000G, 2020-01-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io, HassOS 2.12 Deployment	production, Home Assistant 0.102.3, Raspberry Pi 3 Model B+ 32bit, ESPHome 1.14.3

**ESP (ESP32/ESP8266, Board/Sonoff):**

platform: ESP8266, board: esp_wroom_02, WiFi module TYWE3S https://docs.tuya.com/docDetail?code=K8uhkbb1ihp8u , https://aliexpress.ru/item/32903293047.html WiFi

**Affected component:**

**Description of problem:**
Purchased this nozzle on the tap. Passed it with taya-convert https://github.com/ct-Open-Source/tuya-convert. the Nozzle is connected to the Home Assistant, the automations are triggered. Decided to add an additional uptime sensor. The code is checked and compiled without errors. When you try to flash the OTA software through ESP Home Home Assistant flashing does not go further than 2-3% with the result ERROR Error sending data: [Errno 104] Connection reset by peer. Information from a log I put.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: valve
  platform: ESP8266
  board: esp_wroom_02

wifi:
  ssid: "XXXXXXXXXX"
  password: "XXXXXXXXXXX"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

status_led:
  pin: 
    number: 5
    inverted: false

binary_sensor:
  - platform: gpio
    pin: 13
    name: valve_button
    id: valve_button
    on_release:
      then:
      - switch.toggle: valve

switch:
  - platform: gpio
    pin: 12
    id: valve
    name: valve
    icon: "mdi:water"
    restore_mode: RESTORE_DEFAULT_ON
    on_turn_on:
      then:
      - output.turn_on: valve_led
    on_turn_off:
      then:
      - output.turn_off: valve_led    

output:
  - platform: gpio
    pin:
      number: 4
      inverted: true      
    id: valve_led

sensor:
  - platform: wifi_signal
    name: "valve_wifi_signal"
    update_interval: 60s
  - platform: uptime
    name: valve_Uptime

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
[00:08:26][D][ota:072]: Starting OTA Update from 192.168.1.78…
[00:08:26][D][ota:243]: OTA in progress: 0.1%
[00:08:26][W][ota:233]: Error writing binary data to flash: 0 != 1024!
[00:08:26][W][ota:276]: Update end failed! Error: ERROR[2]: Flash Erase Failed

[00:14:52][D][ota:072]: Starting OTA Update from 192.168.1.78…
[00:14:52][D][ota:243]: OTA in progress: 0.1%
[00:14:52][W][ota:233]: Error writing binary data to flash: 0 != 530!
[00:14:52][W][ota:276]: Update end failed! Error: ERROR[2]: Flash Erase Failed

[00:19:01][D][ota:072]: Starting OTA Update from 192.168.1.78…
[00:19:01][D][ota:243]: OTA in progress: 0.1%
[00:19:01][W][ota:233]: Error writing binary data to flash: 0 != 578!
[00:19:01][W][ota:276]: Update end failed! Error: ERROR[2]: Flash Erase Failed

```

**Additional information and things you've tried:**

mrlemon46, 2020-01-05:
Same issue for me.

I flashed a Geeni Power Strip which has a TYWE3S chip, took it apart and soldered wires on to flash the initial .bin file and put it back together.  Now it is in Home Assistant and online in ESPHome but OTA flashes fail with the same Error 104.  HA and ESPHome are both up to date.

Some Google results say Error 104 is related to not having enough memory but a TyWE3S has 1Mbyte flash, so should be good enough?

Rustam2000G, 2020-01-06:
> Same issue for me.
>
> I flashed a Geeni Power Strip which has a TYWE3S chip, took it apart and soldered wires on to flash the initial .bin file and put it back together. Now it is in Home Assistant [...]

I also thought about my lack of memory. But according to the data https://docs.tuya.com/docDetail?code=K8uhkbb1ihp8u the module has 2 Mbyte of memory, and the file size is much lower than 1 Mbyte. You and the firmware through the USB-UART issues cause a error?

mrlemon46, 2020-01-06:
I compiled the binary in ESPHOME and downloaded.  Flashed through ESPHOME Flasher with USB-UART, that part worked fine.  I just can’t do OTA flashes.  I might try to redo it later today.

brandond, 2020-01-06:
`esp_wroom_02` is configured for 2m flash; I find that these usually have only 1m flash. Tuya devices usually work fine for me using the esp01_1m board definition:
https://docs.platformio.org/en/latest/boards/espressif8266/esp01_1m.html

mrlemon46, 2020-01-07:
I took my power strip apart, re-compiled the binary, reflashed with ESPHOME and USB-UART.

OTA updates now work!  I used board definition ( board: esp01_1m )like brandond mentioned, I must have picked something else before.  Good luck!

Rustam2000G, 2020-01-07:
Thanks for the recommendations! By OTA firmware with board: asp 01_1m does not accept with the same errors. What is the possibility to flash without USB-UART? Damaged the GPIO0 pad when soldering ((but the nozzle on the tap remained operational

brandond, 2020-01-07:
If you can't flash via UART then you're stuck with OTA. If that doesn't work due to an initial bad flash load, you might be out of luck.

Rustam2000G, 2020-01-10:
Soldered the relay from the Board. I found the tip of the metallization at the output of GPIO0 from THE end of the tywe3s textolite, and managed to solder to it. Flashed using USB-UART and ESP Home with the module parameter asp01_1m. OTA earned, flies with a whistle.
Thank you all again!

jaisgit, 2020-02-19:
Same issue for me using a WEMOS D1 mini Pro.
Uploads via USB serial works every time but OTA never completes.
Tried minimal config as well without an sensors coded or connected etc...

Could this a problem with the board selection setting maybe? Any suggestions for an alternative board config to try?

```
esphome:
    name: livingroom
    platform: ESP8266
    board: d1_mini_pro
```

> Program failed with code 1

```
INFO Reading configuration /config/esphome/livingroom.yaml...
INFO Detected timezone 'AEST' with UTC offset 10 and daylight savings time from 04 October 02:00:00 to 04 April 03:00:00
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/livingroom
Processing livingroom (board: d1_mini_pro; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 16MB Flash
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <Wire> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/livingroom/.pioenvs/livingroom/src/main.cpp.o
src/main.cpp:152:3: warning: multi-line comment [-Wcomment]
   //   lambda: !lambda "it.strftime(\"%d.%m.%Y %H:%M\", id(time_now).now());\nit.printf(0,\
   ^
Linking /data/livingroom/.pioenvs/livingroom/firmware.elf
Building /data/livingroom/.pioenvs/livingroom/firmware.bin
Retrieving maximum program size /data/livingroom/.pioenvs/livingroom/firmware.elf
Checking size /data/livingroom/.pioenvs/livingroom/firmware.elf
DATA:    [=====     ]  46.7% (used 38296 bytes from 81920 bytes)
PROGRAM: [====      ]  40.8% (used 426296 bytes from 1044464 bytes)
Creating BIN file "/data/livingroom/.pioenvs/livingroom/firmware.bin" using "/data/livingroom/.pioenvs/livingroom/firmware.elf"
========================= [SUCCESS] Took 17.28 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of livingroom.local
INFO  -> 172.16.0.232
INFO Uploading /data/livingroom/.pioenvs/livingroom/firmware.bin (430448 bytes)
Uploading: [===========                                                 ] 18% 
ERROR Error sending data: timed out
```

OttoWinter, 2020-07-24:
Fixed in https://github.com/esphome/esphome/pull/959

closed 2020-07-24 by OttoWinter