esphome/issues #973: time - sntp not updated with static ip

kekec14, 2020-01-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io - ESPhome 1.14.3

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 - Wemos D1 mini

**Affected component:**

https://esphome.io/components/time.html - sntp
https://esphome.io/components/wifi.html

**Description of problem:**

Due to connectivity problems I set static IP in ESPhome (manual_ip parameter in wifi section) and same on the router. After reboot of the controller, time from sntp server is not right. On the boot, time is 08:00:00 01.01.1970 and after few seconds it is corrected to 09:00:00 01.01.1970 - Right time would be 19:20:25 05.01.2020.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test1
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: XXXXXXX
  password: XXXXXXXX
  #fast_connect: True
  manual_ip:
    static_ip: 192.168.0.173
    gateway: 192.168.0.1
    subnet: 255.255.255.0

logger:

api:

ota:

i2c:
  sda: D2
  scl: D1
  
display:
  - platform: lcd_pcf8574
    dimensions: 20x4
    address: 0x27
    lambda: |-
      it.strftime("%H:%M:%S  %d.%m.%Y", id(sntp_time).now());
    update_interval: 500ms
    id: mydisplay

time:
  - platform: sntp
    id: sntp_time
    servers: pool.ntp.org

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
I tried to comment manual_ip parameter and after that everything works as expected.

fredno, 2020-01-07:
You are missing the nameserver. Your manual config should be something like this.
  manual_ip:
    static_ip: 192.168.0.173
    gateway: 192.168.0.1
    subnet: 255.255.255.0
    dns1: 192.168.0.1

kekec14, 2020-01-08:
Thank you! dns1 parameter was missing.

closed 2020-01-08 by kekec14