esphome/issues #976: Light effect is disabled when turning light off

ryanwinter, 2020-01-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Affected component:**

https://esphome.io/components/light/index.html

**Description of problem:**

**Problem-relevant YAML-configuration entries:**

**Logs (if applicable):**

**Additional information and things you've tried:**
When I turn a light off and then on again, the effect is no longer in effect.

Browsing through the code I found the following which may be responsible:

  // Reason: When user turns off the light in frontend, the effect should also stop
  if (!this->has_flash_() && !this->state_.value_or(v.is_on())) {
    if (this->has_effect_()) {
      ESP_LOGW(TAG, "'%s' - Cannot start an effect when turning off!", name);
      this->effect_.reset();
    } else if (this->parent_->active_effect_index_ != 0) {
      // Auto turn off effect
      this->effect_ = 0;
    }
  }
My expectation is that the effect would still be present when turning the light off and then on again.

ryanwinter, 2020-01-28:
Looked into this a bit. I removed the code from [here](https://github.com/esphome/esphome/blob/2d0d794a9dab9f538d65744fddbe6ecbee6c50e9/esphome/components/light/light_state.cpp#L487). 

However when I turned off the light, it would just automatically turn back on. It seems the effect continues to run, even when the light is off, and my particular effect was a lambda that turned the light on, as I used [this ](https://esphome.io/components/light/index.html#lambda-effect)as my example.

Once I rewrote my lambda, to just use make_call(), instead of turn_on(), the light worked as expected, although the effect continues to run and change the light even though its in the off state.

Kinda of hacky, maybe it would be better to have the ability to pause and resume and effect when the light it turned off/on?

OttoWinter, 2020-07-24:
No the behavior is intended.

As you mentioned in the followup, effects can't be paused, which means that when the light turns off the only option is to also disable the effect.

ryanwinter, 2020-07-24:
So maybe the better option here is to add a pause state to the effect which would be used for when the light is turned off.

I think this would add considerable benefit as would be consistent with expectations of how the device is used.

closed 2020-12-06 by stale[bot]

VoetenK, 2021-01-08:
Wouldn't it be possible to replace effects "NONE" with a user-defined default effect.