esphome/issues #979: moving devices to different subnet

nicoolaro, 2020-01-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on Proxmox (intel NUC)
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 Generic
**Affected component:**

https://esphome.io/components/esphome.html?highlight=use_address#changing-esphome-node-name

**Description of problem:**
All my devices have static manual IP and are working great.
Now , I am trying to move all my devices to different subnet.
I changed static IP's and gateway   i also use "use_address" to point to old one as instructed in the manual. Afterwords i remove old IP in "use_address" and although ESP8266 is working it regenerates new sensors in Home Assistant (usually by adding "_2 " to the name ), and light in ESPHOME dashboard stays RED although logs work.

I am not sure how to approach this issue so HA does not generate double entries. 
Is there a guide i can follow?
**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```
Ex. 
wifi:
  ssid: "******"
  password: !secret IOT_SH158_wifi_password
  manual_ip:
    static_ip: 192.168.20.102
    gateway: 192.168.20.1
    subnet: 255.255.255.0
#  use_address: 192.168.1.48
**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

nicoolaro, 2020-02-01:
For anyone struggling with double entries after node IP/Subnet change:

1. Delete node integration 
2. Delete node YAML file located in ".storage" folder in HASS.IO configuraiton folder. Name will be something like `esphome.4a79ffeec2224eceb924416c8cb1fc2a` (open yaml and verify its correct node)
3. Reboot HA
4. Add new ESPHOME node via ESPHOME INTEGRATION with new IP address.
Sensor names will be the same.

If your router does not support mDNS you can use `"status_use_ping": true` in Esphome addon configuration .

closed 2020-08-07 by stale[bot]

johnboiles, 2021-01-16:
I didn't even need to delete the node integration. Just a restart of HA fixed it for me.