esphome/issues #983: sonoff th10 crash with esphome 1.14.3

aartokke, 2020-01-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

hass.io 0.102.3
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESPHOME 1.14.3
Sonoff TH10 with temperature and humidity sensor

**Affected component:**

https://esphome.io/devices/sonoff.html

**Description of problem:**
3x Sonoff's TH10 with firmware esp 1.14.3 are sometimes during the day unavailable in homeassistant. Uptime is also reset. It looks like it's resetting.
Reading the logs from USB: i'm getting at those times in the log a line with "abort called".

btw: sonoff is used for activating a sirene for 3 seconds

Found a solution in issue #931 which has the exact same problem. So my problem is solved 
https://github.com/esphome/issues/issues/931

i also changed the board type from esp01_1m to esp8285 and adding arduino_version: 2.4.2

esphome:
  name: ${plug_name}
  platform: ESP8266
  board: esp01_1m

esphome:
  name: ${plug_name}
  platform: ESP8266
  board: esp8285
  arduino_version: 2.4.2

In yaml configuration below  i have added fastconnect en reboot_timeout: 0s :  We had some wifi connection issues in the beginning, and because of the time actions (and time-sync) i didn't want a reboot if during the day there was no connection with wifi. (i'm rebooting the sonoff's at night) but those issues are resolved.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  # Plug name
  plug_name: "sonoff_th10_02"
 
  # Plug static IP
  plug_ip: "<--ip address deleted by me-->"
  
  # macadres: <--mac address deleted by me-->
 
esphome:
  name: ${plug_name}
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret esphome_ssid_name
  password: !secret esphome_ssid_password
  
  manual_ip:
    static_ip: ${plug_ip}
    gateway: !secret esphome_wifi_gateway
    subnet: !secret esphome_wifi_subnet
    dns1: !secret esphome_wifi_dns1
    dns2: !secret esphome_wifi_dns2
  fast_connect: True
  #indien geen wifi verbinding wordt de esp niet gereboot.
  #doen we met een automation
  reboot_timeout: 0s
  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${plug_name} Fallback Hotspot
[... 760 lines cut ...]
[09:39:06][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[09:39:06][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[09:39:06][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[09:39:06][V][wifi_signal.sensor:009]:   Unique ID: 'cc50e352dcde-wifisignal'
[09:39:06][C][wifi_info:009]: WifiInfo IPAddress 'sonoff_th10_01 IP'
[09:39:06][V][wifi_info:009]:   Unique ID: 'cc50e352dcde-wifiinfo-ip'
[09:39:06][C][wifi_info:010]: WifiInfo SSID 'sonoff_th10_01 SSID'
[09:39:06][V][wifi_info:010]:   Unique ID: 'cc50e352dcde-wifiinfo-ssid'
[09:39:06][C][wifi_info:011]: WifiInfo BSSID 'sonoff_th10_01 BSSID'
[09:39:06][V][wifi_info:011]:   Unique ID: 'cc50e352dcde-wifiinfo-bssid'
[09:39:08][D][binary_sensor:036]: 'sonoff_th10_01 status': Sending state ON
[09:39:09][V][api.connection:567]: Hello from client: 'Home Assistant 0.102.3 (<--ip address deleted by me-->)'
[09:39:09][D][api.connection:583]: Client 'Home Assistant 0.102.3 (<--ip address deleted by me-->)' connected successfully!
[09:39:09][D][time:029]: Synchronized time: Thu Jan  9 09:39:09 2020
[09:39:15]pm open,type:0 0
[09:39:55][D][dht:048]: Got Temperature=20.0°C Humidity=64.8%
[09:39:55][V][sensor:013]: 'sonoff_th10_01 Temperature': Received new state 20.000000
[09:39:55][D][sensor:092]: 'sonoff_th10_01 Temperature': Sending state 20.00000 °C with 1 decimals of accuracy
[09:39:55][V][sensor:013]: 'sonoff_th10_01 Humidity': Received new state 64.800003
[09:39:55][D][sensor:092]: 'sonoff_th10_01 Humidity': Sending state 64.80000 % with 0 decimals of accuracy

```

**Additional information and things you've tried:**
i solved it with the solution of https://github.com/esphome/issues/issues/931

muchasuerte, 2020-01-26:
Hello @aartokke thank you!!

Your solution has resolved also my issue with my Sonoff TH16, it was very unstable also I was thinking to a bug with the wifi problem, I have another TH16 close to the AP which work fine.

aartokke, 2020-07-08:
same issue #931 is closed en i see it's mentioned in the esphome sonoff docs now to use arduino_version: 2.4.2
therefore closing issue

closed 2020-07-08 by aartokke