esphome/issues #985: AMS AS3935 Franklin Lightning Sensor | error Disturber was detected - try increasing the spike rejection value!

Luc3as, 2020-01-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio on docker, ESPhome 1.14.3
**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos d1 mini
**Affected component:**

https://esphome.io/components/sensor/as3935.html

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
    
as3935_i2c:
  irq_pin: D6
  spike_rejection: 2
  indoor: false

```

**Logs (if applicable):**

```
[14:48:15][I][as3935:041]: Disturber was detected - try increasing the spike rejection value!
[14:48:15][I][as3935:041]: Disturber was detected - try increasing the spike rejection value!
[14:48:15][I][as3935:041]: Disturber was detected - try increasing the spike rejection value!
14:48:22][W][i2c:073]: Received NACK on transmit of data for address 0x03
[14:48:22][W][as3935_i2c:029]: Read failed!
[14:48:15][I][as3935:041]: Disturber was detected - try increasing the spike rejection value!

```

**Additional information and things you've tried:**
I have SDA at D4 and SCL at D5,
This is the same issue as https://github.com/esphome/issues/issues/890
I tried to upload example sketch from this https://github.com/stevemarple/AS3935
With default settings, I just changed i2c pins, I was able to run the sketch and trigger the sensor with piezo from lighter, here is output from sketch, these are the reg settings  from sensor. 
[image]
with Esphome sketch I was not able to trigger the sensor even If I changed indoor to false.

Luc3as, 2020-02-10:
Confirmed, today was nice storm with loads of near strokes, with indoor false parameter it is not throwing errors, but it is not working either

Brianckramer, 2020-02-13:
I tried setting this up and had the same problem, I even tried powering it by a battery in an open field and got the same warnings, no matter what I did for noise floor and spike rejection.

edwardscaleb9, 2020-06-22:
Same for me.  Disturber detected constantly no matter what I put for noise floor, spike rejection, or capacitance.  I'm using an Olimex ESP32-PoE with IRQ on GPIO4.  I wonder if maybe this pin is pulled high by default?  Haven't had a chance to move it around yet.

Luc3as, 2020-06-22:
I contacted supplier from AlieExpress, if they have a clue what could cause this. I got reply from them with some libraries I should try, problem is it is compatible with arduino only, It would need to change some parts of code like ISR routines, which is doable, but to be honest I have no time to test it even with arduino. If someone find a few minutes and spare arduino, I am attaching archive they sent me. 
[FZ3480-AS3935-lighting-detector.zip](https://github.com/esphome/issues/files/4811471/FZ3480-AS3935-lighting-detector.zip)

Veldkornet, 2020-09-26:
Same issue for me. 
`Indoor: false` stops all of the messages, but it doesn’t seem to collect any lightning data either, always “unknown”

closed 2021-02-07 by stale[bot]

Yersi88, 2021-07-07:
I have same problem: Disturber was detected - try increasing the spike rejection value! 

I tried trigger the module with lighter piezo but the entities stay in "unknown" state and storm alarm is always "off"

Anyone knows fix?