esphome/issues #986: Tuya checksum errors when two commands are sent before the first response is received

noamfreedman, 2020-01-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io with ESPHome version 1.14.3

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 on a Feit Electric Wi-Fi Smart Dimmer labeled BPDIM/WIFI/2 and item # 1358969

**Affected component:**

https://esphome.io/components/light/tuya.html

**Description of problem:**

On several dimmers, I see a problem where I get occasional checksum errors.   In looking at the logs, the common denominator seems to be that two commands are sent close enough together that the response from the first wasn't yet received. This could be a user generated command and a heartbeat or two user generated commands.  The only time I see this scenario happen successfully is when the first command is a heartbeat.

**Problem-relevant YAML-configuration entries:**
```yaml
---
esphome:
  name: dimmer
  platform: ESP8266
  board: esp01_1m

# WiFi connection
wifi:
  ssid: !SSID
  password: !PASSWORD

  ap:
    ssid: !SSID
    password: !PASSWORD

captive_portal:
   password: !PASSWORD

# Enable logging
logger:
  baud_rate: 0
  level: VERBOSE

# Enable Home Assistant API
api:
   password: !PASSWORD

# Enable over-the-air updates
ota:
   password: !PASSWORD

# Enable Web server
web_server:
  port: 80

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

tuya:

light:
  - platform: "tuya"
    name: "Dimmer"
    dimmer_datapoint: 2
    switch_datapoint: 1
    min_value: 0
    max_value: 1000
```

**Logs (if applicable):**

```
First example:
[17:50:31][V][tuya:275]: Datapoint 1 set to 0
[17:50:31][V][tuya:261]: Sending Tuya: CMD=0x06 VERSION=0 DATA=[01.01.00.01.00 (5)] INIT_STATE=5
[17:50:31][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[17:50:31][W][tuya:091]: Tuya Received invalid message checksum 01!=10

Second example:
[16:42:54][V][tuya:275]: Datapoint 2 set to 1000
[16:42:54][V][tuya:261]: Sending Tuya: CMD=0x06 VERSION=0 DATA=[02.02.00.04.00.00.03.E8 (8)] INIT_STATE=5
[16:42:54][V][tuya:275]: Datapoint 1 set to 1
[16:42:54][V][tuya:261]: Sending Tuya: CMD=0x06 VERSION=0 DATA=[01.01.00.01.01 (5)] INIT_STATE=5
[16:42:54][W][tuya:091]: Tuya Received invalid message checksum 07!=1B

Here is two successful examples where the first command was a heartbeat:
[17:50:30][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[17:50:30][V][tuya:275]: Datapoint 2 set to 1000
[17:50:30][V][tuya:279]: Not sending unchanged value
[17:50:30][V][tuya:275]: Datapoint 1 set to 1
[17:50:30][V][tuya:261]: Sending Tuya: CMD=0x06 VERSION=0 DATA=[01.01.00.01.01 (5)] INIT_STATE=5
[17:50:30][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=3 DATA=[01 (1)] INIT_STATE=5
[17:50:30][V][tuya:115]: MCU Heartbeat (0x01)
[17:50:30][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=3 DATA=[01.01.00.01.01 (5)] INIT_STATE=5
[17:50:30][V][tuya:235]: Datapoint 1 update to 1
 

[17:50:46][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[17:50:46][V][tuya:275]: Datapoint 2 set to 760
[17:50:46][V][tuya:261]: Sending Tuya: CMD=0x06 VERSION=0 DATA=[02.02.00.04.00.00.02.F8 (8)] INIT_STATE=5
[17:50:46][V][tuya:275]: Datapoint 1 set to 1
[17:50:46][V][tuya:279]: Not sending unchanged value
[17:50:46][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=3 DATA=[01 (1)] INIT_STATE=5
[17:50:46][V][tuya:115]: MCU Heartbeat (0x01)
[17:50:46][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=3 DATA=[02.02.00.04.00.00.02.F8 (8)] INIT_STATE=5
[17:50:46][V][tuya:235]: Datapoint 2 update to 760

```

**Additional information and things you've tried:**

brandond, 2020-01-12:
I've not seen that before. Is this after soft reset, or a cold powerup?

noamfreedman, 2020-01-12:
I see it in both on loading a new firmware as well as on cold power up. It is obviously intermittent in either case since it requires user interaction combined with unfavorable timing.

brandond, 2020-01-12:
The current log output isn't super helpful, in case of a bad cksum it should print the bad buffer or at least the first bit of it so I have some idea what it's actually received. Let me put something together.

noamfreedman, 2020-01-13:
How about this?  Here is a log where it prints out the message received in those situations.  I ran a script to toggle 1000 times:

```
[02:26:31][V][tuya:266]: Sending Tuya: CMD=0x06 VERSION=0 DATA=[01.01.00.01.01 (5)] INIT_STATE=3
[02:26:31][V][tuya:266]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[02:26:31][W][tuya:091]: Tuya Received invalid message checksum 00!=11
[02:26:31][V][tuya:095]: Message with bad checksum: CMD=0x07 VERSION=3 DATA=[01.55.AA.03.00 (5)] INIT_STATE=3

[02:26:33][V][tuya:266]: Sending Tuya: CMD=0x06 VERSION=0 DATA=[01.01.00.01.00 (5)] INIT_STATE=3
[02:26:33][V][tuya:266]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[02:26:33][W][tuya:091]: Tuya Received invalid message checksum 00!=12
[02:26:33][V][tuya:095]: Message with bad checksum: CMD=0x07 VERSION=3 DATA=[01.01.55.AA.03 (5)] INIT_STATE=3

[02:26:42][V][tuya:266]: Sending Tuya: CMD=0x06 VERSION=0 DATA=[01.01.00.01.00 (5)] INIT_STATE=3
[02:26:42][V][tuya:266]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[02:26:42][W][tuya:091]: Tuya Received invalid message checksum 00!=11
[02:26:42][V][tuya:095]: Message with bad checksum: CMD=0x07 VERSION=3 DATA=[01.55.AA.03.00 (5)] INIT_STATE=3

[02:26:44][V][tuya:266]: Sending Tuya: CMD=0x06 VERSION=0 DATA=[01.01.00.01.00 (5)] INIT_STATE=3
[02:26:44][V][tuya:266]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[02:26:44][W][tuya:091]: Tuya Received invalid message checksum 00!=11
[02:26:44][V][tuya:095]: Message with bad checksum: CMD=0x07 VERSION=3 DATA=[01.55.AA.03.00 (5)] INIT_STATE=3

[02:26:45][V][tuya:266]: Sending Tuya: CMD=0x06 VERSION=0 DATA=[01.01.00.01.01 (5)] INIT_STATE=3
[02:26:45][V][tuya:266]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[02:26:45][W][tuya:091]: Tuya Received invalid message checksum 00!=11
[02:26:45][V][tuya:095]: Message with bad checksum: CMD=0x07 VERSION=3 DATA=[01.55.AA.03.00 (5)] INIT_STATE=3

[02:26:47][V][tuya:266]: Sending Tuya: CMD=0x06 VERSION=0 DATA=[01.01.00.01.01 (5)] INIT_STATE=3
[02:26:47][V][tuya:266]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[02:26:47][W][tuya:091]: Tuya Received invalid message checksum 00!=12
[02:26:47][V][tuya:095]: Message with bad checksum: CMD=0x07 VERSION=3 DATA=[01.01.55.AA.03 (5)] INIT_STATE=3

[02:26:55][V][tuya:266]: Sending Tuya: CMD=0x06 VERSION=0 DATA=[01.01.00.01.00 (5)] INIT_STATE=3
[02:26:55][V][tuya:266]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[02:26:55][W][tuya:091]: Tuya Received invalid message checksum 01!=10
[02:26:55][V][tuya:095]: Message with bad checksum: CMD=0x07 VERSION=3 DATA=[55.AA.03.00.00 (5)] INIT_STATE=3

[02:26:58][V][tuya:266]: Sending Tuya: CMD=0x06 VERSION=0 DATA=[01.01.00.01.00 (5)] INIT_STATE=3
[02:26:58][V][tuya:266]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[02:26:58][W][tuya:091]: Tuya Received invalid message checksum 00!=11
[02:26:58][V][tuya:095]: Message with bad checksum: CMD=0x07 VERSION=3 DATA=[01.55.AA.03.00 (5)] INIT_STATE=3

[02:27:00][V][tuya:266]: Sending Tuya: CMD=0x06 VERSION=0 DATA=[01.01.00.01.00 (5)] INIT_STATE=3
[02:27:00][V][tuya:266]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[02:27:00][W][tuya:091]: Tuya Received invalid message checksum 03!=0F
[02:27:00][V][tuya:095]: Message with bad checksum: CMD=0x07 VERSION=3 DATA=[01.01.00.55.AA (5)] INIT_STATE=3

[02:27:05][V][tuya:266]: Sending Tuya: CMD=0x06 VERSION=0 DATA=[01.01.00.01.00 (5)] INIT_STATE=3
[02:27:05][V][tuya:266]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=3
[02:27:05][W][tuya:091]: Tuya Received invalid message checksum 01!=10
[02:27:05][V][tuya:095]: Message with bad checksum: CMD=0x07 VERSION=3 DATA=[55.AA.03.00.00 (5)] INIT_STATE=3
```

brandond, 2020-01-13:
Can you get the logs from a clean power-up boot and then a few changes to the dimmer, both via the input on the device and via HA? It looks like we're getting some truncated responses from the device, as I see 55 AA in some of the RX payloads, which is the header for a new packet.

noamfreedman, 2020-01-13:
Sure thing.  The first set of transitions were on the device and then the next were remote (technically not HA as I'm doing some curls for the testing).

[tuya_output.log](https://github.com/esphome/issues/files/4056388/tuya_output.log)

brandond, 2020-01-14:
I'll have a look at the code and see if there's anything we're doing wrong on our side, but it looks to me like the embedded MCU has a bug where it doesn't handle multiple back-to-back messages. If we send a set command and a heartbeat in quick succession, it stops sending the response to the set command halfway through and starts sending the response to the ack instead.

We can probably work around this by enforcing some sort of delay between messages, but this is definitely a quirk - I have logic analyzer dumps from other MCUs sending and receiving messages with absolutely zero delays in between.

noamfreedman, 2020-01-14:
I did some further testing with a script that does 10k sequential toggles.  I added logic to keep track of when the last un-responded command was sent.  If another command was trying to be sent within a set period, I delayed until that period was up. By setting that period to 50ms, I  reduced the error rate down to roughly three in 10k commands. Increasing the delay up to 250ms didn't do any better.  I'm guessing that doing better would require queuing the commands to allow for the receive buffer to be cleared while the command is delayed.  Given that I last did any c coding was over 15 years ago, that is a step beyond my capabilities right now. :)

I simply added a tracking variable to the Tuya class, cleared it at the beginning of Tuya::handle_command_ and then added the following logic at the beginning of Tuya::send_command_:

```c++
  uint32_t now = millis();
  if (command_waiting_ == 0 || now - command_waiting_ > 50) {
    command_waiting_ = millis();
  } else {
    delay( 50 - now + command_waiting_ );
    command_waiting_ = millis();
  }```

Attached is the log of running through 10k changes with this modification:  [output_50ms.log.txt](https://github.com/esphome/issues/files/4057277/output_50ms.log.txt)

brandond, 2020-01-14:
Yeah, having to wait for the response to come in before sending a new request is irritating. That seems like the only way to fix this 100%.

Where did you get this dimmer? If it's available domestically I might try to pick one up and grab a trace with a logic analyzer to see what's actually going on.

noamfreedman, 2020-01-15:
Yes, they are from Costco.

My guess is that the test patch I applied is sufficient for most any situation.  Ideally there would be no checksum errors, but to only see three when I'm hammering it with 10,000 toggle commands is a huge improvement and likely means it would be extremely rare in the wild.  Not sure it is worth the additional complexity to queue commands while waiting for a response.

backcountrymountains, 2020-03-02:
I also have the Costco Feit dimmers and am seeing the checksum errors. 
Can I get a dumbed down version of how to implement this fix?
Thanks.

backcountrymountains, 2020-03-16:
So I've attempted to apply this fix into my dimmers. I have no idea what I'm doing.

I copied the tuya.cpp and tuya.h from @noamfreedman into my homeassistant config directory.

Then I connected to Hassio on port 22222 as root and did the following:

```
docker -ps
docker exec -it "id of esphome docker" /bin/bash
find . -name tuya.cpp

cd ./opt/esphome/esphome/components/tuya/

cp tuya.cpp tuya.cpp.bak
cp /config/tuya.cpp tuya.cpp
cp tuya.h tuya.h.bak
cp /config/tuya.h tuya.h

#check that the tuya files have 'command_waiting_for_response'
cat tuya.h | grep command_waiting_for_response
cat tuya.cpp | grep command_waiting_for_response
```
I did a 'find' to find all instances of tuya.cpp and tuya.h, backed them up, and overwrote them with the modified tuya.cpp and tuya.h from @noamfreedman. 

I have no idea what I'm doing. Maybe it will work.

qwertangel, 2020-03-19:
so did it work ?

qwertangel, 2020-03-19:
can we get this fix added to tuya module ?

mihir2787, 2020-06-02:
I am seeing a similar issue. I have 4 of these dimmers that I purchased at Costco, and only 1 of the 4 dimmers is exhibiting this symptom (for now). The behavior that I see is that from HA, I can trigger the switch to turn ON the lights, but the Lights DO NOT turn ON or OFF . I have to physically do it. I had a reboot timeout which was introduced automatically in the template, but I removed it for this particular dimmer switch.
@noamfreedman do you see a similar behavior?

[kitchen-recess-lights-yaml.txt](https://github.com/esphome/issues/files/4718487/kitchen-recess-lights-yaml.txt)

[kitchen-recess-lights-log.txt](https://github.com/esphome/issues/files/4718492/kitchen-recess-lights-log.txt)

mihir2787, 2020-06-02:
Did the fix work for you? @noamfreedman

qwertangel, 2020-06-02:
@mihir2787 , issue you described is usually resolved once you power cycle the switch (at breaker)

closed 2021-04-04 by stale[bot]