esphome/issues #987: SCD30 (Sensirion - CO2, Temperature and Relative Humidty Sensor) does not transmit any measurements.

KuePe, 2020-01-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io, esphome
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 WROOM32 DevKit
**Affected component:**

SCD30 CO₂, Temperature and Relative Humidty Sensor
https://esphome.io/components/sensor/scd30.html

**Description of problem:**
Has anyone successfully tested the SCD30 integration (Sensirion CO2 sensor)? 
I only get the message "Data not ready yet!"

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: xxx_test_6
  upper_devicename: Test6

esphome:
  name: $devicename
  platform: ESP32
  board: esp32dev

i2c:
  id: bus_a
  sda: GPIO21
  scl: GPIO22
  scan: True
  frequency: 50kHz

sensor:
  - platform: scd30
    address: 0x61
    update_interval: 60s
    i2c_id: bus_a
    co2:
     name: ${upper_devicename} scd30 CO2
     accuracy_decimals: 1
    temperature:
     name: ${upper_devicename} scd30 Temperature
     accuracy_decimals: 2
    humidity:
     name: ${upper_devicename} scd30 Humidity
     accuracy_decimals: 1

```

**Logs (if applicable):**

```
[18:06:34][I][i2c:033]: Scanning i2c bus for active devices...
[18:06:34][I][i2c:040]: Found i2c device at address 0x61

[18:19:51][VV][i2c:056]: Beginning Transmission to 0x61:
[18:19:51][VV][i2c:093]:     Writing 0b00000010 (0x02)
[18:19:51][VV][i2c:093]:     Writing 0b00000010 (0x02)
[18:19:51][VV][i2c:061]:     Transmission ended. Status code: 0x00
[18:19:51][VV][i2c:083]: Requesting 3 bytes from 0x61:
[18:19:51][VV][i2c:113]:     Received 0b00000000 (0x00)
[18:19:51][VV][i2c:113]:     Received 0b00000000 (0x00)
[18:19:51][VV][i2c:113]:     Received 0b10000001 (0x81)
[18:19:51][W][scd30:086]: Data not ready yet!

```

**Additional information and things you've tried:**
Other I2C sensors (such as the CS811) have been successfully tested.

Comment on API Description:
Detailed Description «This class implements support for the Sensirion scd30 i2c GAS (VOC and CO2eq) sensors.»  is not correct.
The SCD30 is a CO2 Sensor and not a not a VOC and CO2eq Sensor

API Reference:
https://esphome.io/api/scd30_8h.html
https://esphome.io/api/classesphome_1_1scd30_1_1_s_c_d30_component.html#details

Interface Description Sensirion SCD30 Sensor Module
https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/9.5_CO2/Sensirion_CO2_Sensors_SCD30_Interface_Description.pdf

Sizurka, 2020-01-14:
I had similar issues getting a "fresh from factory" one working.  Since most of the configuration gets stored in the sensor non-volatile memory, I think the original author's testing had one that was already configured (probably by another library).

Can you try https://github.com/Sizurka/esphome/tree/scd30-fixes ?  That contains the fixes I had to do get mine working, but with a sample size of one, it's not exactly conclusive.

KuePe, 2020-01-15:
Thank you for your answer. 
I will be happy to test the fix. 

What should I do? 
Should I replace the two files "scd30.cpp" and "scd30.h" in the directory "/config/esphome/muutu_test_6/src/esphome/components/scd30?" 
What should I do with the file "sensor.py"? 
How do I download the files from GitHub?

Sizurka, 2020-01-15:
The simplest way to test without a full development environment would be to override it with a custom component.  So in this case, you would create: `<YourConfigDirectory>/esphome/custom_components/scd30/` then put the four files in to that directory.  That should make ESPHome use the overridden parts instead of the built in one the next time you compile/upload.

To download from github, you can either download the whole thing - the "Clone or download" button then "Download ZIP" will get the branch you have selected, so the above link should get you the right one - then extract the files from it (`esphome-scd30-fixes/esphome/components/scd30/`).  Alternatively you can navigate to the changed contents ( https://github.com/Sizurka/esphome/tree/scd30-fixes/esphome/components/scd30 ), click on the file, then click on the "Raw" button and save the result.

In either case, you should be able to verify that you've done it correctly by trying to set the new option for automatic self calibration.  If ESPHome rejects it, then something went wrong.  So only the updated one would accept something like:

```yaml
sensor:
  - platform: scd30
    automatic_self_calibration: true
````

KuePe, 2020-01-16:
Thank you so much, it works. 
I tested two brand new SCD30 sensors. 
In the next few days I will continue to test and pass on the information. 
Great support.

TheHackmeister, 2020-01-20:
I was also having this issue. Oddly enough, it worked fine on an ESP32, but not on the ESP8266 I had. With these fixes, no problems :).

closed 2020-03-12 by brandond