esphome/issues #988: Modifying yaml file unable to install binary

btoconnor, 2020-01-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Host OS: Ubuntu
Running via pip in a virtualenv, python 3.6

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**Affected component:**

Installation of binary

**Description of problem:**
I ran esphome a few months ago and installed my configuration on 2 separate ESP32 chips without a problem.

I wanted to test a few things out today, so I modified the scripts and ran `esphome dht22test.yaml run` and I continue to get this error:
```
UnsupportedArchiveType: Can not unpack file '/home/brian/Programming/esp32env/dht22test/.piolibdeps/dht22test/_tmp_installing-13d65vs4-package/None':
  File "/home/brian/Programming/esp32env/lib/python3.6/site-packages/platformio/builder/main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/brian/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/brian/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/brian/.platformio/platforms/espressif32/builder/main.py", line 209:
    target_elf = env.BuildProgram()
  File "/home/brian/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/home/brian/Programming/esp32env/lib/python3.6/site-packages/platformio/builder/tools/platformio.py", line 122:
    _build_project_deps(env)
  File "/home/brian/Programming/esp32env/lib/python3.6/site-packages/platformio/builder/tools/platformio.py", line 47:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "/home/brian/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/home/brian/Programming/esp32env/lib/python3.6/site-packages/platformio/builder/tools/piolib.py", line 1043:
    project.install_dependencies()
  File "/home/brian/Programming/esp32env/lib/python3.6/site-packages/platformio/builder/tools/piolib.py", line 878:
    lm.install(uri)
  File "/home/brian/Programming/esp32env/lib/python3.6/site-packages/platformio/managers/lib.py", line 359:
    force=force)
  File "/home/brian/Programming/esp32env/lib/python3.6/site-packages/platformio/managers/package.py", line 703:
    pkg_dir = self._install_from_piorepo(name, requirements)
  File "/home/brian/Programming/esp32env/lib/python3.6/site-packages/platformio/managers/lib.py", line 220:
    if app.get_setting("strict_ssl") else dl_data['url'], requirements)
  File "/home/brian/Programming/esp32env/lib/python3.6/site-packages/platformio/managers/package.py", line 483:
    self.unpack(dlpath, tmp_dir)
  File "/home/brian/Programming/esp32env/lib/python3.6/site-packages/platformio/managers/package.py", line 212:
    with FileUnpacker(source_path) as fu:
  File "/home/brian/Programming/esp32env/lib/python3.6/site-packages/platformio/unpacker.py", line 106:
    raise exception.UnsupportedArchiveType(self.archpath)
```

**Problem-relevant YAML-configuration entries:**
Original YAML file (creds redacted)
```yaml
esphome:
  name: dht22test
  platform: ESP32
  board: node32s

wifi:
  ssid: "myssid"
  password: "mywifipassword"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Dht22Test Fallback Hotspot"
    password: "M3XWYHdDTQJt"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

sensor:
  - platform: dht
    pin: 22
    temperature:
      name: "DHT22 Temperature"
    humidity:
      name: "DHT22 Humidity"
    update_interval: 60s
```
I am trying to add:
```yaml
web_server:
  port: 80
```
to the configuration, and anytime I modify it, I get the error above.  Removing the `web_server` block allows me to run the installation as expected.

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
I'm seeing this on a number of different configuration setups.  At first, I just tried copying my yaml file and modifying a few values, and that produced the issue.  I then went and made a new yaml file using the wizard, and that does allow me to run the installation.  However, modifying that file at any point produces the error again.

brandond, 2020-01-14:
The error appears to be somewhere in the platformio code. Have you tried updating both esphome and platformio using pip?

btoconnor, 2020-01-14:
Yeah - I took a look at that.  I removed the `~/.platformio/` directory and things are acting more normal-ish.  It's a little confusing that running `esphome config.yaml clean` doesn't remove any caches like this, but maybe that's out of scope.

closed 2020-08-07 by stale[bot]